machinekit/machinekit #1131: Protobuf is not in sync with machinetalk-protobuf

machinekoder, 2017-02-12:
This patch brings the Machinetalk Protobuf definitions out of sync: https://github.com/machinekit/machinekit/commit/746398418a76b91632e75076c9c43875f971d48a

machinekoder, 2017-02-12:
Related to https://github.com/machinekit/machinekit/issues/1123

ArcEye, 2017-02-13:
As I noted in the original commit, it was bound to break something, somewhere, because the 2 lots of code had developed unaware of each other and then one repo had not been touched for 2 years.

You are probably best placed to know what needs doing to protobuf.

ArcEye, 2017-02-13:
I think this closed automatically because your PR text had the word `fix` just before the reference.
I'll leave it as it is, so you can re-open if required.

closed 2017-02-19 by machinekoder