machinekit/machinekit #1137: Cython build problem on current master

machinekoder, 2017-02-15:
Did anyone already experience this build problem on Debian Jessie?
```
Compiling hal/cython/machinekit/hal.c
hal/cython/machinekit/hal.c:1593:61: error: unknown type name ‘comp_type_t’
 static CYTHON_INLINE PyObject* __Pyx_PyInt_From_comp_type_t(comp_type_t value);
                                                             ^
hal/cython/machinekit/hal.c:1596:62: error: unknown type name ‘comp_state_t’
 static CYTHON_INLINE PyObject* __Pyx_PyInt_From_comp_state_t(comp_state_t value);
                                                              ^
hal/cython/machinekit/hal.c: In function ‘__pyx_f_10machinekit_8hal_util_pypin_value’:
hal/cython/machinekit/hal.c:31565:86: warning: passing argument 1 of ‘pin_value’ discards ‘const’ qualifier from pointer target type
   __pyx_t_1 = __pyx_f_10machinekit_8hal_util_hal2py(pin_type(__pyx_v_pin), pin_value(__pyx_v_pin)); if (unlikely(!__pyx_t_1)) __PYX_ERR(16, 28, __pyx_L1_error)
                                                                                      ^
In file included from hal/lib/hal_group.h:5:0,
                 from hal/cython/machinekit/hal.c:433:
hal/lib/hal_priv.h:389:27: note: expected ‘struct hal_pin *’ but argument is of type ‘const struct hal_pin *’
 static inline hal_data_u *pin_value(hal_pin_t *pin) {
                           ^
hal/cython/machinekit/hal.c: In function ‘inithal’:
hal/cython/machinekit/hal.c:35886:3: warning: implicit declaration of function ‘__Pyx_PyInt_From_comp_type_t’ [-Wimplicit-function-declaration]
   __pyx_t_2 = __Pyx_PyInt_From_comp_type_t(TYPE_INVALID); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 35, __pyx_L1_error)
   ^
hal/cython/machinekit/hal.c:35886:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_type_t(TYPE_INVALID); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 35, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:35898:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_type_t(TYPE_RT); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 36, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:35910:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_type_t(TYPE_USER); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 37, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:35922:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_type_t(TYPE_REMOTE); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 38, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:35934:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_type_t(TYPE_HALLIB); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 39, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:35946:3: warning: implicit declaration of function ‘__Pyx_PyInt_From_comp_state_t’ [-Wimplicit-function-declaration]
   __pyx_t_2 = __Pyx_PyInt_From_comp_state_t(COMP_INVALID); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 41, __pyx_L1_error)
   ^
hal/cython/machinekit/hal.c:35946:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_state_t(COMP_INVALID); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 41, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:35958:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_state_t(COMP_INITIALIZING); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 42, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:35970:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_state_t(COMP_UNBOUND); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 43, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:35982:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_state_t(COMP_BOUND); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 44, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:35994:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_state_t(COMP_READY); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 45, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:36198:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_type_t(TYPE_INVALID); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 68, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:36210:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_type_t(TYPE_RT); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 69, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:36222:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_type_t(TYPE_USER); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 70, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c:36234:13: warning: assignment makes pointer from integer without a cast
   __pyx_t_2 = __Pyx_PyInt_From_comp_type_t(TYPE_REMOTE); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 71, __pyx_L1_error)
             ^
hal/cython/machinekit/hal.c: At top level:
hal/cython/machinekit/hal.c:39730:75: error: unknown type name ‘comp_type_t’
               static CYTHON_INLINE PyObject* __Pyx_PyInt_From_comp_type_t(comp_type_t value) {
                                                                           ^
hal/cython/machinekit/hal.c:39761:76: error: unknown type name ‘comp_state_t’
               static CYTHON_INLINE PyObject* __Pyx_PyInt_From_comp_state_t(comp_state_t value) {
                                                                            ^
Makefile:402: recipe for target 'objects/hal/cython/machinekit/hal.o' failed
make: *** [objects/hal/cython/machinekit/hal.o] Error 1
make: Leaving directory '/home/alexander/bin/machinekit-1/src'
```

machinekoder, 2017-02-15:
Maybe related to https://github.com/machinekit/machinekit/issues/1123

ArcEye, 2017-02-16:
I have had similar output regards comp_state_t, but cannot remember what platform and OS.
I have been building for amd64 and armhf with Jessie and Stretch recently.
Whatever it was, I cleaned up the repo and it built OK again.

What platform are you building on, which compiler etc?

> Maybe related to #1123

Don't think so.  The current master builds fine.
I did it 'ad nauseum' yesterday on Jessie, plus the packages just all built on Travis

ArcEye, 2017-02-16:
Just built fresh clone of repo on Jessie without any issues
gcc-4.9.2 / cython 0.21.1 / 4.6.5-rt10 amd64 kernel

Tried to run your patch on a copy of master, but it throws all sorts of errors
Looks like it was based upon the master before the https://github.com/machinekit/machinekit/commit/1e1df875f9fb6ae53f19d39efd1ae47caf07f0e3 merge

Can only suggest you do a fresh pull, test a build and then apply your patch to that and re-submit.
Hopefully the cython error will disappear and what the clashes in the patch are will be apparent to you

machinekoder, 2017-02-19:
I still get this error.

GCC version:
```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.9/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 4.9.2-10' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --with-arch-32=i586 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.9.2 (Debian 4.9.2-10)
```

Debian version
```
$ cat /etc/debian_version 
8.7
```

Cython version:
```
$ cython --version
Cython version 0.25.2
$ dpkg -s cython
Package: cython
Status: install ok installed
Priority: optional
Section: python
Installed-Size: 8173
Maintainer: Python Applications Packaging Team <python-apps-team@lists.alioth.debian.org>
Architecture: amd64
Version: 0.25.2-1~bpo8+1
Depends: python (>= 2.7), python (<< 2.8), libc6 (>= 2.14)
Recommends: python-dev, gcc
Suggests: cython-doc
Description: C-Extensions for Python
 Cython is a language that makes writing C extensions for the Python language as
 easy as Python itself. Cython is based on the well-known Pyrex, but supports
 more cutting edge functionality and optimizations.
 .
 The Cython language is very close to the Python language, but Cython
 additionally supports calling C functions and declaring C types on variables
 and class attributes. This allows the compiler to generate very efficient C
 code from Cython code.
 .
 This makes Cython the ideal language for wrapping external C libraries, and
 for fast C modules that speed up the execution of Python code.
Homepage: http://cython.org/
```

ArcEye, 2017-02-19:
>  Cython version 0.25.2

I think this will be the problem, I am using 0.21.1, which is the latest version for Jessie.

I would suspect you have enabled  Jessie backports, which will bring you 0.25.2-1
https://packages.debian.org/jessie-backports/cython from sid

I had this problem when trying to build on Stretch AFAIR.
I did clear it, but cannot remember how :confused:

closed 2017-02-19 by machinekoder