machinekit/machinekit #1142: Watchdog interface has changed

machinekoder, 2017-02-20:
The interface of the watchdog component has changed due to port to instcomp:
```bash
$ halcmd loadrt watchdog num_inputs=3
<commandline>:0: 
Invalid argument 'num_inputs=3' to instantiated component
NB. Use of personality or cfg is deprecated
```

The new parameter is called `pincount`. Should we really break backward compatibility?

machinekoder, 2017-02-20:
We should also change:
https://github.com/machinekit/machinekit/blob/master/lib/python/fdm/config/base.py#L34
accordingly

ArcEye, 2017-02-21:
> Should we really break backward compatibility?

There is no backward compatibility for all the multi-various module args that had grown up in legacy rt modules, 
halcmd does not support loadrt args for instcomps, other than `count=` and `names=`.

In that respect it is fully compatible with every other instantiated component, in that pincount controls the number of pins within an instance.

I had not realised the python libs used it, I will do a PR substituting `newinst` for `loadrt` in that code

PR #1143 (2017-02-21): Amend python function to use instantiated version of watchdog
Fixes #1142
commit 01975331 (2017-02-21): Amend python function to use instantiated version of watchdog

Fixes #1142
closed 2017-02-21 by machinekoder