machinekit/machinekit #1145: HAL rtapi

pmcstone, 2017-02-22:
I do not know if this is related to the merge but this happened after I upated

Traceback (most recent call last):
  File "hal/init.py", line 14, in <module>
    import motion
  File "/home/pmcs/Downloads/pmcs-sim_OLD/hal/motion.py", line 14, in <module>
    convert = rt.newinst('float2u32', 'convert.active-origin-float')
  File "machinekit/rtapi.pyx", line 264, in machinekit.rtapi.RTAPIcommand.newinst (hal/cython/machinekit/rtapi.c:5506)
RuntimeError: rtapi_newinst '('float2u32', 'convert.active-origin-float')' failed: Operation not permitted

ArcEye, 2017-02-23:
The component is non-standard, so I need a link to it.

The errors from newinst are often misleading, try from a terminal.
```
DEBUG=5 realtime restart
halcmd newinst float2u32 convert.active-origin-float
```
Then look in linuxcnc.log in the lines before the error and the cause is often clearer.

machinekoder, 2017-02-23:
I cannot find a `float2u32` component in the Machinekit repository. This must be a component you have imported from a third party configuration.

However, the `conv_float_u32` component should provide similar functionality. I have just created a PR for the components being available as Instcomps.

```
halcmd newinst conv_float_u32 convert.active-origin-float
$ halcmd show pin
Component Pins:
  Comp   Inst Type  Dir         Value  Name                                            Epsilon Flags  linked to:
    78     80 bit   I/O         FALSE  convert.active-origin-float.clamp                --l-
    78     80 s32   OUT             0  convert.active-origin-float.funct.time           ----
    78     80 s32   I/O             0  convert.active-origin-float.funct.tmax           ----
    78     80 bit   OUT         FALSE  convert.active-origin-float.funct.tmax-i         ----
    78     80 float IN              0  convert.active-origin-float.in           0.000010        --l-
    78     80 u32   OUT    0x00000000  convert.active-origin-float.out                  --l-
    78     80 bit   OUT         FALSE  convert.active-origin-float.out-of-range         --l-
```

closed 2017-02-23 by machinekoder

ArcEye, 2017-02-23:
@machinekoder 
Thanks

@pmcstone 
I would still like to see the source please, so that I can see why it failed, to make sure it is not anything generally applicable.

pmcstone, 2017-02-23:
`#!/usr/bin/python

import sys
import os
import subprocess
import argparse
import time
from machinekit import launcher
from machinekit import rtapi

os.chdir(os.path.dirname(os.path.realpath(__file__)))

parser = argparse.ArgumentParser(description='This is the motorctrl demo run script '
                                 'it demonstrates how a run script could look like '
                                 'and of course starts the motorctrl demo')
parser.add_argument('-s', '--halscope', help='Starts the halscope', action='store_true')
parser.add_argument('-m', '--halmeter', help='Starts the halmeter', action='store_true')
parser.add_argument('-p', '--path', help='Adds an additional UI search path to configserver')
parser.add_argument('-d', '--debug', help='Enable debug mode', action='store_true')

args = parser.parse_args()

if args.debug:
    launcher.set_debug_level(5)

#launcher.set_machinekit_ini('./configs/machinekit.ini')

def check_mklaucher():
    try:
        subprocess.check_output(['pgrep', 'mklauncher'])
        return True
    except subprocess.CalledProcessError:
        return False

try:
    launcher.check_installation()
    launcher.cleanup_session()
    launcher.install_comp('./components/float2u32.icomp')
    launcher.install_comp('./components/statehandler.icomp')
    launcher.install_comp('./components/pinwatchdog.icomp')
    launcher.install_comp('./components/speedhandler.icomp')
    launcher.install_comp('./components/simchanger.icomp')
    launcher.register_exit_handler()  # needs to executed after HAL files

    # ensure gcode paths
    nc_path = os.path.expanduser('./nc_files/external')
    if not os.path.exists(nc_path):
        os.mkdir(nc_path)
    nc_path = os.path.expanduser('./nc_files/network')
    if not os.path.exists(nc_path):
        os.mkdir(nc_path)

    # ensure user config paths
    user_path = os.path.expanduser('./configs/user')
    if not os.path.exists(user_path):
        os.mkdir(user_path)
    config_paths = ['atc', 'origins', 'tools']
    for path in config_paths:
        config_path = os.path.join(user_path, 'atc')
        if not os.path.exists(config_path):
            os.mkdir(config_path)

    if not check_mklaucher():  # start mklauncher if not running to make things easier
        launcher.start_process('mklauncher .')

    # the point-of-contact for QtQUickVCP
    launcher.start_process('configserver -n PMC-Stone . ~/repos/pmcs-ui %s' % args.path)

    # remote access to import/export files
    launcher.start_process('./python/pmcsfile_service.py -p ./configs/user')

    # start machinekit
    launcher.start_process('machinekit v6.ini')

    if args.halscope:
        # load scope only now - because all sigs are now defined:
        launcher.start_process('halscope')
    if args.halmeter:
        launcher.start_process('halmeter')

    while True:
        launcher.check_processes()
        time.sleep(1)

except subprocess.CalledProcessError:
    launcher.end_session()
    sys.exit(1)

sys.exit(0)
`
I had to manually install these icomps through the terminal

pmcstone, 2017-02-23:
Manually installing the icomps seemed to resolved the issue

ArcEye, 2017-02-23:
> Manually installing the icomps seemed to resolved the issue

Good, it is not what I was expecting, I don't use python unless I have to, 
this is more in @machinekoder s area :smile: