machinekit/machinekit #1152: Usage of deprecated API breaks compilation with czmq > v4.0.x

dwrobel, 2017-02-25:
Usage of [czmq deprecated API](https://github.com/zeromq/czmq/blob/master/NEWS) breaks the compilation while using with czmq > v4.0.x version.

Observed on Fedora >= 26 which will use [czmq v4.x](https://koji.fedoraproject.org/koji/buildinfo?buildID=860200) version.

I'm not sure if and when you will hit the same on Debian...

Compilation log excerpt:
```
Compiling machinetalk/support/npbdecode.c
gcc -c  -I. -Imachinetalk/build -Imachinetalk/nanopb -Imachinetalk/include -Ilibnml/linklist -Ilibnml/cms -Ilibnml/rcs -Ilibnml/inifile -Ilibnml/os_intf -Ilibnml/nml -Ilibnml/buffer -Ilibnml/posemath -Irtapi -Irtapi_export -Ihal/lib -I../
        -MP -MD -MF "objects/machinetalk/support/npbdecode.d" -MT "objects/machinetalk/support/npbdecode.o" \
        machinetalk/support/npbdecode.c -o objects/machinetalk/support/npbdecode.o
In file included from machinetalk/lib/mk_service.cc:9:0:
machinetalk/include/mk-service.hh:19:5: error: 'zctx_t' does not name a type; did you mean 'zrex_t'?
     zctx_t        *z_context;
     ^~~~~~
     zrex_t
cp machinetalk/include/container.h ../include/container.h
Compiling machinetalk/support/linmove.cc
g++ -c  -I. -Imachinetalk/build -Imachinetalk/nanopb -Imachinetalk/include -Ilibnml/linklist -Ilibnml/cms -Ilibnml/rcs -Ilibnml/inifile -Ilibnml/os_intf -Ilibnml/nml -Ilibnml/buffer -Ilibnml/posemath -Irtapi -Irtapi_export -Ihal/lib -I../
        -MP -MD -MF "objects/machinetalk/support/linmove.d" -MT "objects/machinetalk/support/linmove.o" \
        machinetalk/support/linmove.cc -o objects/machinetalk/support/linmove.o
machinetalk/lib/mk_service.cc: In function 'int bind_ifs(mk_socket_t*, const argvec_t&)':
machinetalk/lib/mk_service.cc:111:12: error: 'zsocket_bind' was not declared in this scope
  s->port = zsocket_bind(s->socket, uri.c_str());
            ^~~~~~~~~~~~
machinetalk/lib/mk_service.cc:111:12: note: suggested alternative: 'zsock_bind'
  s->port = zsocket_bind(s->socket, uri.c_str());
            ^~~~~~~~~~~~
            zsock_bind
machinetalk/lib/mk_service.cc: In function 'int mk_bindsocket(mk_netopts_t*, mk_socket_t*)':
machinetalk/lib/mk_service.cc:156:3: error: 'zsocket_set_ipv6' was not declared in this scope
   zsocket_set_ipv6 (s->socket, 1);
   ^~~~~~~~~~~~~~~~
machinetalk/lib/mk_service.cc:156:3: note: suggested alternative: 'zsock_set_ipv6'
   zsocket_set_ipv6 (s->socket, 1);
   ^~~~~~~~~~~~~~~~
   zsock_set_ipv6
```

ArcEye, 2017-02-25:
>  I'm not sure if and when you will hit the same on Debian...

We will hit it with Stretch.  I have built MK on Stretch and had to use the older version of czmq to get it to build, ( else rewrite the code ) exactly the same errors.

The plus side is that the old lib works, which is probably going to be important, since we have trouble getting people to leave Wheezy for Jessie, so who knows when everyone will be on Stretch and it will be safe to use czmq4.0 :smile:

mbuesch, 2017-08-16:
What are the chances to support both library versions?
I currently have one machine on jessie (I think) and one on sid. But I guess I could update to stretch, if only the newest library is supported.

ArcEye, 2017-08-17:
See #1228 

The libraries are completely incompatible, but Stretch will run happily on the old one recompiled for it.

Longer term it needs someone with knowledge of and interest in haltalk etc to port across to the new API.

Not been any takers thus far.

mbuesch, 2017-08-17:
Ok, thanks.
Luckily the package can be downgraded to this version
http://snapshot.debian.org/package/czmq/3.0.2-10/
Compilation succeeds with this version.

ArcEye, 2017-08-17:
> Luckily the package can be downgraded to this version
http://snapshot.debian.org/package/czmq/3.0.2-10/
Compilation succeeds with this version.

That's good to know.
In due course, the intermediate path could be to import the snapshot packages to deb.machinekit.io
for stretch builds.
Will save us from having to maintain a package.

zultron, 2017-09-26:
I have a patch that updates MK to the CZMQ 4.0 API, will post PR shortly.

It does break support for CZMQ 3.0 API.  However, IIRC, the MK project carries its own CZMQ package for Jessie.  Once CZMQ 4.0 support is known to work, would dropping 3.0 support and updating the Jessie package be an acceptable path forward (barring unforeseen technical complications)?

ArcEye, 2017-09-26:
Look forward to seeing it.

The choice is either support the new API and back port to Jessie, or continue to use the old API and 'forward port' to Stretch.
Neither is particularly good, but option one is more sustainable longer term, as eventually Jessie will drop off.

Wheezy is losing all support from Debian in May 2018 and we should just mothball it at or before that point and no longer build packages for it.
So Wheezy can be left out of the equation.

zultron, 2017-09-26:
The changes in PR #1285 are completely untested except to compile under Stretch, which seems to have other issues.  To test, someone might try rebuilding the Stretch CZMQ package for Jessie; hopefully it will be straightforward.

ArcEye, 2017-09-26:
As per https://github.com/machinekit/machinekit/pull/1285#issuecomment-332263235

The way forward will probably to make a new branch in my fork and experiment there, pulling in your patch and then testing with rebuilt czmq-4 sources built under Jessie.

I have builders both for Jessie and Stretch set up and can just switch which fork they track.

Once all variations are proved, the new zmq packages can be pushed to the main deb repo and one merge made into machinekit

ArcEye, 2017-09-27:
As a first stage, I have built the Stretch sources for czmq and zeromq for Jessie
The packages are libczmq4 (4.0.2-10\~jessie) and libzmq3 (4.0.4+dfg\~jessie)

Available on my repo at https://deb.mgware.co.uk
Only amd64 packages available at this time, as we are just testing.

Next stage will be to pull @zultron s CZMQ-4 API branch into a fork and test build in a binary install of dependencies that include the above packages.

ArcEye, 2017-09-27:
> Next stage will be to pull @zultron s CZMQ-4 API branch into a fork and test build in a binary install of dependencies that include the above packages.

That test is complete.
The packages install and machinekit builds a RIP install against them.

Starting machinekit results in the below error and machinekit hangs until the processes are killed manually

```
MACHINEKIT - 0.1
Machine configuration directory is '/usr/src/machinekit/configs/sim/axis'
Machine configuration file is 'axis_mm.ini'
Starting Machinekit...

halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout

E: 17-09-27 15:18:04 dangling 'DEALER' socket created at hal/utils/halcmd_rtapiapp.cc:281
io started
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout

E: 17-09-27 15:18:09 dangling 'DEALER' socket created at hal/utils/halcmd_rtapiapp.cc:281
halcmd loadusr io started
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout

E: 17-09-27 15:18:14 dangling 'DEALER' socket created at hal/utils/halcmd_rtapiapp.cc:281
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout

E: 17-09-27 15:18:19 dangling 'DEALER' socket created at hal/utils/halcmd_rtapiapp.cc:281
Shutting down and cleaning up Machinekit...
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout
```

hal/utils/halcmd_rtapiapp.cc:281 is
`z_command = zsock_new (ZMQ_DEALER);`
I know little about zmq but suspect that instantiation is valid and it is not getting deleted somewhere else or similar.

zultron, 2017-09-27:
@ArcEye, I saw the same thing in my tests on Stretch.  Looks like my first attempt needs some help!  I'll try again this weekend.  Thanks for testing.

ArcEye, 2017-09-27:
You're welcome.  It will be worth it when it comes together.

ArcEye, 2017-09-30:
FYI
Have now cross built the czmq / zeromq3 packages for i386 and armhf for Jessie
Also available on my repo at https://deb.mgware.co.uk

machinekoder, 2017-10-06:
Next try: https://github.com/machinekit/machinekit/pull/1287

closed 2017-10-11 by luminize

dwrobel, 2017-10-11:
The compilation fails on:

machinetalk/webtalk/webtalk_wsproxy.cc:339:65: error: macro "zsock_new" passed 2 arguments, but takes just 1
      wss->wsq_out = zsock_new (self->netopts.z_context, ZMQ_PAIR);
                                                                 ^
machinetalk/webtalk/webtalk_wsproxy.cc:343:64: error: macro "zsock_new" passed 2 arguments, but takes just 1
      wss->wsq_in = zsock_new (self->netopts.z_context, ZMQ_PAIR);
                                                                ^
machinetalk/webtalk/webtalk_wsproxy.cc:465:54: error: macro "zsock_destroy" passed 2 arguments, but takes just 1
   zsock_destroy (self->netopts.z_context, wss->socket);
                                                      ^
machinetalk/webtalk/webtalk_wsproxy.cc:466:57: error: macro "zsock_destroy" passed 2 arguments, but takes just 1
      zsock_destroy (self->netopts.z_context, wss->wsq_in);
                                                         ^
machinetalk/webtalk/webtalk_wsproxy.cc:467:58: error: macro "zsock_destroy" passed 2 arguments, but takes just 1
      zsock_destroy (self->netopts.z_context, wss->wsq_out);

dwrobel, 2017-10-11:
Just looking for zsock_new() it seems to partially converted properly and partially not:
- correct conversion: https://github.com/machinekit/machinekit/commit/afa2892b740f8a1e28903598ae9724de8f3fce29#diff-5c6692d5c926791ebb862589dc4bab01R281
- incorrect conversion: https://github.com/machinekit/machinekit/commit/afa2892b740f8a1e28903598ae9724de8f3fce29#diff-3e382ecce8d43e188760ad636bb43e3eR145

ArcEye, 2017-10-12:
>The compilation fails on:
machinetalk/webtalk/webtalk_wsproxy.cc:339:65: error: macro "zsock_new" passed 2 arguments, but takes just 1

That is because you have obviously used the `--enable-webtalk` switch to `configure` which builds do not.

Somehow go missed out of https://github.com/machinekit/machinekit/commit/ebca177771b1d878a6e11685ad25cfad1e3bfbe2 which corrected the other API changes.

ArcEye, 2017-10-12:
> Just looking for zsock_new() it seems to be partially converted properly and partially not:
    correct conversion: afa2892#diff-5c6692d5c926791ebb862589dc4bab01R281
    incorrect conversion: afa2892#diff-3e382ecce8d43e188760ad636bb43e3eR145

Now you are commenting on the first historical commit of 13 in that merge, which is corrected in the next commit
https://github.com/machinekit/machinekit/commit/ebca177771b1d878a6e11685ad25cfad1e3bfbe2#diff-3e382ecce8d43e188760ad636bb43e3eR144

Not much future in that.

dwrobel, 2017-10-12:
> That is because you have obviously used the --enable-webtalk switch to configure which builds do not.

Surprisingly, I'm using --disable-webtalk option, but this doesn't prevent webtalk_wsproxy.cc file from compilation process.

ArcEye, 2017-10-12:
You should not need to use a switch at all, it is the default.
The logic must be screwy in configure, will look at that later.

In the process of fixing the calls in webtalk, albeit there is one that may take some investigation
https://github.com/machinekit/machinekit/blob/master/src/machinetalk/webtalk/webtalk_main.cc#L141
I wasn't aware it could be called without an arg.

dwrobel, 2017-10-12:
> I wasn't aware it could be called without an arg.

It can't be, I'm on Fedora and the following:
`$ printf "#include <czmq.h>\nvoid foo() { zsock_new();}" | gcc -c -o /dev/null -xc++ -`

generates error as following:
`
In file included from /usr/include/czmq_library.h:160:0,
                 from /usr/include/czmq.h:30,
                 from <stdin>:1:
<stdin>: In function ‘void foo()’:
<stdin>:2:14: error: expected primary-expression before ‘)’ token
`

whereas the following compiles cleanly:
`$ printf "#include <czmq.h>\nvoid foo() { zsock_new(0);}" | gcc -c -o /dev/null -xc++ -`

ArcEye, 2017-10-12:
Unfortunately that does not answer the underlying question.

Entering `zsock_new(0)` is the same as 'zsock_new(ZMQ_PAIR)`
All the socket types are defines of int values.

@machinekoder 
can you shed some light on what this line is trying to do?

It can't be to produce a NULL pointer, that would fail the assert, so what type of socket pointer does it want?

dwrobel, 2017-10-12:
I understood that somehow zsock_new() compiles fine in your environment. So I prepared a test that proves that line https://github.com/machinekit/machinekit/blob/master/src/machinetalk/webtalk/webtalk_main.cc#L141 should generate a compilation error and it does on my environment.

In my example I passed '0' just to satisfies the compiler - please don't try to find what it means in czmq - it's irrelevant in this case.

Summing up - is zsock_new() generates error _(as it should)_ in your environment or not?

ArcEye, 2017-10-12:
It does not compile in my environment, because webtalk is not compiled at all by default.
I am sure it will fail, because it should.

What should be passed is far from irrelevant, I want it to work, not just build.
So I need to know what the thinking behind this call was, to know how it should be replaced.

There are no usages of `np->z_context` anywhere else in the webtalk code, so I am fairly content that it can just be removed, but am seeking clarification.

ArcEye, 2017-10-12:
Regards https://github.com/machinekit/machinekit/issues/1152#issuecomment-336063342

AFAICT, this call was made to provide a local context, within which the sockets etc. operated.

The major change to czmq4 is that all members operate within a global context which is deleted at shut down.

This is why calls to zsock_new() and zsock_destroy() just take an argument of the zsock_t * relating to the socket and do not require the context within which it was created.

Along with the other amendments, commenting out `webtalk_main:141` 
and the subsequent zsock_destroy() of that context at `webtalk_main:455 `
allows the build to complete successfully, with libwebtalk enabled, and it runs OK.

Making a PR based upon this.

commit afa2892b (2017-10-11): Update to CZMQ v. 4.0 API

Update to CZMQ v. 4.0 API in preparation for Stretch.  Implementation
follows the [zsock manual][1] and [CZMQ 4.0.0 NEWS][2].

Fixes #1152 [3].

[1]: http://czmq.zeromq.org/manual:zsock
[2]: https://github.com/zeromq/czmq/commit/052d2504
[3]: https://github.com/machinekit/machinekit/issues/1152