machinekit/machinekit #1155: Type mismatch in Machinetalk Protobuf and hal.h

machinekoder, 2017-02-28:
`hal.h` defines following types:
```c
    HAL_TYPE_UNSPECIFIED = -1,
    HAL_BIT = 1,
    HAL_FLOAT = 2,
    HAL_S32 = 3,
    HAL_U32 = 4,
    HAL_S64 = 5,
    HAL_U64 = 6,
```

However, `types.proto` defines:
```proto
    HAL_BIT   = 1;
    HAL_FLOAT = 2;
    HAL_S32   = 3;
    HAL_U32   = 4;
    STRING    = 5;
    BYTES     = 6;
```
Since the values are used to decode/encode Machinetalk messages this values should match.

I don't know any uses of the STRING and BYTES types. Therefore, I encourage to shif the values back to and to insert S64 and U64 in the Protobuf description.

ArcEye, 2017-02-28:
I take it you mean make STRING and BYTES 7 and 8 respectively and insert S64 and U64?

That seems the way to go, they must match and if STRING and BYTES are only found in protobuf, the number allocated to them does not matter

machinekoder, 2017-02-28:
Exactly. I think `STRING` and `BYTES` have been reserved for future HAL types.

In general it is a bad idea to change field IDs. However, in this case the types aren't in use yet.

ArcEye, 2017-02-28:
Are you doing this or do you want me to?

machinekoder, 2017-03-03:
@ArcEye Please go ahead.

PR #1159 (2017-03-03): Fix type mismatch in Machinetalk Protobuf and hal.h
enums for s64 and u64 were missing and allocated to other fields

Fixes #1155
closed 2017-03-03 by cdsteinkuehler