machinekit/machinekit #1161: xhc-hb04.tcl is broken again.

patrickLangan, 2017-03-06:
Upon running the sample configuration "xhc-hb04-layout2.ini" I get the following error:
`Unknown 'show' type 'pexists'`

Here is the full output:
```
machinekit@beaglebone:~$ machinekit ~/machinekit/configs/sim/axis/xhc-hb04/xhc-hb04-layout2.ini
MACHINEKIT - 0.1
Machine configuration directory is '/home/machinekit/machinekit/configs/sim/axis/xhc-hb04'
Machine configuration file is 'xhc-hb04-layout2.ini'
Starting Machinekit...
io started
halcmd loadusr io started
xhc-hb04: waiting for XHC-HB04 device
xhc-hb04: found XHC-HB04 device
Unknown 'show' type 'pexists'
    while executing
"hal show pexists xhc-hb04.button-goto-zero"
    ("eval" body line 1)
    invoked from within
"eval hal show $args"
    (procedure "show" line 1)
    invoked from within
"show pexists $name"
    (procedure "pin_exists" line 2)
    invoked from within
"pin_exists $fullbname"
    (procedure "connect_pins" line 24)
    invoked from within
"connect_pins    "
    (file "xhc-hb04.tcl" line 400)
    invoked from within
"source $filename"
    invoked from within
"set result [source $filename]"
    (file "/home/machinekit/machinekit/scripts/haltcl" line 55)
libgcc_s.so.1 must be installed for pthread_cancel to work
/home/machinekit/machinekit/scripts/linuxcnc: line 750:  4522 Aborted                 haltcl -i "$INIFILE" $CFGFILE
Shutting down and cleaning up Machinekit...
xhc-hb04: connection lost, cleaning up
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: warning [do_close] A cancellation for an in-flight transfer hasn't completed but closing the device handle
*** Error in `xhc-hb04': free(): corrupted unsorted chunks: 0x0002e750 ***
Cleanup done
Machinekit terminated with an error.  You can find more information in the log:
    /home/machinekit/linuxcnc_debug.txt
and
    /home/machinekit/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
machinekit@beaglebone:~$ 
```

It seems to me that this error is due to [the temporary fix](https://github.com/machinekit/machinekit/commit/0d3f6b4d3c5aa89f1b75203be1efd4417a92182e) from [the last time xhc-hb04 broke](https://github.com/machinekit/machinekit/issues/640) being undone by [this commit](https://github.com/machinekit/machinekit/commit/d2ae78f3f842704eec6d2338a28f37ebffe9d765).

patrickLangan, 2017-03-06:
By the way, I did make an attempt to fix this myself, with the following code modification to xhc-hb04.tcl:
```
 92  92   proc pin_exists {name} {
 93      -   set line [show pexists $name]
 94      -   if {"$line" != "Exists"} {
 95      -    puts stderr "FAILED: show pexists $name, returned $line"
     93  +   set line [show pin $name]
     94  +   if {[string first $name $line 0] < 0} {
     95  +    puts stderr "FAILED: show pin $name, returned $line"
 96  96       return 0 ; #fail
 97  97     }
 98      -  puts stderr "PASSED: show pexists $name, returned $line"
     98  +  #puts stderr "PASSED: show pin $name, returned $line"
 99  99     return 1; # pass
100 100   } ; #pin_exists
```

And while that did seem to help, I'm still getting this seemingly unrelated error message, also present in the original error output:
`libgcc_s.so.1 must be installed for pthread_cancel to work`

Full output:
```
machinekit@beaglebone:~$ machinekit ~/machinekit/configs/sim/axis/xhc-hb04/xhc-hb04-layout2.ini
MACHINEKIT - 0.1
Machine configuration directory is '/home/machinekit/machinekit/configs/sim/axis/xhc-hb04'
Machine configuration file is 'xhc-hb04-layout2.ini'
Starting Machinekit...
io started
halcmd loadusr io started
xhc-hb04: waiting for XHC-HB04 device
xhc-hb04: found XHC-HB04 device
xhc-hb04.tcl: using std_start_pause_button
libgcc_s.so.1 must be installed for pthread_cancel to work
/home/machinekit/machinekit/scripts/linuxcnc: line 750:  4777 Aborted                 haltcl -i "$INIFILE" $CFGFILE
Shutting down and cleaning up Machinekit...
xhc-hb04: connection lost, cleaning up
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: warning [do_close] A cancellation for an in-flight transfer hasn't completed but closing the device handle
*** Error in `xhc-hb04': free(): corrupted unsorted chunks: 0x0002eb58 ***
Cleanup done
Machinekit terminated with an error.  You can find more information in the log:
    /home/machinekit/linuxcnc_debug.txt
and
    /home/machinekit/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
machinekit@beaglebone:~$ 
```
I did verify that the crash occurs after xhc-hb04.tcl is finished running.  Perhaps there is more than one source of error?

ArcEye, 2017-03-06:
I clobbered my own fix, albeit it may have to be done another way now. :smile:
I'll look at in a while.

I think the errors from libusb after exit, always did occur, it seems to want to be shut down first.
My fix just replaced the archaic printout string comparison with another way to tell if a named pin existed

ArcEye, 2017-03-06:
Reinstated the fix and updated the search mechanism to use new functions.

Tested with the mm sim and works fine
```
MACHINEKIT - 0.1
Machine configuration directory is '/usr/src/machinekit/configs/sim/axis/xhc-hb04'
Machine configuration file is 'xhc-hb04-layout2_mm.ini'
Starting Machinekit...
io started
halcmd loadusr io started
xhc-hb04: waiting for XHC-HB04 device
xhc-hb04: found XHC-HB04 device
PASSED: show pexists xhc-hb04.button-goto-zero, returned Exists
PASSED: show pexists halui.mdi-command-00, returned Exists
PASSED: show pexists xhc-hb04.button-half, returned Exists
PASSED: show pexists halui.spindle.stop, returned Exists
PASSED: show pexists xhc-hb04.button-home, returned Exists
PASSED: show pexists halui.home-all, returned Exists
PASSED: show pexists xhc-hb04.button-macro-1, returned Exists
PASSED: show pexists halui.mdi-command-01, returned Exists
PASSED: show pexists xhc-hb04.button-macro-2, returned Exists
PASSED: show pexists halui.mdi-command-02, returned Exists
PASSED: show pexists xhc-hb04.button-macro-3, returned Exists
PASSED: show pexists halui.mdi-command-03, returned Exists
PASSED: show pexists xhc-hb04.button-macro-6, returned Exists
PASSED: show pexists halui.mdi-command-06, returned Exists
PASSED: show pexists xhc-hb04.button-macro-7, returned Exists
PASSED: show pexists halui.mdi-command-07, returned Exists
PASSED: show pexists xhc-hb04.button-probe-z, returned Exists
PASSED: show pexists motion.probe-input, returned Exists
PASSED: show pexists xhc-hb04.button-reset, returned Exists
PASSED: show pexists halui.estop.activate, returned Exists
PASSED: show pexists xhc-hb04.button-rewind, returned Exists
PASSED: show pexists halui.program.step, returned Exists
PASSED: show pexists xhc-hb04.button-safe-z, returned Exists
PASSED: show pexists halui.mdi-command-10, returned Exists
PASSED: show pexists xhc-hb04.button-spindle, returned Exists
PASSED: show pexists halui.spindle.start, returned Exists
xhc-hb04.tcl: using std_start_pause_button
PASSED: show pexists xhc-hb04.button-step, returned Exists
PASSED: show pexists xhc-hb04.stepsize-up, returned Exists
PASSED: show pexists xhc-hb04.button-stop, returned Exists
PASSED: show pexists halui.program.stop, returned Exists
task pid=15520
emcTaskInit: using builtin interpreter
```

The errors on closing are from libusb and the xhc-hb04 component respectively.
The libusb ones were always there, the free() error seems to come from a call to libusb also.
It all relates to how the component cleans up and it does not affect its operation
```
Shutting down and cleaning up Machinekit...
xhc-hb04: connection lost, cleaning up
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: warning [do_close] A cancellation for an in-flight transfer hasn't completed but closing the device handle
*** Error in `xhc-hb04': free(): corrupted unsorted chunks: 0x000000000222a920 ***
Cleanup done
```

ArcEye, 2017-03-06:
> libgcc_s.so.1 must be installed for pthread_cancel to work

This is a peculiar glibc related error, it does not occur on my system.

If it reoccurs, google that error, there is a lot about it out there.

closed 2017-03-06 by luminize

rubienr, 2017-08-29:
> libgcc_s.so.1 must be installed for pthread_cancel to work

I think this is related to ./scripts/haltcl:2
`load /home/machinekit/git/machinekit/tcl/hal.so`

Further investigation of `ldd -v hal.so` dependency tree shows that `libgcc_s.so` is needed with several symbol versions:

```
machinekit@beaglebone:~/git/machinekit/src$ ldd -v  /home/machinekit/git/machinekit/tcl/hal.so | grep "libgcc_s.so" | sort | uniq | grep -E "libgcc_s.so|$" --color
		libgcc_s.so.1 (GCC_3.0) => /lib/arm-linux-gnueabihf/libgcc_s.so.1
		libgcc_s.so.1 (GCC_3.3) => /lib/arm-linux-gnueabihf/libgcc_s.so.1
		libgcc_s.so.1 (GCC_3.5) => /lib/arm-linux-gnueabihf/libgcc_s.so.1
		libgcc_s.so.1 (GCC_4.3.0) => /lib/arm-linux-gnueabihf/libgcc_s.so.1
	/lib/arm-linux-gnueabihf/libgcc_s.so.1:
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb69c9000)
machinekit@beaglebone:~/git/machinekit/src$ ll /lib/arm-linux-gnueabihf/libgcc_s.so.1
```
but i see the **gcc 3.5** is missing in the `libgcc_s.so.1` version table:  
```
machinekit@beaglebone:~/git/machinekit/src$ readelf -V /lib/arm-linux-gnueabihf/libgcc_s.so.1 | grep  -A100 "Version definition section" 
Version definition section '.gnu.version_d' contains 13 entries:
  Addr: 0x000000000000cb34  Offset: 0x00cb34  Link: 4 (.dynstr)  000000: Rev: 1  Flags: BASE   Index: 1  Cnt: 1  Name: libgcc_s.so.1
  0x001c: Rev: 1  Flags: none  Index: 2  Cnt: 1  Name: GLIBC_2.0
  0x0038: Rev: 1  Flags: none  Index: 3  Cnt: 2  Name: GCC_3.0
  0x0054: Parent 1: GLIBC_2.0
  0x005c: Rev: 1  Flags: none  Index: 4  Cnt: 2  Name: GCC_3.3
  0x0078: Parent 1: GCC_3.0
  0x0080: Rev: 1  Flags: none  Index: 5  Cnt: 2  Name: GCC_3.3.1
  0x009c: Parent 1: GCC_3.3
  0x00a4: Rev: 1  Flags: none  Index: 6  Cnt: 2  Name: GCC_3.3.4
  0x00c0: Parent 1: GCC_3.3.1
  0x00c8: Rev: 1  Flags: none  Index: 7  Cnt: 2  Name: GCC_3.4
  0x00e4: Parent 1: GCC_3.3.4
  0x00ec: Rev: 1  Flags: none  Index: 8  Cnt: 2  Name: GCC_3.4.2
  0x0108: Parent 1: GCC_3.4
  0x0110: Rev: 1  Flags: none  Index: 9  Cnt: 2  Name: GCC_4.0.0
  0x012c: Parent 1: GCC_3.4.2
  0x0134: Rev: 1  Flags: none  Index: 10  Cnt: 2  Name: GCC_4.2.0
  0x0150: Parent 1: GCC_4.0.0
  0x0158: Rev: 1  Flags: none  Index: 11  Cnt: 2  Name: GCC_4.3.0
  0x0174: Parent 1: GCC_4.2.0
  0x017c: Rev: 1  Flags: none  Index: 12  Cnt: 2  Name: GCC_4.7.0
  0x0198: Parent 1: GCC_4.3.0
  0x01a0: Rev: 1  Flags: none  Index: 13  Cnt: 1  Name: GCC_3.5

Version needs section '.gnu.version_r' contains 1 entries:
 Addr: 0x000000000000ccf0  Offset: 0x00ccf0  Link: 4 (.dynstr)
  000000: Version: 1  File: libc.so.6  Cnt: 1
  0x0010:   Name: GLIBC_2.4  Flags: none  Version: 14
```

PS: It is not an error from libusb-1.0.

PR #1163 (2017-03-06): Reinstate workaround to keep xhc-hb04 pendants working
Inadvertently removed during multicore merge

This workaround is a temporary measure to keep the xhc-hb04 pendant working.
xhc-hb04.tcl uses command line halcmd outputs against tests which are fixed
in expectation of a particular return. (ie from linuxcnc circa 2014)

Changes to halcmd print routines broke xhc-hb04.

This will tell the tcl code if a pin exists, removing the need for the
previous convoluted parsing of the stdout output.

The actual search routine has been changed from a list walking legacy one,
to use the `halg_foreach()` i [... cut]