machinekit/machinekit #1182: pru_generic.bin not deployed for rt-preempt FLAVOR

machinekoder, 2017-03-22:
It looks like the `pru_generic.bin` is not deployed with the packages for Debian Jessie for the rt-preempt RT flavor. Does the PRU driver currently work with rt-preempt on the BBB?

cdsteinkuehler, 2017-03-22:
On 3/22/2017 4:39 AM, Alexander Rössler wrote:
> It looks like the `pru_generic.bin` is not deployed with the packages
> for Debian Jessie for the rt-preempt RT flavor. Does the PRU driver
> currently work with rt-preempt on the BBB?

The PRU driver doesn't care what RTOS is running, which is the whole
point of the RTAPI.

IIRC, the PRU bin file only appearing in the Xenomai tree was due to
some of the gyrations M. Haberler and J. Morris went through to make the
RTAPI layer work with arbitrary real-time flavors at run-time.  I am
sure there was a reason at one point that the unique realtime flavor
directories were required, but now they all contain pretty much bit
identical versions of everything but the top level RTAPI code.

IMHO, the posix/xenomai/preempt directories ought to go away and there
should just be one directory.

-- 
Charles Steinkuehler
charles@steinkuehler.net

machinekoder, 2017-03-22:
@cdsteinkuehler I compiled hal_pru_generic for RT on the BBB. However, I get the following error when trying to load the component:
```
Mar 22 17:10:37 beaglebone rtapi:0: rtapi_app_main(hal_pru_generic): -1 Operation not permitted
```
The file exists:
```
machinekit@beaglebone:~/repos/UNI-PRINT-3D$ file /home/machinekit/machinekit/rtlib/rt-preempt/pru_generic.bin
/home/machinekit/machinekit/rtlib/rt-preempt/pru_generic.bin: 8086 relocatable (Microsoft)
```

Do you have an idea what could be the problem?

cdsteinkuehler, 2017-03-22:
On 3/22/2017 12:13 PM, Alexander Rössler wrote:
> @cdsteinkuehler I compiled hal_pru_generic for RT on the BBB. However, I get the following error when trying to load the component:
> ```
> Mar 22 17:10:37 beaglebone rtapi:0: rtapi_app_main(hal_pru_generic): -1 Operation not permitted
> ```

It's hard to be sure with just the error, but it's likely due to
permissions (did you "sudo make setuid"?), not having the PRU overlay
loaded, or maybe using the newer 4.x kernel (which moves around lots of
stuff in sysfs) or a kernel without the required UIO pru support.

Note that the PRU binary is not even remotely dependent on the rtos
"flavor", so you can use the pru_generic.bin from the xenomai/ directory
on posix or preempt-rt without issue.

-- 
Charles Steinkuehler
charles@steinkuehler.net

machinekoder, 2017-03-23:
I'm using the 4.4 LTS TI kernel with RT-PREEMPT so the sysfs changes might be the problems.

Can you give me some hints what might need modifications?

luminize, 2017-07-26:
closed in favour of https://github.com/machinekit/machinekit/issues/1238

closed 2017-07-26 by luminize