machinekit/machinekit #1187: Haltalk rcomp bin times out

machinekoder, 2017-03-25:
I figured out that the problem of Machinetalk clients timing out is a result of the bind command taking several seconds to execute. https://github.com/machinekit/machinekit/blob/master/src/machinetalk/haltalk/haltalk_command.cc#L297

A closer dive into the command gave me the result that `validate_component` and `halpr_describe_component` are the two bottlenecks of the functions. I dug even deeper into the `validate_component` and found that `halpr_pin_count` is the bottleneck of this function: https://github.com/machinekit/machinekit/blob/master/src/machinetalk/haltalk/haltalk_command.cc#L97

Both commands call the `halg_foreach_from` function in `hal_object.c` https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal_object.c#L257
The problem seems to occur when the function is called with `args` containing an `owning_comp` value. This results in the loop calling itself through `halpr_find_owning_comp` https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal_object.c#L300

I still could not figure out what is the actual problem here.

machinekoder, 2017-03-25:
I measured around 5 million calls to `halg_foreach_from` for one `halpr_pin_count`

machinekoder, 2017-03-25:
In my example configuration the complete HAL config objects had a size of approximately 2000 items. This means the `halpr_find_owning_comp` increases the complexity of the function call by a factor of ~2500.

It also explains why I could not reproduce this bug on my desktop machine. The desktop machine has enough horsepower to easily run the 5mio commands in almost no time.

ArcEye, 2017-03-27:
I would be interested if you were able to quantify what effect this reordering has on the problem.
 #1191

It would seem obvious, where the majority, if not all pins are owned by instances, to check those first.

einstine909, 2017-03-27:
I tested the builds `0.1.1490548480-1mk.travis.master.git29f680b4~1jessie` and `0.1.1490548454-1mk.travis.master.git252c3e30~1jessie` (should include #1188 ) and I am still seeing haltalk timeouts on a BBB. Anything I can do to provide more information?

machinekoder, 2017-03-27:
@einstine909 It looks like something different is happening in your setup since you experience the problems also with a non-Haltalk configuration.

machinekoder, 2017-03-29:
The rcomp issue has been fixed. @einstine909 Lets transfer this problem over to the QtQuickVcp issue tracker

closed 2017-03-29 by machinekoder