machinekit/machinekit #1193: can't compile with modbus >3.10

Alessandro-Barbieri, 2017-03-27:
```
Compiling hal/user_comps/mb2hal/mb2hal.c
Compiling hal/user_comps/mb2hal/mb2hal_init.c
Compiling hal/user_comps/mb2hal/mb2hal_modbus.c
hal/user_comps/mb2hal/mb2hal.c: In function ‘get_tx_connection’:
hal/user_comps/mb2hal/mb2hal.c:362:55: warning: passing argument 2 of ‘modbus_set_response_timeout’ makes integer from pointer without a cast [-Wint-conversion]
     modbus_set_response_timeout(this_mb_link->modbus, &timeout);
                                                       ^
In file included from hal/user_comps/mb2hal/mb2hal.h:18:0,
                 from hal/user_comps/mb2hal/mb2hal.c:27:
/usr/include/modbus/modbus.h:185:16: note: expected ‘uint32_t {aka unsigned int}’ but argument is of type ‘struct timeval *’
 MODBUS_API int modbus_set_response_timeout(modbus_t *ctx, uint32_t to_sec, uint32_t to_usec);
                ^
hal/user_comps/mb2hal/mb2hal.c:362:5: error: too few arguments to function ‘modbus_set_response_timeout’
     modbus_set_response_timeout(this_mb_link->modbus, &timeout);
     ^
In file included from hal/user_comps/mb2hal/mb2hal.h:18:0,
                 from hal/user_comps/mb2hal/mb2hal.c:27:
/usr/include/modbus/modbus.h:185:16: note: declared here
 MODBUS_API int modbus_set_response_timeout(modbus_t *ctx, uint32_t to_sec, uint32_t to_usec);
                ^
hal/user_comps/mb2hal/mb2hal.c:369:51: warning: passing argument 2 of ‘modbus_set_byte_timeout’ makes integer from pointer without a cast [-Wint-conversion]
     modbus_set_byte_timeout(this_mb_link->modbus, &timeout);
                                                   ^
In file included from hal/user_comps/mb2hal/mb2hal.h:18:0,
                 from hal/user_comps/mb2hal/mb2hal.c:27:
/usr/include/modbus/modbus.h:188:16: note: expected ‘uint32_t {aka unsigned int}’ but argument is of type ‘struct timeval *’
 MODBUS_API int modbus_set_byte_timeout(modbus_t *ctx, uint32_t to_sec, uint32_t to_usec);
                ^
hal/user_comps/mb2hal/mb2hal.c:369:5: error: too few arguments to function ‘modbus_set_byte_timeout’
     modbus_set_byte_timeout(this_mb_link->modbus, &timeout);
     ^
In file included from hal/user_comps/mb2hal/mb2hal.h:18:0,
                 from hal/user_comps/mb2hal/mb2hal.c:27:
/usr/include/modbus/modbus.h:188:16: note: declared here
 MODBUS_API int modbus_set_byte_timeout(modbus_t *ctx, uint32_t to_sec, uint32_t to_usec);
                ^
make: *** [Makefile:422: objects/hal/user_comps/mb2hal/mb2hal.o] Error 1

```

See also https://github.com/LinuxCNC/linuxcnc/issues/105

ArcEye, 2017-03-28:
Please say what distro are you using and all the other information that puts this in some sort of context.

Even Stretch uses libmodbus-3.0.6-2, so it is not likely we will be rewriting to cater backwardly incompatible API changes in libmodbus any time soon.

Alessandro-Barbieri, 2017-03-28:
I'm using gentoo with libmodbus-3.1.2 https://packages.gentoo.org/packages/dev-libs/libmodbus

In linuxcnc the same issue seems too be fixed with https://github.com/LinuxCNC/linuxcnc/pull/106

Maybe Debian will upgrade the library https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=842534
they don't see the new version because the watch file fails

ArcEye, 2017-03-28:
> In linuxcnc the same issue seems too be fixed with LinuxCNC#106

Don't think so, last entry Dec 2016  "Compiling with libmodbus version 3.1.4 fails."

>  Maybe Debian will upgrade the library  https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=842534

Or maybe they won't.  That is not a bug report it clearly shows as `Severity: wishlist`
Debian prefer stability and packages that work, to chasing version numbers in some insane belief that 'higher numbers are better'.

If you can make an argument as to why libmodbus > 3.1.0 is required for anything other than the convenience of people who choose not to run the supported distros, you may get someone interested.

closed 2017-07-07 by ArcEye

Alessandro-Barbieri, 2017-07-07:
?

grbd, 2018-01-06:
For anyone else that comes across this
I've got some unofficial ebuilds setup here that can be installed via layman
https://github.com/CNC-Nidehog/GentooCNC
the machinekit ebuild is set to use the older libmodbus which seems to work okay as far as compiling machinekit