machinekit/machinekit #1197: Haltalk throws "Assertion `self' failed." after: MT_HALRCOMP_BIND_CONFIRM

zhivko, 2017-04-02:
Could this be related in too long haltalk message maybe?

I am using mk from current master branch of https://github.com/machinekit/machinekit.

I am sending MT_HALRCOMP_BIND message from my mk JAVA client (let me emphasize other messages work without problem without assert problem).

17/04/02 23:02:31 INFO commands.MachineKitStart: D: 17-04-02 21:02:31 [017] 56E58679-B43F81A0
17/04/02 23:02:31 INFO commands.MachineKitStart: D: 17-04-02 21:02:31 [1121] 088002A206DB060A056D79696E69820116080112106D79696E692E7468632D656E6162...
17/04/02 23:02:31 INFO commands.MachineKitStart: handle_command_input: req=type: MT_HALRCOMP_BIND

actual load is:
```
088002A206DB060A056D79696E69820116080112106D79696E692E7468632D656E61626C652020820118080212126D79696E692E61637475616C2D766F6C7473201082011B080212156D79696E692E766F6C74732D7265717565737465642020820116080112106D79696E692E76656C2D73746174757320108201130802120D6D79696E692E76656C2D746F6C2020820118080212126D79696E692E7363616C652D6F666673657420208201150802120F6D79696E692E76656C2D7363616C652020820117080212116D79696E692E766F6C746167652D746F6C202082011A080212146D79696E692E636F7272656374696F6E2D76656C2020820117080112116D79696E692E746F7263682D70726F626520108201150802120F6D79696E692E7468632D7A2D706F732010820118080212126D79696E692E6F66667365742D76616C756520108201140802120E6D79696E692E6D617876656C5F3320208201140802120E6D79696E692E6D61786163635F33202082011C080212166D79696E692E7374657067656E5F6D617876656C5F33202082011C080212166D79696E692E7374657067656E5F6D61786163635F3320208201140802120E6D79696E692E6D617876656C5F3420208201140802120E6D79696E692E6D61786163635F34202082011C080212166D79696E692E7374657067656E5F6D617876656C5F34202082011C080212166D79696E692E7374657067656E5F6D61786163635F3420208201140802120E6D79696E692E6D617876656C5F3020208201140802120E6D79696E692E6D61786163635F30202082011C080212166D79696E692E7374657067656E5F6D617876656C5F30202082011C080212166D79696E692E7374657067656E5F6D61786163635F3020208201140802120E6D79696E692E6D617876656C5F3120208201140802120E6D79696E692E6D61786163635F31202082011C080212166D79696E692E7374657067656E5F6D617876656C5F31202082011C080212166D79696E692E7374657067656E5F6D61786163635F3120208201140802120E6D79696E692E6D617876656C5F3220208201140802120E6D79696E692E6D61786163635F32202082011C080212166D79696E692E7374657067656E5F6D617876656C5F32202082011C080212166D79696E692E7374657067656E5F6D61786163635F322020A206FB010A086D796D6F74696F6E8201110802120B6D796D6F74696F6E2E767820108201120802120C6D796D6F74696F6E2E64767820108201110802120B6D796D6F74696F6E2E767920108201110802120B6D796D6F74696F6E2E767A20108201120802120C6D796D6F74696F6E2E64767A201082011D080212176D796D6F74696F6E2E63757272656E742D72616469757320108201100802120A6D796D6F74696F6E2E76201082011B080212156D796D6F74696F6E2E6C6173657248656967687431201082011B080312156D796D6F74696F6E2E70726F6772616D2D6C696E652010820119080112136D796D6F74696F6E2E7370696E646C652D6F6E2010
```

And then haltalk replies with:

```
send_pbcontainer: type: MT_HALRCOMP_BIND_CONFIRM
uuid: "\202\367\352\274\027\347\021\347\241\334TJ\026\305\320*"
comp {
  name: "myini"
  comp_id: 474
  type: 3
  state: 2
  last_update: 0
  last_bound: 0
  last_unbound: 0
  pid: 8963
  userarg1: 100
  userarg2: 0
  pin {
    type: HAL_FLOAT
    name: "myini.actual-volts"
    handle: 497
    dir: HAL_IN
    halfloat: 0
    linked: true
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.correction-vel"
    handle: 507
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.maxacc_0"
    handle: 477
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.maxacc_1"
    handle: 479
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.maxacc_2"
    handle: 481
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.maxacc_3"
    handle: 483
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.maxacc_4"
    handle: 485
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.maxvel_0"
    handle: 476
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.maxvel_1"
    handle: 478
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.maxvel_2"
    handle: 480
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.maxvel_3"
    handle: 482
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.maxvel_4"
    handle: 484
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.offset-value"
    handle: 501
    dir: HAL_IN
    halfloat: 0
    linked: true
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.scale-offset"
    handle: 504
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.stepgen_maxacc_0"
    handle: 487
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.stepgen_maxacc_1"
    handle: 489
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.stepgen_maxacc_2"
    handle: 491
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.stepgen_maxacc_3"
    handle: 493
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.stepgen_maxacc_4"
    handle: 495
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.stepgen_maxvel_0"
    handle: 486
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.stepgen_maxvel_1"
    handle: 488
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.stepgen_maxvel_2"
    handle: 490
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.stepgen_maxvel_3"
    handle: 492
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.stepgen_maxvel_4"
    handle: 494
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_BIT
    name: "myini.thc-enable"
    handle: 496
    dir: HAL_OUT
    halbit: false
    linked: true
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.thc-z-pos"
    handle: 498
    dir: HAL_IN
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_BIT
    name: "myini.torch-probe"
    handle: 502
    dir: HAL_IN
    halbit: true
    linked: true
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.vel-scale"
    handle: 505
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_BIT
    name: "myini.vel-status"
    handle: 500
    dir: HAL_IN
    halbit: false
    linked: true
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.vel-tol"
    handle: 503
    dir: HAL_OUT
    halfloat: 0
    linked: true
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.voltage-tol"
    handle: 506
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.volts-requested"
    handle: 499
    dir: HAL_OUT
    halfloat: 0
    linked: true
    epsilon: 1e-05
    flags: 0
  }
}

haltalk: src/zframe.c:186: zframe_data: Assertion `self' failed.
```

How can I further troubleshoot this? I had **Assertion 'self'** problem before with MT_PING type of package but now I have it with MT_HALRCOMP_BIND_CONFIRM coming from haltalk back to my client.

How can I troubleshoot this? Is it length of package that is problematic? This is just guess...

zhivko, 2017-04-03:
I have python client that reproduces this behaviour.

```
machinekit@machinekit:~/git/machinekit/myini/testPing$ python mkPythonClient.py 
send ping 1
send ping 2
waiting for command uri
(u'HAL Rcommand service on machinekit.local pid 527._machinekit._tcp.local.', ServiceInfo(type='_machinekit._tcp.local.', name=u'HAL Rcommand service on machinekit.local pid 527._machinekit._tcp.local.', address='\xc0\xa8+I', port=6202, weight=0, priority=0, server=u'machinekit.local.', properties={'dsn': 'tcp://machinekit.local:6202', 'instance': '41cefcec-1850-11e7-860b-080027165e4c', 'uuid': 'a42c8c6b-4025-4f83-ba28-dad21114744a', 'service': 'halrcmd'}))
---
(u'Previewstatus service on machinekit.local pid 551._machinekit._tcp.local.', ServiceInfo(type='_machinekit._tcp.local.', name=u'Previewstatus service on machinekit.local pid 551._machinekit._tcp.local.', address='\xc0\xa8+I', port=49154, weight=0, priority=0, server=u'machinekit.local.', properties={'dsn': 'tcp://machinekit.local:49154', 'instance': '4272355b-1850-11e7-860b-080027165e4c', 'uuid': 'a42c8c6b-4025-4f83-ba28-dad21114744a', 'service': 'previewstatus'}))
---
(u'Preview service on machinekit.local pid 551._machinekit._tcp.local.', ServiceInfo(type='_machinekit._tcp.local.', name=u'Preview service on machinekit.local pid 551._machinekit._tcp.local.', address='\xc0\xa8+I', port=49153, weight=0, priority=0, server=u'machinekit.local.', properties={'dsn': 'tcp://machinekit.local:49153', 'instance': '4272355a-1850-11e7-860b-080027165e4c', 'uuid': 'a42c8c6b-4025-4f83-ba28-dad21114744a', 'service': 'preview'}))
---
(u'Command service on machinekit.local pid 551._machinekit._tcp.local.', ServiceInfo(type='_machinekit._tcp.local.', name=u'Command service on machinekit.local pid 551._machinekit._tcp.local.', address='\xc0\xa8+I', port=54209, weight=0, priority=0, server=u'machinekit.local.', properties={'dsn': 'tcp://machinekit.local:54209', 'instance': '42723559-1850-11e7-860b-080027165e4c', 'uuid': 'a42c8c6b-4025-4f83-ba28-dad21114744a', 'service': 'command'}))
---
(u'Error service on machinekit.local pid 551._machinekit._tcp.local.', ServiceInfo(type='_machinekit._tcp.local.', name=u'Error service on machinekit.local pid 551._machinekit._tcp.local.', address='\xc0\xa8+I', port=55235, weight=0, priority=0, server=u'machinekit.local.', properties={'dsn': 'tcp://machinekit.local:55235', 'instance': '42723558-1850-11e7-860b-080027165e4c', 'uuid': 'a42c8c6b-4025-4f83-ba28-dad21114744a', 'service': 'error'}))
---
(u'Status service on machinekit.local pid 551._machinekit._tcp.local.', ServiceInfo(type='_machinekit._tcp.local.', name=u'Status service on machinekit.local pid 551._machinekit._tcp.local.', address='\xc0\xa8+I', port=56589, weight=0, priority=0, server=u'machinekit.local.', properties={'dsn': 'tcp://machinekit.local:56589', 'instance': '42723557-1850-11e7-860b-080027165e4c', 'uuid': 'a42c8c6b-4025-4f83-ba28-dad21114744a', 'service': 'status'}))
---
(u'HAL Rcomp service on machinekit.local pid 527._machinekit._tcp.local.', ServiceInfo(type='_machinekit._tcp.local.', name=u'HAL Rcomp service on machinekit.local pid 527._machinekit._tcp.local.', address='\xc0\xa8+I', port=6201, weight=0, priority=0, server=u'machinekit.local.', properties={'dsn': 'tcp://machinekit.local:6201', 'instance': '41cefcec-1850-11e7-860b-080027165e4c', 'uuid': 'a42c8c6b-4025-4f83-ba28-dad21114744a', 'service': 'halrcomp'}))
---
(u'File service on machinekit.local pid 551._machinekit._tcp.local.', ServiceInfo(type='_machinekit._tcp.local.', name=u'File service on machinekit.local pid 551._machinekit._tcp.local.', address='\xc0\xa8+I', port=56432, weight=0, priority=0, server=u'machinekit.local.', properties={'dsn': 'ftp://machinekit.local:56432', 'instance': '42723556-1850-11e7-860b-080027165e4c', 'uuid': 'a42c8c6b-4025-4f83-ba28-dad21114744a', 'service': 'file'}))
---
(u'Log service on machinekit.local pid 490._machinekit._tcp.local.', ServiceInfo(type='_machinekit._tcp.local.', name=u'Log service on machinekit.local pid 490._machinekit._tcp.local.', address='\xc0\xa8+I', port=49152, weight=0, priority=0, server=u'machinekit.local.', properties={'dsn': 'tcp://machinekit.local:49152', 'instance': '414d5250-1850-11e7-860b-080027165e4c', 'uuid': 'a42c8c6b-4025-4f83-ba28-dad21114744a', 'service': 'log'}))
---
(u'HAL Group service on machinekit.local pid 527._machinekit._tcp.local.', ServiceInfo(type='_machinekit._tcp.local.', name=u'HAL Group service on machinekit.local pid 527._machinekit._tcp.local.', address='\xc0\xa8+I', port=6200, weight=0, priority=0, server=u'machinekit.local.', properties={'dsn': 'tcp://machinekit.local:6200', 'instance': '41cefcec-1850-11e7-860b-080027165e4c', 'uuid': 'a42c8c6b-4025-4f83-ba28-dad21114744a', 'service': 'halgroup'}))
---
Found command uri at: tcp://machinekit.local.:6202
send ping
connecting
connected
receive
Service HAL Rcommand service on machinekit.local pid 527._machinekit._tcp.local. removed
Service HAL Rcomp service on machinekit.local pid 527._machinekit._tcp.local. removed
Service HAL Group service on machinekit.local pid 527._machinekit._tcp.local. removed
```

It is seen from upper output that assert then crashes haltalk.

output on server is:
```
machinekit@machinekit:~/git/machinekit$ which machinekit
/home/machinekit/git/machinekitOff/bin/machinekit
machinekit@machinekit:~/git/machinekit$ 
machinekit@machinekit:~/git/machinekit$ 
machinekit@machinekit:~/git/machinekit$ 
[... cut ...]
CRAMPS.hal:203: Pin 'axis.3.amp-enable-out' linked to signal 'emcmot.03.enable'
[... 659 lines cut ...]
  }
  pin {
    type: HAL_FLOAT
    name: "myini.voltage-tol"
    handle: 506
    dir: HAL_OUT
    halfloat: 0
    linked: false
    epsilon: 1e-05
    flags: 0
  }
  pin {
    type: HAL_FLOAT
    name: "myini.volts-requested"
    handle: 499
    dir: HAL_OUT
    halfloat: 0
    linked: true
    epsilon: 1e-05
    flags: 0
  }
}

haltalk: zframe.c:171: zframe_size: Assertion `self' failed.
```

zhivko, 2017-04-03:
find out that I am sending 2 components in protobuf message.
Now i change to send one by one and it turns out that pb for just second comp throws following error:
```
send_pbcontainer: type: MT_HALRCOMP_BIND_REJECT
note: "pin count mismatch:pb comp=10 hal comp=11"
```

seems I need to recheck in count etc... will do that...
however throwing assert and crashing haltalk still doesn't seem to be most user friendly.

zhivko, 2017-04-03:
it works now when second comp has same pin numbers as in hal... however user would expect MT_HALRCOMP_BIND_REJECT.

ArcEye, 2017-04-03:
Well done for tracking it down.

The problem looks to start here
https://github.com/machinekit/machinekit/blob/master/src/machinetalk/haltalk/haltalk_command.cc#L390

It sends a message if an error occurs.
If the calling side does not do anything about that message, it is liable to iterate to a NULL pointer and trigger the assert() in zmq.
```
size_t zframe_size (zframe_t *self)
{
    assert (self);
    assert (zframe_is (self));

    return zmq_msg_size (&self->zmsg);
}
```
I don't like assert()s, they are a bit lazy.  Just zmq saying "What are you doing passing a NULL pointer, take that for your cheek"

Since the assert() is part of the zmq library and cannot be changed, the code using it needs to check
what it passes to it.

So whenever `zframe_size(zframe_t *p)` is called, need to make sure that `p` is valid.
This will probably have to be done by specifically checking if it is NULL, because even the function to check if a pointer is a zframe throws an assert error if NULL. :smile: 
```
bool zframe_is (void *self)
{
    assert (self);
    return ((zframe_t *) self)->tag == ZFRAME_TAG;
}
```

If you fancy playing with that, find exactly where the triggering call is made when the pin lists don't match etc., you might improve things for yourself and others.

ArcEye, 2017-04-04:
@zhivko I have hopefully fixed them, in so far as allowing them to pass a NULL pointer goes anyway.

I would be grateful if you can test with your config that produced the error and report back when the changes are in the repo.

The knock-ons from this and the exact behaviour when it does hit a NULL pointer, may still not be ideal
in terms of program execution, but at least will not throw the previous fatal assert() error.

ArcEye, 2017-04-04:
@zhivko  has just been merged,

I'll wait to hear from you in due course.

zhivko, 2017-04-04:
@ArcEye I tried mk from your repo and I still get assert.
I was hoping that if I get:
```
MT_HALRCOMP_BIND_REJECT
note: "pin count mismatch:pb comp=10 hal comp=11"
```
when sending single component, but same (I think) should be in case sending two components, one with appropriate pin count and another with pin count difference.

Output from mk is same - for first comp I get MT_HALRCOMP_BIND_CONFIRM, and then assert.

ArcEye, 2017-04-04:
Is the assert() error coming from `zframe_size` or another function?

zhivko, 2017-04-04:
I could not tell where the assert is comming from from mk console output.

machinekit@machinekit:~/git/machinekitArcEye$ git config --get
remote.origin.url
https://github.com/ArcEye/machinekit.git

Seems you haven't change from my last try, because pull hasn't show
anything changed...

regards

ArcEye, 2017-04-04:
> Seems you haven't change from my last try, because pull hasn't show anything changed...

It was in a branch called `test-zmq-pointers` which is now merged into machinekit/master.

Just checking you were testing the right branch.

zhivko, 2017-04-04:
Oh yes! I should checkout your new branch **test-zmq-pointers** - in that branch it works as it should :)

I get:
```
send_pbcontainer: type: MT_PING
note: "pin count mismatch:pb comp=10 hal comp=11"
uuid: "V\366\257\262\031?\021\347\222\202\010\000\'\026^L"
```

ArcEye, 2017-04-04:
Good, it looks as though it is stepping over the problem and returning, without halting the execution.

Thanks for testing, I'll close this again.

closed 2017-04-04 by ArcEye