machinekit/machinekit #1199: RT-PREEMPT for armhf broken build in travis

dkhughes, 2017-04-07:
Hi Everyone,

I've been away for a bit, but I'm starting to take a look at some of the outstandings I have left for the arm on zynq stuff and my bucket list.

It looks like there is a problem with fakeroot in the travis build of the machinekit package. This makes installing the machinekit package fail on armhf since it's depending on a different version of the machinekit-rt-preempt package.

Is this a known problem? Anything I can do to help? I see a lot of changes have been happening in the last couple of months so I'm a little lost, but I will try to help if possible.

ArcEye, 2017-04-07:
There has just been a commit regards this
https://github.com/machinekit/machinekit/commit/54105677e8ab8f8680f0f4286c7b56b71b857fd2

Does this cause the problem ( or fix it depending upon how current your fork is )

dkhughes, 2017-04-07:
I was trying to install the package from the repo. The CI build is failing on the jessie target here:
[https://travis-ci.org/machinekit/machinekit/jobs/218423456#L4164](https://travis-ci.org/machinekit/machinekit/jobs/218423456#L4164).

ArcEye, 2017-04-07:
My PR was the second to be merged in quick succession.
It actually fails here https://travis-ci.org/machinekit/machinekit/builds/218419684 which is the hal_pru commit and then on every subsequent build.

Looks like the hal_pru for rt-preempt needs reverting until it builds properly

ArcEye, 2017-04-07:
It is a small code change https://github.com/machinekit/machinekit/commit/54105677e8ab8f8680f0f4286c7b56b71b857fd2

If you do a reversal PR, I will merge it once the tests have run and we can then be sure that everything works, if the build succeeds thereafter.

dkhughes, 2017-04-07:
Okay, submitted. I just changed it back to the former version. Unfortunately I don't know enough about the PRU code to do more.

ArcEye, 2017-04-07:
Merged, lets see in 90 mins or so if all the packages built OK :smile:

ArcEye, 2017-04-07:
That has worked, all packages built OK.

Hopefully you will be able to proceed now.

cdsteinkuehler, 2017-04-07:
On 4/7/2017 9:46 AM, dkhughes wrote:
> Okay, submitted. I just changed it back to the former version. Unfortunately I 
> don't know enough about the PRU code to do more.

All the PRU code is identical regardless of flavor.  You can use the
ARM binary HAL component from Xenomai with any of the other real-time
flavors and it will work fine (go ahead and binary diff them).

The PRU object code cares even less about what real-time flavor is
running.  When I run with preempt-rt, I just use the PRU bin file from
the Xenomai directory.

Ultimately, IMHO the replicated HAL object file directories for each
RT flavor should go away and there would just be one directory.  We
really only need one file (with the RTAPI stuff) to be different for
the various different RT flavors, everything else is already binary
identical across the flavor directories.

-- 
Charles Steinkuehler
charles@steinkuehler.net

dkhughes, 2017-11-02:
Closed - this has been fixed for a while now.

closed 2017-11-02 by dkhughes