machinekit/machinekit #1203: About armel

xielangtao, 2017-04-19:
1> sudo sh -c "echo 'deb http://deb.machinekit.io/debian wheezy main'>/etc/apt/sources.list.d/machinekit.list"

2> apt-get update
Hit http://mirrors.ustc.edu.cn wheezy Release.gpg
Hit http://mirrors.ustc.edu.cn wheezy Release                                  
Hit http://mirrors.ustc.edu.cn wheezy/main Sources                             
Get:1 http://ftp.us.debian.org wheezy-backports Release.gpg [1554 B]           
Hit http://mirrors.ustc.edu.cn wheezy/contrib Sources                          
Hit http://mirrors.ustc.edu.cn wheezy/non-free Sources                         
Hit http://mirrors.ustc.edu.cn wheezy/main armel Packages                      
Hit http://mirrors.ustc.edu.cn wheezy/contrib armel Packages                   
Get:2 http://ftp.us.debian.org wheezy-backports Release [164 kB]               
Hit http://mirrors.ustc.edu.cn wheezy/non-free armel Packages                  
Hit http://deb.machinekit.io wheezy Release.gpg                                
Hit http://mirrors.ustc.edu.cn wheezy/contrib Translation-en            
Hit http://mirrors.ustc.edu.cn wheezy/main Translation-en               
Hit http://mirrors.ustc.edu.cn wheezy/non-free Translation-en                  
Hit http://deb.machinekit.io wheezy Release                                    
Get:3 http://ftp.us.debian.org wheezy-backports/main armel Packages [581 kB]
Get:4 http://ftp.us.debian.org wheezy-backports/main Translation-en [378 kB]
Fetched 1125 kB in 30s (36.9 kB/s)                                             
W: Failed to fetch http://deb.machinekit.io/debian/dists/wheezy/Release  Unable to find expected entry 'main/binary-armel/Packages' in Release file (Wrong sources.list entry or malformed file)

E: Some index files failed to download. They have been ignored, or old ones used instead.

help

xielangtao, 2017-04-19:
I want download this file:

sudo apt-get install libczmq-dev python-zmq libjansson-dev pkg-config libwebsockets-dev libxenomai-dev python-pyftpdlib cython bwidget lsb-release python-gtk2

but apt-get update was failed.

ArcEye, 2017-04-19:
The apt command fails because there are no 'arml' repositories, just 'armhf'.

You can download the packages from http://deb.machinekit.io and install manually, but I don't know enough about differences to know if it will work.

closed 2017-04-19 by ArcEye