machinekit/machinekit #1207: compile warnings line numbers not always accurate.

luminize, 2017-05-15:
I've had in an icomp v2 the following declared:
```
pin_ptr in u32 index_cmd = 0;
variable hal_s32_t index_curr = 0;
```
then in the function I compared these 2, "if" starts at line 90:
```
                    if (index_curr == index_cmd) {
                        // position has been reached
                        // stop and go to next state
                        sb(drive_on, false);
                        sb(drive_brake, true);
                        indextable_active = false;
                    }
```
when compiling I got the error
```
indextablev2.icomp:93:36: error: invalid operands to binary == (have ‘hal_u32_t’ and ‘s32_pin_ptr’)
```
likewise I get with this code starting at line 65:
```
                index_target = gs(index_cmd)
                pos_difference = index_target - index_curr;
                if (pos_difference == 0) {
                    // do nothing, return to previous state
                    state = 1;
```
the following message
```
indextablev2.icomp:69:17: error: expected ‘;’ before ‘pos_difference’
```

So I guess there's a 3 line discrepancy between the info the terminal shows and the real line. No showstopper, but if you don't know it can be hard to find a fault.

ArcEye, 2017-05-15:
There has always been a problem with line numbers when compiling components.

Because comp and instcomp generate extra code and then add the component code (and any blank lines) verbatim from immediately after the `;;` delimiter, it can sometimes be difficult to align the reported error with the original component.

Sometimes you need to take a 2 stage approach and generate the .c file, then the whole component, so that you can refer back to the .c file in the case of any errors and see where they actually occur.

gcc 4.9 is a lot better than previous versions and most of the time the `^` carat char is actually pretty close to the offending section of the error line.

bschousek, 2017-05-18:
I've been bit by this same issue in recent days myself and since it's been a few years since I've produced lots of C code it has been a real stumbling block in a process which is intended to hide some of the boilerplate magic necessary. It does seem like it shouldn't be that difficult to map component.icomp lines to component.c lines and properly report the offending error, though I've looked at the instcomp script and know it would take a fair amount of digging for me before the difficulty was overcome.

Shown below is the test code I was putting together to post this as an issue when I found this already existing issue. There is an intentional error on line 11 that gets incorrectly flagged as occurring on line 14.

```
1 component funkyline "See if the line numbers are correct in a compile";
2
3 pin in unsigned signal_to_hal;
4 
5 function _;
6 license "GPL" ;
7 ;;
8 FUNCTION(_) {
9 	int foo1, foo2, bar, baz;
10 	//incorrectly try and set a value to see what line number the error gets reported as
11 	signal_to_hal=1;
12	foo1=1;
13	foo2=13;
14	bar=3;
15	baz=7;
16	return 0;
17 }
```

```
instcomp --compile funkyline.icomp
Compiling realtime funkyline.c
funkyline.icomp: In function ‘funkyline_’:
funkyline.icomp:14:15: error: lvalue required as left operand of assignment
/usr/share/linuxcnc/Makefile.modinc:123: recipe for target 'funkyline.o' failed
make: *** [funkyline.o] Error 1

```

ArcEye, 2017-05-19:
The report is coming from gcc, not instcomp.
It will only report what it finds when compiling.

gcc also produces errors after a line that an error occurred on quite often too.
eg. forget a closing semicolon
```
22.      foo = bar
23.      if(foo)
24.            bar2+=bar;
```
will get an error similar to `line 23: expected ';' before if statement`

luminize, 2017-05-19:
well, as I said, its no showstopper. It's documented now in this issue, so people can find it. I'll close it.

closed 2017-05-19 by luminize