machinekit/machinekit #1215: question about hardware drivers

davidelang, 2017-06-10:
It appears that the hardware drivers are all focused on driving step/direction type interfaces.

Is there any ability to talk to a stepper controller using higher-level constructs? There are now a few different devices out there (including pthat and odrive) that are designed for very high speeds where sending a series of step/direction pulses would be a very high overhead for both sides of the communications. They both have serial interfaces that allow you to send a command like "go 3000 steps in direction X and 50 steps in direction Y at a speed of Z with an acceleration of W; start"

Is it possible to create an interface to these sorts of boards from machinekit without a major re-write?

sirop, 2017-06-10:
So you want to use either USB or SPI?

ArcEye, 2017-06-10:
> Is it possible to create an interface to these sorts of boards from machinekit without a major re-write?

Short answer no.

Machinekit is a complete machine controller.
If you have such a board, it is already a controller, all you need is a DNC type program to feed the gcode to it.

davidelang, 2017-06-10:
@sirop, yes the interface between the computer running the g-code interpreter and planner and the board implementing the motion would be serial-over-USB or similar

@ArcEye Too bad, I was hoping to avoid the need to reinvent the wheel and be able to use the gcode interpreter and planner (and possibly be able to interface with multiple boards at once, odrive only handles 2 axis for example)

neither of these boards can accept g-code.

I had dug a little into grbl and saw that it decomposes the g-code into straight line moves that it then dispatches to the motion control routines, and so there is a possible point to hook in there. I was hoping that something like machinekit or linuxcnc would be allow for a similar place to hook in. I see hints of there being something like that in the comments with footnotes about nml, but I wasn't seeing anyting using it, so I figured I'd ask.

machinekoder, 2017-06-28:
I would like to see Grbl hooked up to Machinekit as playout. However, I'm not sure if there any Linux ports.

luminize, 2017-06-28:
> On 28 Jun 2017, at 17:57, Alexander Rössler <notifications@github.com> wrote:
> 
> I would like to see Grbl hooked up to Machinekit as playout. However, I'm not sure if there any Linux ports.
> 
https://github.com/grbl/grbl/wiki/Development-Path-and-Future-Needs

Looks like they want to have it on more types  of hardware.

rene-dev, 2017-07-11:
On the stmbl servo, we send floating point target position and velocity over smartserial, as well as enable, fault, feedback position, and GPIOs. can be easily implemented anywhere, the driver mcu just requires a 2.5 mbit uart.