machinekit/machinekit #1223: hal_gpio fails to load

kfmut, 2017-06-22:
Hi,

I've installed Machinekit on RPI2 and using this configuration https://drive.google.com/file/d/0B4EdEdEGvEuERkR4eUhXeWRQbHc/view?usp=sharing as starting point for my own. But Machinekit fails to load it with "katum2.hal:7: insmod failed, returned -22:rtapi_app_main(hal_gpio): -22 Invalid argument", line 7 in config file is simple "loadrt hal_gpio dir=16911384". I've tried to remove "dir=16911384" from that line, but error still there. I've contacted with author and he confirmed that this configuration works for him fine, his setup was made in mid-May. Also tried to change hal_gpio to hal_chip_gpio to see if it loads at all and it loads ok but fails later on configuration as expexted. 

My Machinekit version is 0.1.1497640487-1mk.travis.master.gitf3a69ee4~1raspbian﻿
Raspbian is latest image dated 2017-04-10
uname -a says "Linux kCNC 4.4.4-rt9-v7+ #7 SMP PREEMPT RT Mon Mar 7 14:53:11 UTC 2016 armv7l GNU/Linux"
latency-test is working, simulations are also working.

kfmut, 2017-06-22:
Sorry but only error message is about insmod without any futher explanations.

linuxcnc.log: https://drive.google.com/open?id=0B5NfNsKzoKH0eEF1c3VQSWc3U3c

My /var/log/dmesg is empty, contents of /var/log/messages if it helps: https://drive.google.com/open?id=0B5NfNsKzoKH0cDl3dDF3NmduTFk

This is output to stderr if machinekit started with -d option: https://drive.google.com/open?id=0B5NfNsKzoKH0NkFBZ0xwMW1YWFk

kfmut, 2017-06-22:
Author of previously mentioned sample config have tried it with newer Machinekit version and it works OK. Looks like problem is specific to my setup...

kfmut, 2017-06-22:
Looks like I somehow screwed up with DEBUG=5 in config, sorry :-(

New linuxcnc.log: https://drive.google.com/open?id=0B5NfNsKzoKH0NWhLejFWX2lpbms

And I have found cause of it "HAL_GPIO: ERROR: board revision 5 not supported".
I'm using RPI2 revision 1.2 as it stated on board itself.

ArcEye, 2017-06-22:
I suspected it might be something like this, it came up before when rev 4 needed to be supported.

There is the board version and the cpu revision number, some boards seem to have different chips to that expected for their board number.

If you do `cat /proc/cpuinfo` it should tell you what it thinks you have

That is what is interrogated by get_cpuinfo_revision()
https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/cpuinfo.c#L27
you can check what you get back against the assumptions that makes.

Paste it in full here and we can see if the pinouts are published for that particular rev.

ArcEye, 2017-06-22:
I suspect that you will have this
```
     cat /proc/cpuinfo
    processor       : 0
    model name      : ARMv7 Processor rev 5 (v7l)
    BogoMIPS        : 38.40
    Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
    CPU implementer : 0x41
    CPU architecture: 7
    CPU variant     : 0x0
    CPU part        : 0xc07
    CPU revision    : 5

    processor       : 1
    model name      : ARMv7 Processor rev 5 (v7l)
    BogoMIPS        : 38.40
    Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
    CPU implementer : 0x41
    CPU architecture: 7
    CPU variant     : 0x0
    CPU part        : 0xc07
    CPU revision    : 5

    processor       : 2
    model name      : ARMv7 Processor rev 5 (v7l)
    BogoMIPS        : 38.40
    Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
    CPU implementer : 0x41
    CPU architecture: 7
    CPU variant     : 0x0
    CPU part        : 0xc07
    CPU revision    : 5

    processor       : 3
    model name      : ARMv7 Processor rev 5 (v7l)
    BogoMIPS        : 38.40
    Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
    CPU implementer : 0x41
    CPU architecture: 7
    CPU variant     : 0x0
    CPU part        : 0xc07
    CPU revision    : 5

    Hardware        : BCM2709
    Revision        : a01041
```

This is correct for a P2 but returns a higher cpu revision than a Pi3 :unamused:

kfmut, 2017-06-22:
Here we go
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2709
Revision	: a22042
Serial		: 000000001dd92892
```

ArcEye, 2017-06-22:
OK, that explains it

https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/cpuinfo.c#L75
shows a check for rev 4 against
```
else if ((strcmp(revision, "a22082") == 0) ||
	    (strcmp(revision, "a02082") == 0))
      return 4;
```

It should also check `Revision	: a22042`, which is not in the code at all, so it defaults to version 5 which is 'not supported' even though the pins are undoubtedly the same.

The original work in this file by Ben Croston is from 2012 and there seem to have been a plethora of board versions, hardware versions and cpu revisions since then.

These are the main ones now
[image]

I will add this version to the checks and push that.

Let you know when the packages are ready for testing, won't be until tomorrow at least

kfmut, 2017-06-22:
ArcEye, thank you! I was strugling with it whole day without any results before issueing this ticket and now I'm really glad that it could be fixed!

ArcEye, 2017-06-23:
I have just pushed the packages into the repo.

I deliberately only included your hardware revision into the code.
If I included all the missing revisions and any of them had different pinouts, that would present a very hard to resolve error, whereas the one you had is a simple one to identify, as a starting point for a new revision number.

Just needs you to update, install the new packages and test :smile:

ArcEye, 2017-06-23:
Damn.

Just realised the raspbian packages are not being produced at all at present, so you cannot update.

Unfortunately not likely to be fixed before next week.

I will let you know when resolved.

kfmut, 2017-06-23:
No worries, I will definitely check it when it would be available :-)

ArcEye, 2017-06-23:
@kfmut 
I have been experimenting with a local Docker cross-build system

I have produced a hal_gpio.so module from the latest version of the machinekit repo, which should run on raspbian rt-preempt
[hal_gpio.zip](https://github.com/machinekit/machinekit/files/1098359/hal_gpio.zip)

Try downloading the above, 
extracting the binary, checking it is still mod 644 (in case the uploading and zipping changed the permissions)
rename `/usr/lib/linuxcnc/rt-preempt/hal_gpio.so` to `hal_gpio.so.old` or similar
drop the new one in its place and see if machinekit will load it.

It might solve the problem for you until packages are back on line properly and test my cross-build too.

kfmut, 2017-06-23:
Yep, it's loaded! Thank you! Now it's time to hook up some stepper drivers and experiment with them :-)

ArcEye, 2017-06-24:
Excellent - I will close the issue then.
When they are back on line, it will be in the mainstream packages too.

ArcEye, 2018-04-01:
Can I assume your edit looks something like this?
```
char *get_cpuinfo_revision(char *revision)
{
   FILE *fp;
   char buffer[1024];
   char hardware[1024];
   int  rpi_found = 0;

   if ((fp = fopen("/proc/cpuinfo", "r")) == NULL)
      return 0;

   while(!feof(fp)) {
      fgets(buffer, sizeof(buffer) , fp);
      sscanf(buffer, "Hardware	: %s", hardware);
      if (strcmp(hardware, "BCM2708") == 0)
         rpi_found = 1;
      else if (strcmp(hardware, "BCM2709") == 0)
         rpi_found = 1;
 >>   else if (strcmp(hardware, "BCM2835") == 0)
         rpi_found = 1;  <<
      sscanf(buffer, "Revision	: %s", revision);
   }
   fclose(fp);

   if (!rpi_found)
      revision = NULL;
   return revision;
}
```

The revision number is already in the code for rev 4

Just wish they would stop using different processors / revision codes  :smile:

AlanValmorbida, 2018-04-01:
Exactly this edit.

ArcEye, 2018-04-01:
OK, I will add that processor to the cpuinfo.c code

closed 2018-04-01 by ArcEye

Sharu-IF, 2018-05-09:
Hi, i have a raspberry pi model 3B with the following specs
Hardware BCM2835
Revision a32082

How can i solve this problem?

ArcEye, 2018-05-09:
There will be a PR for this revision and another new one listed shortly 
https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md

Then just need to wait for it to be merged and new packages made.

PR #1372 refers

ArcEye, 2018-05-10:
Packages with serial `_0.1.1525927436.git10bcf99-1` now in the repository.

They contain the amended hardware revisions and the RPi driver should work with them.

luminize, 2018-05-11:
> On 11 May 2018, at 18:21, Claudio Sciortino <notifications@github.com> wrote:
> 
> @ArcEye thank you!
> Can you explain me how to download this new package? I'm a noob at this things.

Upgrade your packages should do that.
sudo apt upgrade

Sharu-IF, 2018-05-12:
Hi @luminize, with sudo apt upgrade, my raspberry finds 0 packages to update or to install.

PR #1365 (2018-04-02): Add BCM2835 hardware to recognised processor types for RPi3
Fixes #1223
commit c53d4e1f (2018-04-02): Add BCM2835 hardware to recognised processor types for RPi3

Fixes #1223
PR #1372 (2018-05-10): Add new Rpi3B revision numbers to have them recognised.
Fixes #1223 (https://github.com/machinekit/machinekit/issues/1223#issuecomment-387739491)