machinekit/machinekit #1225: Is there anyway to speed up axis gui on BBB

spaivaras, 2017-06-23:
At current state GUI is totally unusable on, jog buttons latch with selected action, controls have a very delayed reaction, and i don't even speak about preview window.

BBB seems to have an opengles driver, maybe someone tried to use it for the preview?
Maybe turning of preview helps at least a bit (can't find any information how to do that)

Or does people don't use BBB + machinekit setups at all?

darenschwenke, 2017-06-23:
There are quite a few existing threads on this.  I'll try to sum up.
OpenGL rendering is software based in the current offerings, which kills
Axis performance.
The realtime kernel and hardware opengl do not play nice.  This may work
soon, although I have no idea of the timing.
The current common workaround is use Cetus or Machineface remote
interfaces, depending on your target.  Rendering is then client side

cdsteinkuehler, 2017-06-24:
On 6/23/2017 2:38 PM, Aivaras Spaicys wrote:
> At current state GUI is totally unusable on, jog buttons latch with selected
> action, controls have a very delayed reaction, and i don't even speak about
> preview window.
>
> BBB seems to have an [...]

In addition to Daren's advice, you may find acceptable performance by
disabling preview (using (AXIS stop)) and switching to the DRO tab (so
the 3D preview is hidden), or just disable the 3D preview entirely:

http://www.machinekit.io/docs/gui/axis-machinekit-enhancements/#preview-disable

...and it helps if you at least *try* to dig through the existing
documentation and forums before assuming your question isn't a FAQ.

-- 
Charles Steinkuehler
charles@steinkuehler.net

spaivaras, 2017-06-24:
Thank you for the responses. I did dig thru issues and documentation. The last thread I saw about it was @2014 so three years ago. I also looked into qt remote GUis. But the quotas for binaries were overlimited and building qt myself is not up to my skills. So please excuse me for the issue. In any case thank you for the replies. I now have things to try.

darenschwenke, 2017-06-24:
Try the forum:
https://groups.google.com/forum/?utm_source=digest&utm_medium=email#!forum/machinekit/topics

machinekoder, 2017-06-27:
@spaivaras The quota problem has been fixed.

spaivaras, 2017-06-27:
@machinekoder thank you, already downloaded. I guess this could be closed as all of my questions where answered, or it could be labeled with something like "not-a-bug" and left for future reference.

closed 2017-06-28 by machinekoder