machinekit/machinekit #1228: Building Machinekit on Debian Stretch

machinekoder, 2017-06-28:
This thread is dedicated to building Machinekit on Debian Stretch. Some defaults and package names seem to have changed. Therefore, some work might be necessary to get everything up and running.

machinekoder, 2017-06-28:
The hardest hurdle I'm facing is the new CZMQ4 shipping with Stretch. CZMQ4 breaks the API compared to CMZQ3 used in Machinekit.

ArcEye, 2017-06-28:
Machinekit builds after some changes.

Using gcc 6.3 produces errors at some code that passed 4.9.2
You need to fix the issue here https://github.com/machinekit/machinekit/pull/1092
I did that here and fixed another problem too
https://github.com/machinekit/machinekit/pull/1125

AFAIR czmq lib has changed and is no longer backwardly compatible
I used our existing lib 
This issue refers https://github.com/machinekit/machinekit/issues/1152

I did not do much more than get everything to build and check that it worked.
With the demise of Wheezy imminent (all support from Debian finishes next year), it is probably time to start looking a bit closer at Stretch

machinekoder, 2017-06-28:
@AarcEye Did you use the Jessie package repository or did you build the CZMQ library from source?

ArcEye, 2017-06-29:
Good question.

I __think__ I used the existing package, downgrading to v3.  
The way I wrote up my reply on Issue #1152 suggests I thought it was a readily available option to all.

I will mount that partition later and confirm however, because I _did_ build the lib for another project, so have the sources readily available.

ArcEye, 2017-06-29:
In fact I built the libs, was a while ago, just could not remember.

I used zeromq-4.0.5 and czmq-2.2.0

machinekoder, 2017-06-29:
@ArcEye I was able to compile with zeromq-4.0.8 and czmq-2.2.0 from source. The good news is that it worked with Protobuf 3.x after some modifications of the proto files.

machinekoder, 2017-06-29:
Btw. Stretch uses the `libzmq5` and `libczmq4` package names. So I guess we could provide the `libzmq3` and `libczmq2` packages from the Machinekit repository without major problems.

ArcEye, 2017-06-30:
I have made packages for stretch-amd64 of zeromq and czmq.

They are at http://deb.mgware.co.uk
instructions, gpg key etc are on that page

(Also contains the latest Jessie packages for all 3 architectures)

ArcEye, 2017-07-04:
NB.
In my last commit <18d3df2> I added a switch to the debian package building system, which makes
dh_shlibdeps more forgiving of libraries which are locally built, or come from packaged locally built libs.

This allows package builds under Stretch, whilst the czmq / zeromq libraries are resolved.

propcoder, 2017-07-17:
Would it be possible to compile and provide the packages for x86 (32bit) architecture too?

propcoder, 2017-07-17:
I want to install machinekit in Stretch on BeagleBone, but struggle on zeromq and czmq - how should I install packages? I did include mgware deb sources..

ArcEye, 2017-07-17:
The packages are amd64 only.
Outside of that you need to build from sources.

I don't know why you would want to run Stretch on BBB.  
There are already images and packages for Jessie,  this thread is really just looking forward to anticipate the forthcoming problems with porting Machinekit to Stretch, or any distro that now uses gcc-6.x.x and newer incompatible versions of libraries.

propcoder, 2017-07-17:
MachinekitClient does not work over OpenVPN tun interface on wheezy and jessie, while it works on stretch (by @machinekoder research) - that's why I need it.

ArcEye, 2017-07-18:
OK.
Until there is an armhf build system targetting Stretch you need to build locally.

The source code can be downloaded from 
https://github.com/zeromq/czmq/releases
https://github.com/zeromq/libzmq/releases
just make sure you get a version before the API changes

ArcEye, 2017-07-19:
@propcoder 
There are now amd64 and armhf versions of czmq / zeromq deb packages in my repo
http://deb.mgware.co.uk

You need to follow the instructions there and then install whilst ensuring you fetch from the correct repo
(unless you are into pinning and preferences, a quick and dirty way is just comment out all other repos,
`apt-get update && apt-get install czmq zeromq`  and then uncomment and update again)

The armhf libs are cross-built and untested, so if you could let me know how they go.

ArcEye, 2017-08-04:
> The armhf libs are cross-built and untested, so if you could let me know how they go.

I have just built machinekit and Machinekit-HAL on a native Debian Stretch armhf server using these packages and all appears OK

ArcEye, 2017-08-05:
Ref https://github.com/machinekit/machinekit/issues/1228#issuecomment-316364736
Now added a cross-built i386 package for czmq and zeromq to my repo.

machinekoder, 2017-09-13:
@ArcEye I have just tested your machinekit packages. It looks like the reference to python-zmq is missing. Installing python-zmq from the stretch repositories fixes the problem. No version conflicts noticed.

ArcEye, 2017-09-13:
Those packages are out of date.

I left out python-zmq, because I don't use python and the dependency on libzmq5
```
Package: python-zmq
Source: pyzmq
Version: 16.0.2-2
Architecture: amd64
Maintainer: Debian Python Modules Team <python-modules-team@lists.alioth.debian.org>
Installed-Size: 1184
Depends: python (<< 2.8), python (>= 2.7~), python:any (<< 2.8), python:any (>= 2.7.5-5~), libc6 (>= 2.14), libzmq5 (>= 4.1.2)
Section: python
Priority: optional
Homepage: http://www.zeromq.org/bindings:python
Description: Python bindings for 0MQ library
```
I found that it pulled in the wrong lib and configure kept using that one and the build failed.

I am very surprised that this version 16.0.2 of python-zmq works

It contains ZMQ_IMMEDIATE, zmq_ctx_destroy, zmq_msg_recv etc. which I thought were some of the incompatible changes in the C code which prevents the new libs being used.

machinekoder, 2017-09-13:
@ArcEye libzmq5 is installed in parallel to the libzmq3 so Python uses the newer version and the rest of Machinekit the old version. The Python ZMQ library is backward compatible.

What do we need to get the Stretch packages up and running?

ArcEye, 2017-09-13:
We need a builder, but that means a version of the mk-cross-builder using Stretch containers, instead of Jessie.
The advantage of that will be fast armhf builds, just as we have for Machinekit-HAL in Jessie currently.
I need to pick @zultron s brains on that when I revisit it.

In the interim I have been cross-building locally, using multistrap chrooted directories.
Now the changes to dh_python renaming of libs are through, I am looking at setting up a local Jenkins job to automate the building and pushing up to my repo.

Once that it proven, I can alter the structure of the deb.machinekit.io repo to include Stretch and pushes can be diverted there.

Only once we have a builder container capable of doing Stretch builds triggered from github webhooks, will we have an automated system as exists for Wheezy / Jessie.

But one thing at a time.

If you say python-zmq is not a problem for a package dependency (it was for a build), I can reinstate that.

Next I need to get the same changes into Machinekit, that I have made to Machinekit-HAL.
ie the packaging mainly plus a changed build, that make the pru_generic.* pru_decamux.* binaries available to all kernel flavours in armhf

@luminize has just tested package installs of Machinekit-HAL on a BBB with rt-preempt and the hpg.xx
stuff all works

THEN, I will look at the automation of a private repo, to start with, of Stretch packages.

Seems like a plan 😄

ArcEye, 2017-09-16:
@machinekoder 
Re https://github.com/machinekit/machinekit/issues/1228#issuecomment-329143344

I have now finalised the packaging for Stretch and Jessie on machinekit-hal.
That too has `python-zmq` back in the install deps only 
(it was it being in the build deps that caused problems before)

Also found that somehow `libjansson` had been removed from the deps, but only discovered that when testing on a bare install partition. That is now back.

The `machinekit-hal` Stretch packages are fully functioning and being automatically updated in my repo.

Need to get all the packaging stuff across to `machinekit` now, and do the same for Stretch packages there.

ArcEye, 2017-09-17:
@machinekoder
Further re #1228 (comment)

My repo now has finalised packages for Stretch, of machinekit with the pru_generic.* binaries built for all armhf builds.
Contains all the stuff in PR #1279

ArcEye, 2017-09-26:
I certainly am :)

Robert Nelson is slowly deprecating xenomai in favour of rt-preempt, with the latency difference closing.  Also Xenomai3 will not work with machinekit.

Since the BBB uses the PRU for hardware stepgen, encoders etc there is no need for it there.
I have recently changed the builds so that the pru_generic.* binaries are available for all kernel flavours

It is useless for software stepgen in x86_64 and unnecessary if using Mesa boards.
DE0-NANO-Soc FPGA boards use rt-preempt on arm

zultron, 2017-10-14:
Take a look at PR #1293, successful builds with the Jessie and Stretch "sysroot" images for amd64, i386 and armhf.  If the results of this build system could be verified, I'd feel pretty happy with the work.  After adding the package uploading machinery back in, this could be a candidate for a replacement build system.  Thanks, @ArcEye and @machinekoder for getting me past the stumbling blocks while I worked this out.

ArcEye, 2017-10-15:
I will pull it into my fork and test.
I have a few test builders which can be adapted quite easily.

ArcEye, 2017-10-15:
Well done @zultron.

Pulled your commits into a branch and have just done 6 command line docker builds, 1 for each flavour and distro.

All built seamlessly and **very** fast.

I will get an automatic build running this afternoon using the images

ArcEye, 2017-10-15:
Now have a TravisCI build running from a repo containing your commits.
https://travis-ci.org/ArcEye/machinekit-cross-builder-tester/builds/288193776?utm_source=github_status&utm_medium=notification

They are building fine, but confirm something I noted in the manually started builds

The changelog appears to be being written with the new version
```
New changelog entry:

machinekit (0.1.1508072768.gitc037246-1travis.ArcEye.machinekitcrossbuildertester.master~1stretch) experimental; urgency=low

  * Travis CI rebuild

    - TAG amd64_9

    - PR/branch master

    - Commit c037246

    - https://github.com/machinekit/machinekit/commit/c037246

 -- Mick <arceye@mgware.co.uk>  Sun, 15 Oct 2017 13:08:51 +0000
```
but the packages are not reflecting this (apart from the source tarballs etc.)
```
dpkg-deb: building package 'machinekit-dbgsym' in '../machinekit-dbgsym_0.1.1-1_amd64.deb'.
dpkg-deb: building package 'machinekit-posix-dbgsym' in '../machinekit-posix-dbgsym_0.1.1-1_amd64.deb'.
dpkg-deb: building package 'machinekit-posix' in '../machinekit-posix_0.1.1-1_amd64.deb'.
dpkg-deb: building package 'machinekit' in '../machinekit_0.1.1-1_amd64.deb'.
dpkg-deb: building package 'machinekit-rt-preempt-dbgsym' in '../machinekit-rt-preempt-dbgsym_0.1.1-1_amd64.deb'.
dpkg-deb: building package 'machinekit-rt-preempt' in '../machinekit-rt-preempt_0.1.1-1_amd64.deb'.
```
Need to go back to a single manually started build and do a post mortem on the changelog after build.
Not quite sure what is happening.

ArcEye, 2017-10-15:
The runtests are failing on both amd64_8 and amd64_9 (same tests fail)
```
Runtest: 157 tests run, 149 successful, 8 failed + 0 expected
Failed: 
	tests/encoder-ratio-stepgen-v2.0
	tests/encoder-ratio.0
	tests/halui-jogging
	tests/hm2-idrom
	tests/motion-logger/mountaindew
	tests/motion/g0
	tests/save.1
	tests/usercomp.0
```

ArcEye, 2017-10-15:
> The changelog appears to be being written with the new version
but the packages are not reflecting this (apart from the source tarballs etc.)

This is related to the new method of creating changelog.new and pointing to it with
```
	# Set changelog location
	BUILD_OPTS+=" --changes-option=-ldebian/changelog.new"
```
`changelog.new` is correct, but dpkg is using the default original `changelog`

ArcEye, 2017-10-15:
Re https://github.com/machinekit/machinekit/issues/1228#issuecomment-336712115 failing runtests

If you look at what tests were run in the initial PR which passed, it only ran 149 tests and skipped the rest, such as hm2_id_rom, which always seems to fail in a docker env.

The failing tests seem to be the other 8, which were not run originally and failed in the test build.

zultron, 2017-10-15:
In the sysroot Docker images, I changed the user ID to 2000 to match Travis, so that `sudo` would work again.  It might be that enabling `sudo` revealed some failures masked by `rtapi_app_posix` not being setuid.

I'll take a look.  Sorry for sneaking that change in there, and thanks for your vigilance.

zultron, 2017-10-17:
So `sudo` is fixed, the images are updated with the latest MK `debian/config`, and builds are all passing.  I never saw the test failures you did, but if the tests are still failing on your side, I'm happy to help work through it.

ArcEye, 2017-10-17:
If I clone your repo, checkout `stretch-sysroot-builder` and run `./scripts/build_docker -t amd64_8 -c deb` , I get a set of packages with proper unique serial, so changelog stuff is fixed and all builds. 👍 

The test failures were peculiar.
If I run tests on your branch, I only get the 149 tests, the other 8 being dragged in must have been something to do with the existing branch I cherry-picked your commits into, not simply cloning your repo?

If I run `./scripts/build_docker -t amd64_8 -c test` however I get
`Runtest: 149 tests run, 74 successful 75 failed + 0 expected`

The failures are because I started the docker as root.  Digging into the stderr files, the tests are being run as root too, which will not work.

An unforeseen consequence of passing the UID through to docker ?
I can normally run a docker as whatever and it runs in its own permissions/access, irrespective of the launching users UID.

zultron, 2017-10-17:
I don't know of the downside of running commands in docker with your own UID, if you trust the container.  That can be a big "if", but at least the container builds are transparent.

For me, my development process is a lot smoother with less annoyance when the environment both inside and outside the container are the same, especially the same UID and the same git repo and home directory bind-mounted to the same paths, as the `build_docker` script does:

- Files and directories in the repo are all created with the same UID, no matter whether created from inside or outside the container, so no annoying permissions issues when going back and forth
- With `$HOME` bind-mounted in the container, shell commands pick up my customized settings in `~/.bashrc`, `~/.gitconfig`, etc.
- Filesystem paths inside the container can be copy-and-pasted without modification outside the container, and vice-versa

Of course this is all designed around my own dev workflow, and everyone's is different.  If you're not comfortable with this scheme, I'm happy to think about how to satisfy other requirements.

machinekoder, 2017-10-17:
I usually use something along the lines to fix the permission problem in docker:

```bash
docker_image=greenhive/qt-docker-android-armv7:latest
project=greenhive-ui
platform=android-armv7
recipe=Linux/android/Recipe

docker pull ${docker_image}
docker rm ${project}-${platform} &> /dev/null || true
docker run --name ${project}-${platform} -i \
     -v /etc/group:/etc/group:ro -v /etc/passwd:/etc/passwd:ro \
     -u $( id -u $USER ):$( id -g $USER ) \
     -v "${PWD}:/${project}" ${docker_image} \
     /bin/bash -c "cd ${project}; ./build/${recipe}"
docker rm ${project}-${platform} &> /dev/null
```

I'm not sure if this a good approach, but it works.

ArcEye, 2017-10-17:
For running inside Travis, it will be fine.
If I had not had problems with the build-tests earlier, I would never have thought to run them locally through the docker and discovered this.

What I don't usually do is bind mount $HOME however.  I have so many paths etc in `.bashrc`, it has caused problems in the past, so I just bind mount $PWD and put whatever I want to be discoverable in there.

Again, inside Travis this is not going to be an issue, because it only contains what has been set up prior to calling the docker image.

See https://github.com/machinekit/machinekit/pull/1293#discussion_r145146718 re the gst package name and the earlier https://github.com/machinekit/machinekit/pull/1293#discussion_r144713445

With those resolved I anticipate fully installable packages 🥂 

I can just upload them to my repo and test an install on a jessie amd64 chroot partition I use for testing.

zultron, 2017-10-17:
@machinekoder, bind-mounting `/etc/{passwd,group}` works well as long as that's where the password database lives.  It won't work in my environment where nss points at ldap.

@ArcEye, likewise, I see why my scheme of mounting `$HOME` won't work in your environment.

Would it make more sense to remove some of those arguments in `scripts/build_docker`, especially `docker run -e USER=${USER} -v ${HOME}:${HOME} -e HOME=${HOME}`, and replace them with e.g. `docker run ${DOCKER_USER_OPTS}`?  (Note that it's already possible to override `$UID_GID` from the environment.)  Then everyone could do this stuff however they wanted to:  bind `$HOME` or not, bind `/etc/passwd` or not, etc., and they can be set as needed in `.travis.yml` to work in that environment.

ArcEye, 2017-10-17:
@zultron , I would not bother too much about it.

The main thing that matters is that it reliably builds packages though Travis.

Very few people will be sticking their head under the hood. 
When I run a docker from the command line I usually strip out the passed environment to the minimum etc anyway, create an interactive session and build manually inside it rather than using the script.

ArcEye, 2017-10-17:
>   See #1293 (comment) re the gst package name and the earlier #1293 (comment)
    With those resolved I anticipate fully installable packages 🥂

In fact I have made those changes locally, built packages, uploaded to my repo and done a test install
It works.
```
(Jessie-sda2-chroot)root@INTEL-i7:/# apt-get install machinekit machinekit-posix
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  gstreamer0.10-plugins-base libgstreamer-plugins-base0.10-0 libgstreamer0.10-0 python-gst0.10
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  freeglut3 libczmq4 libgtkglext1 libgtksourceview2.0-0 libgtksourceview2.0-common libidl0 liborbit2 libzmq3 python-gconf python-gnome2 python-gst-1.0 python-gtkglext1 python-gtksourceview2 python-opengl
  python-pydot python-pyorbit python-pyparsing python-simplejson python-zmq xdot
Suggested packages:
  python-gnome2-doc libgtksourceview2.0-dev libgle3
The following NEW packages will be installed:
  freeglut3 libczmq4 libgtkglext1 libgtksourceview2.0-0 libgtksourceview2.0-common libidl0 liborbit2 libzmq3 machinekit machinekit-posix python-gconf python-gnome2 python-gst-1.0 python-gtkglext1
  python-gtksourceview2 python-opengl python-pydot python-pyorbit python-pyparsing python-simplejson python-zmq xdot
0 upgraded, 22 newly installed, 0 to remove and 41 not upgraded.
Need to get 5,613 kB/8,654 kB of archives.
After this operation, 51.4 MB of additional disk space will be used.
Do you want to continue? [Y/n] n    
(Jessie-sda2-chroot)root@INTEL-i7:/# apt-get install machinekit machinekit-posix
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  freeglut3 libczmq4 libgtkglext1 libgtksourceview2.0-0 libgtksourceview2.0-common libidl0 liborbit2 libpgm-5.1-0 libsodium13 libzmq3 python-configobj python-gconf python-gnome2 python-gst-1.0
  python-gtkglext1 python-gtksourceview2 python-opengl python-pydot python-pyorbit python-pyparsing python-simplejson python-xlib python-zmq xdot
Suggested packages:
  python-configobj-doc python-gnome2-doc libgtksourceview2.0-dev libgle3
The following NEW packages will be installed:
  freeglut3 libczmq4 libgtkglext1 libgtksourceview2.0-0 libgtksourceview2.0-common libidl0 liborbit2 libpgm-5.1-0 libsodium13 libzmq3 machinekit machinekit-posix python-configobj python-gconf python-gnome2
  python-gst-1.0 python-gtkglext1 python-gtksourceview2 python-opengl python-pydot python-pyorbit python-pyparsing python-simplejson python-xlib python-zmq xdot
0 upgraded, 26 newly installed, 0 to remove and 41 not upgraded.
Need to get 5,613 kB/9,150 kB of archives.
After this operation, 53.1 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```
So it is looking near completion

zultron, 2017-10-17:
The [`python-gst-1.0`][1] issue is fixed.

I tested out the [deps][3] issue.  There was an issue with the `libzmq3` dep, since the package is called `libzmq5` in Stretch.  I decided to simply remove the explicit deps altogether, since I doubt the `(>= 4.0.4)` version dependency is actually meaningful, and  `shlibdeps` determines the dep to be `libzmq5 (>= 3.2.3+dfsg)` for Stretch.

This is all going on in the "Rework packaging for Stretch" commit; see [here][4].

[1]: https://github.com/machinekit/machinekit/pull/1293#discussion_r145146718
[3]: https://github.com/machinekit/machinekit/pull/1293#discussion_r144713445
[4]: https://github.com/machinekit/machinekit/pull/1293/commits/aa3e18edcd1b4ab450901c1022695e6645b7450c#diff-3a8967c9c7449446ddcdc76faa20bfafL53

zultron, 2017-10-17:
Also, I looked into re-enabling `tests/hm2-idrom`.  It can be done with some work in `configure.ac`, but that seems like it's for another issue, if it seems important enough.

ArcEye, 2017-10-18:
> This is all going on in the "Rework packaging for Stretch" commit; see here.

I have pulled it all and built all versions, of all packages, in a Jenkins builder and uploaded to my repo.

Installing onto a bare chrooted partition of Jessie 
```
(Jessie-sda2-chroot)root@INTEL-i7:/# apt-get install machinekit machinekit-posix
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  freeglut3 libczmq4 libgtkglext1 libgtksourceview2.0-0 libgtksourceview2.0-common libidl0 liborbit2 libpgm-5.1-0 libsodium13 libzmq3 python-configobj python-gconf python-gnome2 python-gst-1.0
  python-gtkglext1 python-gtksourceview2 python-opengl python-pydot python-pyorbit python-pyparsing python-xlib python-zmq xdot
Suggested packages:
  python-configobj-doc python-gnome2-doc libgtksourceview2.0-dev libgle3
The following NEW packages will be installed:
  freeglut3 libczmq4 libgtkglext1 libgtksourceview2.0-0 libgtksourceview2.0-common libidl0 liborbit2 libpgm-5.1-0 libsodium13 libzmq3 machinekit machinekit-posix python-configobj python-gconf python-gnome2
  python-gst-1.0 python-gtkglext1 python-gtksourceview2 python-opengl python-pydot python-pyorbit python-pyparsing python-xlib python-zmq xdot
0 upgraded, 25 newly installed, 0 to remove and 43 not upgraded.
Need to get 5,503 kB/9,025 kB of archives.
After this operation, 52.8 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```
and then Stretch (the Stretch one is a bit barer, hence extra packages in the list)
```
(amd64-stretch-chroot)root@INTEL-i7:/# apt-get install machinekit machinekit-posix
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  blt bwidget freeglut3 gir1.2-gst-plugins-base-1.0 gir1.2-gstreamer-1.0 graphviz libann0 libboost-serialization1.62.0 libboost-thread1.62.0 libcdt5 libcgraph6 libczmq4 libgtkglext1 libgtksourceview2.0-0
  libgtksourceview2.0-common libgts-0.7-5 libgts-bin libgvc6 libgvpr2 libidl-2-0 libmodbus5 liborbit2 libpango1.0-0 libpangox-1.0-0 libpathplan4 libprotobuf10 libtk-img libxdot4 python-avahi python-configobj
  python-gconf python-gdbm python-gnome2 python-gst-1.0 python-gtkglext1 python-gtksourceview2 python-imaging python-opengl python-pil python-pil.imagetk python-protobuf python-pydot python-pyftpdlib
  python-pyorbit python-pyparsing python-sendfile python-simplejson python-tk python-vte python-xlib python-zmq python3-gi-cairo tcl-tclreadline tk8.6-blt2.5 xdot yapps2-runtime
Suggested packages:
  blt-demo graphviz-doc libtk-img-doc python-configobj-doc python-gnome2-doc python-gdbm-dbg libgtksourceview2.0-dev libgle3 python-pil-doc python-pil-dbg python-pil.imagetk-dbg python-pyparsing-doc tix
  python-tk-dbg
The following NEW packages will be installed:
  blt bwidget freeglut3 gir1.2-gst-plugins-base-1.0 gir1.2-gstreamer-1.0 graphviz libann0 libboost-serialization1.62.0 libboost-thread1.62.0 libcdt5 libcgraph6 libczmq4 libgtkglext1 libgtksourceview2.0-0
  libgtksourceview2.0-common libgts-0.7-5 libgts-bin libgvc6 libgvpr2 libidl-2-0 libmodbus5 liborbit2 libpango1.0-0 libpangox-1.0-0 libpathplan4 libprotobuf10 libtk-img libxdot4 machinekit machinekit-posix
  python-avahi python-configobj python-gconf python-gdbm python-gnome2 python-gst-1.0 python-gtkglext1 python-gtksourceview2 python-imaging python-opengl python-pil python-pil.imagetk python-protobuf
  python-pydot python-pyftpdlib python-pyorbit python-pyparsing python-sendfile python-simplejson python-tk python-vte python-xlib python-zmq python3-gi-cairo tcl-tclreadline tk8.6-blt2.5 xdot yapps2-runtime
0 upgraded, 58 newly installed, 0 to remove and 103 not upgraded.
Need to get 12.6 MB/16.5 MB of archives.
After this operation, 85.3 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

There is a slight difference between Jessie and Stretch, but that is because of the specific deps in my
jessie libczmq packages, which are satisfied more generically in stretch by libzmq5.

Comment https://github.com/machinekit/machinekit/pull/1293/#discussion_r145364380 notwithstanding, we are just about there.

Well done @zultron

ArcEye, 2017-10-18:
I have now created the Stretch distribution in the main repo, in preparation for package uploads.

So far just contains the 3 relevant `all arch` packages, as a kind of placeholder.
http://deb.machinekit.io/debian/dists/stretch/main/binary-amd64/Packages

ArcEye, 2017-10-19:
> I don't see any strict Machinekit deps in http://deb.mgware.co.uk/dists/stretch/main/binary-amd64/Packages. Does that mean that with Stretch, MK can finally be built purely from mainline packages?

Yes, finally, having got the czmq4 API sorted it can all come from stock Debian packages.
First time ever for us!

luminize, 2017-10-19:
Well done guys!

ArcEye, 2017-10-19:
TODO:
- The docs are now changed, so the rt-preempt control files can just point to 
`linux-image-rt-[686-pae|amd64]`

- @machinekoder  reported that the machinekit package has begun to contain comp again.
I checked my Stretch packages and this is correct, so somewhere a `*.install.in` file has got reverted to before I removed the machinekit-dev package.
I will look into it in a bit.

- I don't think we need to change the skip on hm2_id_rom test.  It used to cause all sorts of problems inside a docker, which was why it was omitted.

zultron, 2017-10-19:
On 10/19/2017 10:44 AM, ArcEye wrote:
 >
 >     @machinekoder <https://github.com/machinekoder> reported that the
 >     machinekit package has begun to contain comp again.
 >     I checked my Stretch packages and this is correct, so somewhere a
 >     |*.install.in| file has got reverted to before I removed the
 >     machinekit-dev package.
 >     I will look into it in a bit.

Oops, that was me.  I'll fix it.

ArcEye, 2017-10-19:
OK, if easier, comp could just be in the machinekit package, it is the Makefile.modinc file that must be in the `machinekit-{flavour}` package to ensure the flavour is set correctly

zultron, 2017-10-19:
Sorry about that, an overzealous clean-up of `debian/rules`.  I simply reverted the offending commit, and everything should be back the way it was.  Changes already pushed.

zultron, 2017-10-19:
Ok, I had the same idea you did, and the separate Stretch packaging stuff is all folded back together, if I did it right.

I have **not** looked at the package upload stuff.  I used to used the package-cloud stuff, but not anymore, and I believe the MK project has its own way of using sftp to push packages to MAH's server.  Someone else will have to look at that bit for me, hopefully as easy as replacing some lines I chopped out of `.travis.yml`.

zultron, 2017-10-19:
I did my best to add back the `.travis/send_binaries.sh` and `.travis/send_status.sh` scripts.  Still need someone to test....

ArcEye, 2017-10-19:
I'll do a rebuild and test tomorrow

ArcEye, 2017-10-20:
Looks OK until the last send_status bit.  I'll look later.

```
The command "scripts/build_docker -i $IMAGE -t $TAG -c $CMD" exited with 0.

cache.2
store build cache
0.01s
12.45schange detected (content changed, file is created, or file is deleted):
/home/travis/.ccache/0/0/53702bd6ff98c572c5c58c57cbef41-5319.manifest
/home/travis/.ccache/0/0/67478c4468ceafd2b25cabf976a367-2785.manifest
/home/travis/.ccache/0/0/6eefcfee0e131523a876d54d686fbf-2477.manifest
/home/travis/.ccache/0/1/4611c90ae215cd5a4ca105b36abb02-241286.d
/home/travis/.ccache/0/1/4611c90ae215cd5a4ca105b36abb02-241286.o
/home/travis/.ccache/0/1/7428fdbae0234930a9843d341ad62b-1788.manifest
/home/travis/.ccache/0/1/ac5004cc7dd2b336c599dfdd83b6ab-2465.manifest
/home/travis/.ccache/0/1/e65945ca036fbfff412c03a160f75f-606211.manifest
/home/travis/.ccache/0/1/f8809d693f59ae4805d095329484c1-101124.manifest
/home/travis/.ccache/0/2/6ec93c4491772bb7c5bbdac7c8f222-91357.manifest
/home/travis/.ccache/0/2/a631b65509797fd5ad684b7f8b5966-8073.manifest
/home/travis/.ccache/0/2/c4698c16b8b94b35cbf9c2adaafac6-3154.manifest
/home/travis/.ccache/0/3/6ffd4536039ae31f4b2e9a514dea56-10552.manifest
/home/travis/.ccache/0/3/d32ae5b574012e296fcc2eb3a0f9db-11517.manifest
/home/travis/.ccache/0/3
...
changes detected, packing new archive
uploading archive

after_success.1
0.01s$ .travis/send_binaries.sh
Skipping package upload for command 'test'

after_success.2
0.78s$ test -z "PACKAGECLOUD_USER" || gem install package_cloud
Successfully installed package_cloud-0.3.00
1 gem installed

after_success.3
0.01s$ test -z "PACKAGECLOUD_USER" || scripts/build_packagecloud_upload
No packagecloud upload:  CMD 'test' != 'deb'

after_success.4
0.01s$ ccache -s
cache directory                     /home/travis/.ccache
cache hit (direct)                    13
cache hit (preprocessed)             707
cache miss                            15
called for link                      383
called for preprocessing             309
preprocessor error                     1
bad compiler arguments                 2
autoconf compile/link                 31
unsupported compiler option          379
no input file                        787
files in cache                     11142
cache size                         290.7 Mbytes
max cache size                     512.0 Mbytes
after_script

0.01s$ .travis/send_status.sh
sshpass: invalid option -- 'P'
sshpass: invalid option -- 'o'
Usage: sshpass [-f|-d|-p|-e] [-hV] command parameters
   -f filename   Take password to use from file
   -d number     Use number as file descriptor for getting password
   -p password   Provide password as argument (security unwise)
   -e            Password is passed as env-var "SSHPASS"
   With no parameters - password will be taken from stdin
   -h            Show help (this screen)
   -V            Print version information
At most one of -f, -d, -p or -e should be used

Done. Your build exited with 0.
```

ArcEye, 2017-10-21:
This should not have been closed, especially when the last entry is highlighting an error

ArcEye, 2017-10-21:
@zultron 

One minor glitch
Looking at the full package builds, the `*_{arch}.buildinfo` and `*_{arch}.changes` files are being
given the default `0.1.1-1` serial, instead of the generated 'git commit No.gitSHA' one

ArcEye, 2017-10-21:
Found the problem with the uploads
```
0.01s$ .travis/send_binaries.sh
tar: deploy: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
```

Somehow the change to the logic in this file has affected its being able to see the directory `deploy`
That dir is still being created in build-dep.sh, so hard to see the problem

Reverting that file and testing

ArcEye, 2017-10-21:
Having stopped at the Travis script error, switched again to the Jenkins builder.

The initial problem with the PR test under Jenkins was because @zultron is no longer a member of the project and as such automatically 'whitelisted'.

After banging my head on the wall trying to figure out why the webhook was not activating, turned out the comment attributed to me, that I knew nothing about, was the githubPRplugin guarding against malicious PRs by waiting for an 'ok to test' message from a project member with push rights to the repo.

Now successfully built a PR and merged for full Jenkins package build and upload, bypassing Travis.

Will then find out if the ionotify process is working, it did look to be alive yesterday
but, `ps -A | grep ionoticoming` gets no result now.

ArcEye, 2017-10-22:
Have now hopefully fixed the server side upload.
There was an environment setting file missing, which meant all the attempts to incorporate uploads
into the repo failed and they just stagnated there.
The ionotfy based system was working, but the script it ran failed due to the above.

The last full test build succeeded from PR to Merge.
https://github.com/machinekit/machinekit/pull/1299

There was a glitch with package versions not matching between main machinekit package and flavour packages, which has been fixed manually.

Think I have found the cause of this in the server script that processes the packages.
It caused the main `machinekit_*` packages to be deleted from the upload, if xenomai | armhf packages were detected in the same dir.
Not sure what the logic of that was, but with these bulk uploads we need to process everything.

Making another inoccuous PR to test the whole system again before long

ArcEye, 2017-10-22:
#1300 worked perfectly
All packages match across the distro type

Only thing remaining is to get `dpkg-buildsources` to use the correct changelog file
as per https://github.com/machinekit/machinekit/issues/1228#issuecomment-338383239 above

[10 more replies]

commit 74201463 (2017-10-20): Point Travis at new Docker build scripts

Fixes #1228