machinekit/machinekit #1232: Passing compiler and linker flags to comp

machinekoder, 2017-07-07:
Is it possible to pass additional linker or compiler flags to comp? I would like to link my comp against the `libmachinetalk-npb.so` to make use of the ring buffers.

ArcEye, 2017-07-07:
Yes, for userspace at least.
https://github.com/machinekit/machinekit/blob/master/src/hal/utils/comp.g#L658

sirop, 2017-07-07:
For one of my componenets I use :
```
export C_INCLUDE_PATH=/usr/local/include/openrave-0.9
EXTRA_LDFLAGS=" -lopenrave0.9_c -lopenrave0.9-core_c /opt/OpenBLAS/lib/libopenblas.so  -Wl,-rpath,/usr/local/lib:/opt/OpenBLAS/lib -lm" comp --install jacobian.comp
```

ArcEye, 2017-07-07:
@sirop just beat me :)

The other ways of course are to specify on the command line
eg.
`$ EXTRA_LDFLAGS="-lmachinetalk-npb" comp --compile thecompname.comp `

or produce a C file first, then specify the linkage in the gcc command
eg.
```
$ comp thecompname.comp
$ gcc -L/usr/lib/machinekit -lmachinetalk-npb -o thecompname thecompname.c
```
or similar

sirop, 2017-07-07:
@machinekoder 
https://github.com/machinekit/machinekit/blob/master/src/machinetalk/msgcomponents/ringread.comp
did not need any extra flags or libs being passed, but I have not updated MK src for a long time.

machinekoder, 2017-07-14:
The problem has been fixed with the recent package changes. So I will close the issue.

closed 2017-07-14 by machinekoder

ArcEye, 2017-07-14:
Can I take it this means you have tried using (inst)comp on armhf to build components and it works OK?

No-one has actually fed back that it works yet.