machinekit/machinekit #1238: support 4.x kernels on the BBB

luminize, 2017-07-26:
After some off-line discussion these issues (add if we find more) are about adapting the
PRU driver to deal with the new sysfs exposed by the 4.x kernels
- [pru_generic.bin not deployed for rt-preempt FLAVOR](https://github.com/machinekit/machinekit/issues/1182)
- [Status of the machinekit-rt-preempt Debian package on armhf](https://github.com/machinekit/machinekit/issues/928)

RobertCNelson, 2017-07-31:
@luminize i think we got everything patched..

machinekit:

https://github.com/machinekit/machinekit/pull/1241

revert: https://github.com/machinekit/machinekit/commit/598b4c73691d8535633186d7e58d17ac169547af

beaglebone, make sure you using u-boot overlays:

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

config-pin: needed this patch:  (sudo apt update ; sudo apt upgrade) will pick it up..

https://github.com/beagleboard/bb.org-overlays/commit/40b386724c19bbff3e7c0295d183544814eb4f95

Regards,

RobertCNelson, 2017-07-31:
here's teh email thread:

https://groups.google.com/forum/#!topic/machinekit/RrNLUo4ASP4

while he tested: 4.4.68-ti-rt-r115, 4.9.40-bone-rt-r7 should be more optimized..

Regards,

luminize, 2017-08-01:
Thanks @darenschwenke @RobertCNelson @cdsteinkuehler

closed 2017-08-01 by luminize