machinekit/machinekit #1239: Axis UI incremental mode does not select/change with 'i' key

propcoder, 2017-07-30:
Error messages appear when pressing 'i' on keyboard instead of changing to incremental jog mode or selecting increment:
```
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1535, in __call__
    return self.func(*args)
  File "/usr/bin/axis", line 2855, in <lambda>
    root_window.bind("i", lambda event: jogspeed_incremental())
  File "/usr/bin/axis", line 1375, in jogspeed_incremental
    cursel = int(cursel)
TypeError: int() argument must be a string or a number, not 'tuple'

```

ArcEye, 2017-07-31:
Do you know if this has ever worked?

Nothing has changed in Axis for 10 months and it is not a shortcut key I have ever used.

ArcEye, 2017-07-31:
Reminded myself why I hate Python and its lack of variable declarations.
Don't see how this variable is a tuple, since it only ever has a size of 1.

Fixed by specifically referencing the base index of the tuple.
Don't really see how this ever worked, maybe no-one used it or a change in python made this variable a tuple instead of a string?

Whatever, it works now.

PR #1240 (2017-07-31): Fix jog increment hot key in Axis
Pressing 'I' triggers an error because a tuple is used as an
argument to int()

Solution is simply to specify index of [0]

Can't see how this ever worked, maybe no-one used it.

Fixes #1239
closed 2017-07-31 by ArcEye