machinekit/machinekit #1245: First attempt at installing machine kit

flykks, 2017-08-13:
Install on BBB for learning purposes. All installation from packages as explained at:
http://www.machinekit.io/docs/getting-started/installing-packages/

uname -a
Linux beaglebone 3.8.13-xenomai-r84 #1 Sun Feb 12 04:20:15 UTC 2017 armv7l GNU/Linux

Debian image is 7.11 from beagleboard.org

Installation seemed to run OK, then run machinekit and get the following linuxcnc_print.txt

RUN_IN_PLACE=no
LINUXCNC_DIR=
LINUXCNC_BIN_DIR=/usr/bin
LINUXCNC_TCL_DIR=/usr/lib/tcltk/linuxcnc
LINUXCNC_SCRIPT_DIR=
LINUXCNC_RTLIB_DIR=/usr/lib/linuxcnc
LINUXCNC_CONFIG_DIR=
LINUXCNC_LANG_DIR=/usr/share/linuxcnc/tcl/msgs
INIVAR=/usr/libexec/linuxcnc/inivar
HALCMD=halcmd
LINUXCNC_EMCSH=/usr/bin/wish8.5
INIFILE=/home/machinekit/machinekit/configs/sim.axis/gantry.ini
PARAMETER_FILE=sim.var
TASK=milltask
HALUI=
DISPLAY=axis
Starting Machinekit server program: linuxcncsvr
Loading Real Time OS, RTAPI, and HAL_LIB modules
Starting Machinekit IO program: io
Starting TASK program: milltask
Starting DISPLAY program: axis
Killing task linuxcncsvr, PID=4632
Killing task milltask, PID=4696
Removing HAL_LIB, RTAPI, and Real Time OS modules
Removing NML shared memory segments

Following is the linuxcnc_debug.txt

Not sure where to go from here.  Any help will be greatly appreciated.

ArcEye, 2017-08-13:
Please take this to the forum, 
https://groups.google.com/forum/#!forum/machinekit
the issues trackers are for bugs in the code, not users problems in getting started.

Make sure you read this before you do.
http://www.machinekit.io/docs/getting-help/
It would be impossible to assist on the amount of info here.

regards

closed 2017-08-13 by ArcEye

flykks, 2017-08-13:
Thanks for your reply. I read the getting-help (highlighted) and I posted to the link from “please report an issue”… I am sorry if I misunderstood.  Needless to say, I am new to this, and not sure how to go about it…  But very interested to get it going…

 

Should I use “then ask”… ?

 

Thank you 

 

https://groups.google.com/forum/#!forum/machinekit

http://www.machinekit.io/docs/getting-help/