machinekit/machinekit #1246: Missing dependency (or recommends) on yapps2-runtime

duelafn, 2017-08-16:
Running on Debian 8 (jessie) on BeagleBone:

    [machinekit@dlp ~]$ instcomp --compile utgmist.icomp 
    Traceback (most recent call last):
      File "/usr/bin/instcomp", line 32, in <module>
        from yapps import runtime
    ImportError: No module named yapps

    [root@dlp ~]# apt-get install yapps2-runtime

    [machinekit@dlp ~]$ instcomp --compile utgmist.icomp 
    Compiling realtime utgmist.c
    Linking utgmist.so

duelafn, 2017-08-16:
Forgot to mention,

ArcEye, 2017-08-17:
Can you close one of the two issues and clarify exactly what is happening please.

In between the two of them and all the now replaced edits, I am confused as to whether you are saying that the package needs to depend on yapps, or that it does and this conflicts with yapps2 (which you are installing for what reason?)

ArcEye, 2017-08-17:
OK, I see what happened.
It was omitted when machinekit-dev was deprecated to get over the flavour problem in armhf machinekit-dev packages.

The above PR should fix it

Thanks for the report.

PR #1248 (2017-08-17): Restore yapps dependency to machinekit-flavour packages.
Left out when comp etc. moved and machinekit-dev deprecated

Fixes #1246
closed 2017-08-17 by ArcEye