machinekit/machinekit #1247: machinekit package conflicts with yapps2

duelafn, 2017-08-16:
The machinekit package seems to have an inappropriate copy of the yapps executable:

    [root@dlp ~]# apt-get install yapps2
    ...
    Unpacking yapps2 (2.1.1-17.2) ...
    dpkg: error processing archive /var/cache/apt/archives/yapps2_2.1.1-17.2_all.deb (--unpack):
     trying to overwrite '/usr/bin/yapps', which is also in package machinekit 0.1.1502215133-1mk.travis.master.git4f7690d9~1jessie
    Errors were encountered while processing:
     /var/cache/apt/archives/yapps2_2.1.1-17.2_all.deb
    E: Sub-process /usr/bin/dpkg returned an error code (1)

    [root@dlp ~]# dpkg -L machinekit | grep yapps
    /usr/bin/yapps

It seems to match the standard script aside from some development artifacts.

    [root@dlp ~]# cp /usr/bin/yapps yapps-mkit
    [root@dlp ~]# apt-get -o Dpkg::Options::="--force-overwrite" install yapps2
    [root@dlp ~]# diff /usr/bin/yapps yapps-mkit
    14c14,16
    < import sys, re
    ---
    > import sys, os, re
    > BASE = os.path.abspath(os.path.join(os.path.dirname(sys.argv[0]), ".."))
    > sys.path.insert(0, os.path.join(BASE, "lib", "python"))

This is Debian 9 (jessie) on BeagleBone (machinekit 0.1.1502215133-1mk.travis.master.git4f7690d9~1jessie).

ArcEye, 2017-08-17:
This is the one I don't understand, treating the 2 issues as separate.

Machinekit does not depend upon yapps and it had to be specifically added for machinekit-dev, to cater for comp which uses it
```
Package: machinekit
Version: 0.1.1502215133-1mk.travis.master.git4f7690d9~1wheezy
Architecture: amd64
Maintainer: John Morris <john@dovetail-automata.com>
Installed-Size: 28871
Depends: libatk1.0-0 (>= 1.12.4), libavahi-client3 (>= 0.6.16), libavahi-common3 (>= 0.6.16), 
    libboost-python1.49.0 (>= 1.49.0-1), libboost-serialization1.49.0 (>= 1.49.0-1), libboost-thread1.49.0 (>= 1.49.0-1)
Breaks: linuxcnc
Replaces: linuxcnc
```

However instcomp, which is in machinekit,  should require yapps too by that logic.
Is it just that no-one using packages has ever used instcomp?

ArcEye, 2017-08-17:
OK now sorted what is happening.

The yapps concerned does not come from a package, but from sources
https://github.com/machinekit/machinekit/tree/master/lib/python/yapps

Whilst `runtime.py` is present, the yapps executable script does not contain it and requires the module `runtime`, which in a package is not present.

In a RIP build it is at `/lib/python/yapps/runtime.pyc` and no yapps packages are required to be installed.

Looks like linuxcnc (because that is where it comes back from) got around this by depending on `yapps2-runtime`, when installing the `-dev` package which contains `comp`.

This in itself is strange, because the whole reason for including the yapps code inline, was given as being a hedge against the correct version of yapps2 not being available

I have tested on a bare jessie install and confirm the error regards the absence of module `runtime` when running both comp/instcomp.

There is however no clash between the installed binary `/usr/bin/yapps` and the package
`yapps2-runtime`.
This is because the package does not contain a binary, just the `runtime.py` script.

I installed the package and comp/instcomp work as they should.

The PR #1248 now merged should fix it OK

Thanks again for the reports.
I seldom if ever use packages and this particular 'quirk' had passed me by

closed 2017-08-17 by ArcEye

duelafn, 2017-08-17:
Sorry for being unclear, but this issue remains.

There are two packages related to yapps: `yapps2` and `yapps2-runtime`. The yapps2 package is the actual parser generator, and yapps2-runtime is needed by scripts which are generated by yapps2. (So, for instance, if machinekit didn't bundle its own copy, `yapps2` would be needed to build comp and instcomp and `yapps2-runtime` would be needed to run comp and instcomp.)

The runtime package does not include `/usr/bin/yapps` but the `yapps2` package does. Since the `machinekit` package also installs `/usr/bin/yapps`, it conflicts with the `yapps2` package (this issue), but requires the `yapps2-runtime` package (issue #1246 which should now be fixed).

The conflict on `/usr/bin/yapps` does not impact the normal installation of machinekit since the `yapps2` package won't be installed by default (only the runtime is). However, if someone were using yapps for their own purposes, they might want to install the `yapps2` package in addition to `machinekit`. In that case, they would have a conflict.

The most pure solution to this issue would be to completely remove the bundled `lib/python/yapps` from the machinekit repository and add the `yapps2` package as a build dependency. That, of course, will mess with the build system and would require developers to install yapps separately (and I have no idea how stable yapps is to know whether that is a good idea or not).

If you don't want to unbundle yapps, it should be enough to prevent the yapps generator script from being included in the `machinekit` package. It is likely that its inclusion is an accident anyway.

So far as I can tell, the `yapps` script is being created by [src/hal/utils/Submakefile](https://github.com/machinekit/machinekit/blob/master/src/hal/utils/Submakefile#L117). Since the makefile creates the script in `bin`, the yapps script is being automatically picked up and installed by the debian package.

I'm not very familiar with actually creating Debian packages, but I imagine that the yapps script could be excluded by creating it in some other location or by explicitly excluding it. It looks like it could be excluded by adding `-Xusr/bin/yapps` to the `dh_install` line in [debian/rules.in](https://github.com/machinekit/machinekit/blob/master/debian/rules.in#L196).

Thanks for looking at these, and let me know if I'm still a bit unclear or accidentally mixed up a package name (I triple checked them, but could still have slipped).

ArcEye, 2017-08-17:
> The conflict on /usr/bin/yapps does not impact the normal installation of machinekit since the yapps2 package won't be installed by default (only the runtime is). However, if someone were using [...]

That is the real issue, it is a corner case which has never previously occurred AFAIK and may never again.  
Most people using packages would never have heard of yapps, much less be using it for something else.

That said the yapps2 package version of yapps and the MK one are binary identical, except for 2 lines of code in the MK one, which sets the base path, doubtless intended for RIP builds.

So the whole code section which is unchanged in 11 years, could be dispensed with and yapps2 plus yapps2-runtime added to the build and package requirements.

I'll have a think on it.

ArcEye, 2017-08-18:
After thinking and a quick consultation, I think this is going in the 'Won't Fix' box.

Debian packaging pedantry might dictate otherwise, but the code was included for very good reasons, to prevent someone 'improving' it and breaking the core engine for module production.

If you have installed machinekit, you don't need to install the yapps2 package, because you already have yapps installed.

If someone absolutely must have all the documentation bumph that makes up the rest of the package, instead of searching on-line, the answer is just delete /usr/bin/yapps and the package will install and machinekit will still work (until someone 'improves' yapps).

Appreciate you may not like it, but pragmatism wins on this one.