machinekit/machinekit #1250: Float support for sserial

saeugetier, 2017-08-22:
I'm using the sserial protocol on a DE0Nano Soc with mksocfpga. The sserial ports are connected to stmbl. Due lack of float number support the signals pos_cmd and pos_fb are not available. 

LinuxCNC had some commits related to that topic.
https://github.com/LinuxCNC/linuxcnc/commit/2a257d58cce8dcc16a8bdb793491b9bd4338b1c3
https://github.com/LinuxCNC/linuxcnc/commit/9207cbca59c61a32cf33a25f3c37d0ba0d4e1dbb

Should I do a backport to machinekit or would someone from the main developers take care of it?

ArcEye, 2017-08-22:
Feel free to do a PR.

The main thing is that it is done by someone who can test it in use to ensure that it does what it should.

saeugetier, 2017-08-22:
Done: https://github.com/machinekit/machinekit/pull/1251

closed 2017-08-23 by saeugetier