machinekit/machinekit #1254: Invalid read/write of size 8, address is 0 bytes inside a block of size 4 alloc'd

rubienr, 2017-08-27:
* Reason: 

Mistakenly allocated **sizeof(pointer_type)** instead **sizeof(type)**, thus reads/writes of i.e. float which is much latger than poitner type exceed alloce'd block size.

* Where: 

 [user_comps/xhc-hb04.cc:L471](https://github.com/machinekit/machinekit/blame/master/src/hal/user_comps/xhc-hb04.cc#L471)
[user_comps/xhc-hb04.cc:L487](https://github.com/machinekit/machinekit/blame/master/src/hal/user_comps/xhc-hb04.cc#L487)
 [user_comps/xhc-hb04.cc:L503](https://github.com/machinekit/machinekit/blame/master/src/hal/user_comps/xhc-hb04.cc#L503)

* Fix: 

The  [line  471](https://github.com/machinekit/machinekit/blame/master/src/hal/user_comps/xhc-hb04.cc#L471) should write
`return hal_pin_simu(pin_name, ( void**)data_ptr_addr, sizeof(hal_float_t));`
instead of 
`return hal_pin_simu(pin_name, ( void**)data_ptr_addr, sizeof(*data_ptr_addr));`
Subsequent lines must be fixed respectively.

br, rr

ArcEye, 2017-08-27:
Could you submit a PR please.
You may want to do the same to linuxcnc, the code comes directly from there.

rubienr, 2017-08-27:
I'll try my best since I am busy with implementing for the WHB04B-6. For that reason I took the xhc-hb04.cc as baseline and found several issues.