machinekit/machinekit #1255: xhc-hb04: Memleak - libusb_alloc_transfer(0)

rubienr, 2017-08-27:
* Reason: 

Any libusb_transfer struct is requested from libusb **but never released gracefully**.

* Where:

[xhc-hb04.cc:L443](https://github.com/machinekit/machinekit/blame/master/src/hal/user_comps/xhc-hb04.cc#L443)

```
void setup_asynch_transfer(libusb_device_handle *dev_handle)
442	{
443		transfer_in  = libusb_alloc_transfer(0);
444		libusb_fill_bulk_transfer( transfer_in, dev_handle, (0x1 | LIBUSB_ENDPOINT_IN),
445			in_buf, sizeof(in_buf),
446			cb_response_in, NULL, 0); // no user data
447		libusb_submit_transfer(transfer_in);
448	}
```

* HotFix:
```
void setup_asynch_transfer(libusb_device_handle *dev_handle)
442	{
                if (transfer_in != nullptr) 
                {
   			libusb_free_transfer(transfer_in);
   			transfer_in = nullptr;
                }
443		transfer_in  = libusb_alloc_transfer(0);
444		libusb_fill_bulk_transfer( transfer_in, dev_handle, (0x1 | LIBUSB_ENDPOINT_IN),
445			in_buf, sizeof(in_buf),
446			cb_response_in, NULL, 0); // no user data
447		libusb_submit_transfer(transfer_in);
448	}
```

ArcEye, 2017-08-27:
Could you submit a PR please.
You may want to do the same to linuxcnc, the code comes directly from there.

rubienr, 2017-08-27:
Do you wish having one PR per issue or one PR combining all xhc-hb04 issues?

ArcEye, 2017-08-27:
A single PR addressing both issues is fine thanks