machinekit/machinekit #1259: DE0_Nano_SoC_Cramps.3x24.dtbo errors with pb_decode(Firmware) failed: 'string overflow'

luminize, 2017-08-28:
The DE0_Nano_SoC_Cramps.3x24.dtbo errors when trying to load. The DB25 4x7I76 dtbo works, so I'm deducting the error is in the dtbo file.

Taken from the log below:
```
Aug 28 15:07:35 mksocfpga rtapi:0: 1:rtapi_app:1826:user hm2/hm2_5#01124.0: pb_decode(Firmware) failed: 'string overflow'
Aug 28 15:07:35 mksocfpga rtapi:0: 1:rtapi_app:1826:user hm2/hm2_5#01124.0: error reading fwid proto msg
Aug 28 15:07:35 mksocfpga rtapi:0: 1:rtapi_app:1826:user hm2_5#01124.0: hm2_soc_ol_board fails HM2 registration
Aug 28 15:07:35 mksocfpga rtapi:0: 1:rtapi_app:1826:user hm2_soc_ol: error registering UIO driver: -22
Aug 28 15:07:35 mksocfpga rtapi:0: hal_call_usrfunct(newinst,-- config="firmware=socfpga/dtbo/DE0_Nano_SoC_Cramps.3x24.dtbo num_stepgens=5" debug=1) failed: -1 - Operation not permitted
Aug 28 15:07:35 mksocfpga rtapi:0: 1:rtapi_app:1826:user hal_call_usrfunct(newinst,-- config="firmware=socfpga/dtbo/DE0_Nano_SoC_Cramps.3x24.dtbo num_stepgens=5" debug=1) failed: -1 - Operation not permitted
```

```
machinekit@mksocfpga:~$ export DEBUG=5
machinekit@mksocfpga:~$ halrun -I
msgd:0 stopped
rtapi:0 stopped
halcmd: loadrt hostmot2
<stdin>:1: Realtime module 'hostmot2' loaded
halcmd: newinst hm2_soc_ol hm2-socfpga0 -- config="firmware=socfpga/dtbo/DE0_Nano_SoC_Cramps.3x24.dtbo num_stepgens=5" debug=1
<stdin>:2: Realtime module 'hm2_soc_ol' loaded
<stdin>:2: rc=-1: hal_call_usrfunct(newinst,-- config="firmware=socfpga/dtbo/DE0_Nano_SoC_Cramps.3x24.dtbo num_stepgens=5" debug=1) failed: -1 - Operation not permitted
halcmd: exit
<commandline>:0: Realtime threads stopped
machinekit@mksocfpga:~$

Aug 28 15:07:24 mksocfpga msgd:0: startup pid=1820 flavor=rt-preempt rtlevel=5 usrlevel=5 halsize=524288 shm=Posix cc=gcc 4.9.2  version=unknown
Aug 28 15:07:24 mksocfpga msgd:0: ØMQ=4.0.5 czmq=3.0.2 protobuf=2.6.1 atomics=gcc intrinsics    libwebsockets=<no version symbol>
Aug 28 15:07:24 mksocfpga msgd:0: configured: sha=680d20c
Aug 28 15:07:24 mksocfpga msgd:0: built:      Aug 28 2017 13:48:52 sha=680d20c
Aug 28 15:07:24 mksocfpga msgd:0: register_stuff: actual hostname as announced by avahi='mksocfpga.local'
Aug 28 15:07:24 mksocfpga msgd:0: zeroconf: registering: 'Log service on mksocfpga.local pid 1820'
Aug 28 15:07:24 mksocfpga rtapi:0: 2:rtapi_app:1826:user rtapi:0: cannot create core dumps - /proc/sys/fs/suid_dumpable contains 0
Aug 28 15:07:24 mksocfpga rtapi:0: 2:rtapi_app:1826:user you might have to run 'echo 1 > /proc/sys/fs/suid_dumpable' as root to enable rtapi_app core dumps
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user rtapi.so default iparms: ''
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user RTAPI:0  rt-preempt unknown init
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user rtapi: loaded from rtapi.so
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user hal_lib.so default iparms: ''
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user rtapi_app_main:195 HAL: initializing RT hal_lib support
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user halg_xinitfv:90 HAL: initializing component 'hal_lib' type=4 arg1=0 arg2=0/0x0
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user hal_heap_addmem:58 HAL: extending arena by 262144 bytes
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user halg_export_xfunctfv:85 HAL: exporting function 'newinst' type 2 fp=0 owner=66
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user halg_export_xfunctfv:85 HAL: exporting function 'delinst' type 2 fp=0 owner=66
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user halg_xinitfv:271 HAL: singleton component 'hal_lib' id=66 initialized
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user rtapi_app_main:199 HAL: RT hal_lib support initialized rc=66
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user hal_lib: loaded from hal_lib.so
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user accepting commands at ipc:///tmp/0.rtapi.a42c8c6b-4025-4f83-ba28-dad21114744a
Aug 28 15:07:24 mksocfpga rtapi:0: 3:rtapi_app:1826:user rtapi_app:0 ready flavor=rt-preempt gcc=4.9.2 git=unknown
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user pid=1826 flavor=rt-preempt gcc=4.9.2 git=unknown
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user pid=1826 flavor=rt-preempt gcc=4.9.2 git=unknown
Aug 28 15:07:24 mksocfpga msgd:0: ulapi:1827:user _ulapi_init(): ulapi rt-preempt unknown loaded
Aug 28 15:07:24 mksocfpga msgd:0: ulapi:1827:user halg_xinitfv:271 HAL: singleton component 'hal_lib1827' id=70 initialized
Aug 28 15:07:24 mksocfpga msgd:0: hal_lib:1827:user --halcmd ping
Aug 28 15:07:24 mksocfpga msgd:0: hal_lib:1827:user halg_exit:293 HAL: removing component 72 'halcmd1827'
Aug 28 15:07:24 mksocfpga msgd:0: hal_lib:1827:user ulapi_hal_lib_cleanup:235 HAL: lib_module_id=70
Aug 28 15:07:24 mksocfpga msgd:0: hal_lib:1827:user halg_exit:293 HAL: removing component 70 'hal_lib1827'
Aug 28 15:07:24 mksocfpga msgd:0: hal_lib:1827:user halg_exit:315 HAL: hal_errorcount()=0
Aug 28 15:07:24 mksocfpga msgd:0: hal_lib:1827:user halg_exit:316 HAL: _halerrno=0
Aug 28 15:07:24 mksocfpga rtapi:0: 4:rtapi_app:1826:user pid=1826 flavor=rt-preempt gcc=4.9.2 git=unknown
Aug 28 15:07:24 mksocfpga msgd:0: ulapi:1832:user _ulapi_init(): ulapi rt-preempt unknown loaded
Aug 28 15:07:24 mksocfpga msgd:0: ulapi:1832:user halg_xinitfv:271 HAL: singleton component 'hal_lib1832' id=74 initialized
Aug 28 15:07:24 mksocfpga msgd:0: hal_lib:1832:user --halcmd -kf
Aug 28 15:07:25 mksocfpga msgd:0: zeroconf: registered 'Log service on mksocfpga.local pid 1820' _machinekit._tcp 0 TXT "uuid=a42c8c6b-4025-4f83-ba28-dad21114744a" "instance=9914d820-8c02-11e7-a98a-bad04a9c4ece" "service=log" "dsn=ipc:///tmp/0.log.a42c8c6b-4025-4f83-ba28-dad21114744a"
Aug 28 15:07:28 mksocfpga rtapi:0: 4:rtapi_app:1826:user hostmot2.so default iparms: ''
Aug 28 15:07:28 mksocfpga rtapi:0: 4:rtapi_app:1826:user halg_xinitfv:90 HAL: initializing component 'hostmot2' type=1 arg1=0 arg2=0/0x0
Aug 28 15:07:28 mksocfpga rtapi:0: 4:rtapi_app:1826:user hostmot2: loaded from hostmot2.so
Aug 28 15:07:35 mksocfpga rtapi:0: 4:rtapi_app:1826:user hm2_soc_ol.so default iparms: 'debug=0 no_init_llio=0 num=0'
Aug 28 15:07:35 mksocfpga rtapi:0: 4:rtapi_app:1826:user halg_xinitfv:90 HAL: initializing component 'hm2_soc_ol' type=1 arg1=0 arg2=0/0x0
Aug 28 15:07:35 mksocfpga rtapi:0: 4:rtapi_app:1826:user hm2_soc_ol: loaded from hm2_soc_ol.so
Aug 28 15:07:35 mksocfpga rtapi:0: 4:rtapi_app:1826:user do_newinst_cmd: instargs='-- config="firmware=socfpga/dtbo/DE0_Nano_SoC_Cramps.3x24.dtbo num_stepgens=5" debug=1'
[... 76 lines cut ...]
Aug 28 15:07:46 mksocfpga rtapi:0: 4:rtapi_app:1826:user report_memory_usage:168 HAL:   strings on global heap: alloc=236 freed=67 balance=169
Aug 28 15:07:46 mksocfpga rtapi:0: 4:rtapi_app:1826:user report_memory_usage:175 HAL:   RT objects: 233  alignment loss: 6  (2%)
Aug 28 15:07:46 mksocfpga rtapi:0: 4:rtapi_app:1826:user report_memory_usage:177 HAL:   hal_malloc():   3
Aug 28 15:07:46 mksocfpga rtapi:0: 4:rtapi_app:1826:user report_memory_usage:179 HAL:   unused:   261655
Aug 28 15:07:46 mksocfpga rtapi:0: 4:rtapi_app:1826:user halg_exit:320 HAL: hal_sweep: 15 objects freed
Aug 28 15:07:46 mksocfpga rtapi:0: 4:rtapi_app:1826:user rtapi_app_exit:213 HAL: RT hal_lib support removed successfully
Aug 28 15:07:46 mksocfpga rtapi:0: 4:rtapi_app:1826:user  'hal_lib' unloaded
Aug 28 15:07:46 mksocfpga rtapi:0: 4:rtapi_app:1826:user RTAPI:0 exit
Aug 28 15:07:46 mksocfpga rtapi:0: 4:rtapi_app:1826:user  'rtapi' unloaded
Aug 28 15:07:46 mksocfpga rtapi:0: 3:rtapi_app:1826:user exiting mainloop (by remote command)
Aug 28 15:07:46 mksocfpga rtapi:0: 2:rtapi_app:1826:user rtapi_app_main 1826: 1 page faults, 4478 page reclaims
Aug 28 15:07:46 mksocfpga msgd:0: ulapi:1871:user _ulapi_init(): ulapi rt-preempt unknown loaded
Aug 28 15:07:46 mksocfpga msgd:0: ulapi:1871:user halg_xinitfv:271 HAL: singleton component 'hal_lib1871' id=88 initialized
Aug 28 15:07:46 mksocfpga msgd:0: hal_lib:1871:user --halcmd shutdown
Aug 28 15:07:46 mksocfpga msgd:0: hal_lib:1871:user halg_exit:293 HAL: removing component 90 'halcmd1871'
Aug 28 15:07:46 mksocfpga msgd:0: hal_lib:1871:user ulapi_hal_lib_cleanup:235 HAL: lib_module_id=88
Aug 28 15:07:46 mksocfpga msgd:0: hal_lib:1871:user halg_exit:293 HAL: removing component 88 'hal_lib1871'
Aug 28 15:07:46 mksocfpga msgd:0: hal_lib:1871:user halg_exit:315 HAL: hal_errorcount()=0
Aug 28 15:07:46 mksocfpga msgd:0: hal_lib:1871:user halg_exit:316 HAL: _halerrno=0
Aug 28 15:07:46 mksocfpga msgd:0: rtapi_app exit detected - scheduled shutdown
Aug 28 15:07:48 mksocfpga msgd:0: msgd shutting down
Aug 28 15:07:48 mksocfpga msgd:0: zeroconf: unregistering 'Log service on mksocfpga.local pid 1820'
Aug 28 15:07:48 mksocfpga msgd:0: log buffer hwm: 0% (14 msgs, 1135 bytes out of 524288)
Aug 28 15:07:48 mksocfpga msgd:0: normal shutdown - global segment detached
```

the-snowwhite, 2017-08-31:
Yes I have had similar problems which led me to suspect that  the cause could be that the fpga config(s) needed longer time to load and settle So i increased the delay here:
https://github.com/machinekit/machinekit/pull/1263/files#diff-f22babbc6525d142cde6401c07757e2eL336

Which enables both the DE10 and the DE0 versions to run.

the-snowwhite, 2017-08-31:
Bootlog here:

[DE0_Cramps_3x24-good-boot.txt](https://github.com/machinekit/machinekit/files/1267692/DE0_Cramps_3x24-good-boot.txt)

the-snowwhite, 2017-08-31:
After testing I reinstalled the current Machinekit packages and the load got to here before failing:
Aug 31 17:40:34 mksocfpga-nano-soc msgd:0: hal_lib:2027:user hal_add_funct_to_thread:214 HAL: adding function 'hm2_5i25.0.read' to thread 'servo-thread'
Aug 31 17:40:34 mksocfpga-nano-soc msgd:0: hal_lib:2027:user hal_add_funct_to_thread:230 HAL error: function 'hm2_5i25.0.read' not found
Aug 31 17:40:34 mksocfpga-nano-soc msgd:0: hal_lib:2027:user halg_exit:293 HAL: removing component 288 'machinekit::hal2027'
Aug 31 17:40:34 mksocfpga-nano-soc msgd:0: hal_lib:2027:user ulapi_hal_lib_cleanup:235 HAL: lib_module_id=286
Aug 31 17:40:34 mksocfpga-nano-soc msgd:0: hal_lib:2027:user halg_exit:293 HAL: removing component 286 'hal_lib2027'
Aug 31 17:40:34 mksocfpga-nano-soc msgd:0: hal_lib:2027:user halg_exit:315 HAL: hal_errorcount()=1
Aug 31 17:40:34 mksocfpga-nano-soc msgd:0: hal_lib:2027:user halg_exit:316 HAL: _halerrno=-22
Aug 31 17:40:35 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user pid=2009 flavor=rt-preempt gcc=4.9.2 git=v0.1~-detached~2b90132
Aug 31 17:40:35 mksocfpga-nano-soc msgd:0: ulapi:2060:user _ulapi_init(): ulapi rt-preempt v0.1~-detached~2b90132 loaded
Aug 31 17:40:35 mksocfpga-nano-soc msgd:0: ulapi:2060:user halg_xinitfv:271 HAL: singleton component 'hal_lib2060' id=908 initialized
Aug 31 17:40:35 mksocfpga-nano-soc msgd:0: hal_lib:2060:user --halcmd stop
[DE0_Cramps_3x24-fail-boot.txt](https://github.com/machinekit/machinekit/files/1267705/DE0_Cramps_3x24-fail-boot.txt)

luminize, 2017-08-31:
> On 31 Aug 2017, at 17:50, Michael Brown wrote:
> 
> Yes I have had similar problems which led me to suspect that the cause could be that the fpga config(s) needed longer time to load and settle So i increased the delay here:
> https://github.com/machinekit/machinekit/pull/1263/files#diff-f22babbc6525d142cde6401c07757e2eL336
> 
> Which enables both the DE10 and the DE0 versions to run.
> 
I'll cherry-pick that commit and try.

cdsteinkuehler, 2017-08-31:
On 8/31/2017 10:54 AM, Michael Brown wrote:
> After testing I reinstalled the current Machinekit packages and the load got to here before failing:
> Aug 31 17:40:34 mksocfpga-nano-soc msgd:0: hal_lib:2027:user hal_add_funct_to_thread:214 HAL: [...]

It may have gotten to there, but earlier on it looks like registering
the board with HM2 failed because the connector name was zero length.

Plus, it looks like the board is named hm2_5i24, and *NOT* hm2_5i25.

See the logfile starting with the parsing of the protobuf firmware ID
tag at line 309:

> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user hm2/hm2_5i24.0: protobuf raw size=156, aligned to 156
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user fwid msg at 0x42af0:00000000: 0a 07 64 63 61 63 61 61 62 12 0e 61 6c 74 65 72  ..dcacaab..alter
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user fwid msg at 0x42af0:00000010: 61 20 73 6f 63 66 70 67 61 1a 0f 0a 00 00 00 08  a socfpga.......
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user fwid msg at 0x42af0:00000020: 4f 30 2e 50 30 15 18 00 00 00 1a 0f 0a 08 47 50  O0.P0.........GP
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user fwid msg at 0x42af0:00000030: 49 4f 30 2e 50 31 15 18 00 00 00 1a 0f 0a 08 47  IO0.P1.........G
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user fwid msg at 0x42af0:00000040: 50 49 4f 30 2e 50 32 15 18 00 00 00 00 00 00 00  PIO0.P2.........
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user fwid msg at 0x42af0:00000050: 00 2a 10 54 65 72 61 73 69 63 20 44 45 30 2d 4e  .*.Terasic DE0-N
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user fwid msg at 0x42af0:00000060: 61 6e 6f 32 37 68 74 74 70 73 3a 2f 2f 6a 65 6e  ano27https://jen
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user fwid msg at 0x42af0:00000070: 6b 69 6e 73 2e 6d 61 63 68 69 6e 65 6b 69 74 2e  kins.machinekit.
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user fwid msg at 0x42af0:00000080: 69 6f 2f 6a 6f 62 2f 6d 6b 73 6f 63 66 70 67 61  io/job/mksocfpga
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user fwid msg at 0x42af0:00000090: 2d 71 75 61 72 74 75 73 2f 31 39 2f              -quartus/19/
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user hm2/hm2_5i24.0: build_sha = 'dcacaab'
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user hm2/hm2_5i24.0: fpga_part_number = 'altera socfpga'
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:2009:user hm2/hm2_5i24.0: connector 0 name = ''
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 1:rtapi_app:2009:user hm2: invalid llio ioport connector name 0 passed in (zero length)
> Aug 31 17:40:32 mksocfpga-nano-soc rtapi:0: 1:rtapi_app:2009:user hm2_5i24.0: hm2_soc_ol_board fails HM2 registration

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2017-08-31:
@cdsteinkuehler Yes failure in the boardname / cookie check are the fiorst symptoms.
I noticed the issue began on loading the socfpga/dtbo/DE0_Nano_SoC_Cramps.3x24.dtbo
config after I had removed the dts-overlay check so I reconed the faults were triggered by the shorter
delay after configuring the fpga.
Increasing the delay from 200xxx to 300xxx seems to remedy this issue.

the-snowwhite, 2018-04-07:
This Issue is most certantly related to the jessie release images with the v2016 u boot.
It seems to stem from the hw bridge handoff not doing things properly leading to flaky behavior and intermediary data corruption on the uio data bus.
The new stretch sd-images with 2018.01 uboot seem to have a much more solid and robust behavior without any flaky'ness so far.....

the-snowwhite, 2018-04-07:
https://github.com/the-snowwhite/soc-image-buildscripts/releases/tag/Release_2.0