machinekit/machinekit #1261: HM2_soc_ol device tree overlay loaded check prevents configuring fpga @ boot

the-snowwhite, 2017-08-30:
To be able to use the alt_vip framebuffer the fpga need to be configured via u-boot before booting the kernel. This will also load the uio0 hm2_soc_io driver.
Trying to reconfigure the fpga by loading the dtbo from machinekit will invalidate the framebuffer and give a blank screen.
trying to run machinekit without loading the dtbo gives an error message due to the device tree not loaded.
following quick fix remedies the issue for the DE10 Nano however the DE0_Nano_SoC then has issues with wrong signature (possible due to not having enough time to configure the fpga).

https://github.com/the-snowwhite/machinekit/commit/8b6694170299cc0d8daf53389fae8f2c8191a743

the-snowwhite, 2017-08-30:
I have just tested this to work as expected on both the DE10 and DE0 Nano SoC's
https://github.com/the-snowwhite/machinekit/commit/bb33c62b552eead43d5e4c6affb0e8dbc8e4bffc

the-snowwhite, 2017-08-31:
https://github.com/machinekit/machinekit/issues/1259

cdsteinkuehler, 2018-01-02:
Another option if the FPGA is programmed by u-Boot is to just not have the device-tree overlay configure the FPGA.  The remaining details (interrupts, register address(es), shared memory regions, etc) can remain in the device tree file.  To do this, just comment the firmware-name line and indicate the FPGA is already programmed in the dts file:

````
        fragment@0 {
                target-path = "/soc/base-fpga-region";
                #address-cells = <1>;
                #size-cells = <1>;
                __overlay__ {

                        // firmware-name = "path/to/whatever.rbf";
                        external-fpga-config = <1>;

````

dkhughes, 2018-01-02:
That makes a lot of sense, I like that solution. Just need to make sure the ol driver can handle a nonexistent firmware-name.

the-snowwhite, 2018-04-07:
This issue is fixed with this commit:

    https://github.com/machinekit/machinekit/commit/081cc70a0cae4a67ea31850ed16223c3ae88c0e2

And confirmed to work in this release:

    https://github.com/the-snowwhite/soc-image-buildscripts/releases/tag/Release_2.0

closed 2018-04-07 by the-snowwhite