machinekit/machinekit #1264: hm2_soc_ol.so: undefined symbol: hm2_unregister

luminize, 2017-08-30:
Happens when trying to load a different FPGA configuration when restarting a realtime environment 

see output below when trying to load `config="firmware=socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.dtbo num_stepgens=5 enable_adc=1" debug=1`

```
login as: machinekit
Debian GNU/Linux 8

Machinekit SoC/FPGA Debian Image, Altera CycloneV 2016-11-21

Support/FAQ: http://www.machinekit.io/

default username:password is [machinekit:machinekit]

machinekit@192.168.178.49's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Aug 30 12:37:08 2017 from t5400.fritz.box
environment set up for RIP build in /home/machinekit/machinekit/src
machinekit@mksocfpga:~$ export DEBUG=5
machinekit@mksocfpga:~$ halrun -I
msgd:0 stopped
rtapi:0 stopped
halcmd: loadrt hostmot2
<stdin>:1: Realtime module 'hostmot2' loaded
halcmd: newinst hm2_soc_ol hm2-socfpga0 -- config="firmware=socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.dtbo num_stepgens=5" debug=1
<stdin>:2: Realtime module 'hm2_soc_ol' loaded
halcmd: exit
<commandline>:0: Realtime threads stopped
machinekit@mksocfpga:~$ halrun -I
msgd:0 stopped
rtapi:0 stopped
halcmd: loadrt hostmot2
<stdin>:1: Realtime module 'hostmot2' loaded
halcmd: newinst hm2_soc_ol hm2-socfpga0 -- config="firmware=socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.dtbo num_stepgens=5" debug=1
<stdin>:2: Realtime module 'hm2_soc_ol' loaded
halcmd: exit
<commandline>:0: Realtime threads stopped
machinekit@mksocfpga:~$ halrun -I
msgd:0 stopped
rtapi:0 stopped
halcmd: newinst hm2_soc_ol hm2-socfpga0 -- config="firmware=socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.dtbo num_stepgens=5 enable_adc=1" debug=1
<stdin>:1: insmod failed, returned -1:
do_load_cmd: dlopen: /home/machinekit/machinekit/rtlib/rt-preempt/hm2_soc_ol.so: undefined symbol: hm2_unregister
rpath=/home/machinekit/machinekit/rtlib/rt-preempt:/home/machinekit/machinekit/lib
See /var/log/linuxcnc.log for more information.
halcmd: exit
<commandline>:0: Realtime threads stopped
```

snip from /var/log/linuxcnc.log
```
Aug 30 12:58:50 mksocfpga msgd:0: zeroconf: registered 'Log service on mksocfpga.local pid 2198' _machinekit._tcp 0 TXT "uuid=a42c8c6b-4025-4f83-ba28-dad21114744a" "instance=f7baa52c-8d82-11e7-8779-bad04a9c4ece" "service=log" "dsn=ipc:///tmp/0.log.a42c8c6b-4025-4f83-ba28-dad21114744a"
Aug 30 12:59:01 mksocfpga rtapi:0: do_load_cmd: dlopen: /home/machinekit/machinekit/rtlib/rt-preempt/hm2_soc_ol.so: undefined symbol: hm2_unregister
Aug 30 12:59:01 mksocfpga rtapi:0: rpath=/home/machinekit/machinekit/rtlib/rt-preempt:/home/machinekit/machinekit/lib
Aug 30 12:59:01 mksocfpga rtapi:0: 1:rtapi_app:2203:user do_load_cmd: dlopen: /home/machinekit/machinekit/rtlib/rt-preempt/hm2_soc_ol.so: undefined symbol: hm2_unregister
Aug 30 12:59:01 mksocfpga rtapi:0: 1:rtapi_app:2203:user rpath=/home/machinekit/machinekit/rtlib/rt-preempt:/home/machinekit/machinekit/lib
```

RIP build commit:
```
machinekit@mksocfpga:~/machinekit$ git log
commit 680d20cf9de9d34b7a906ef58e68461760db01b7
Merge: 4a09bfd 0e87824
Author: cdsteinkuehler <charles@steinkuehler.net>
Date:   Sun Aug 27 10:09:39 2017 -0500

    Merge pull request #1253 from the-snowwhite/DE0_Nana_SoC_Cramps

    De0 nano soc cramps
```

the-snowwhite, 2017-08-31:
Odd
This is the only change I can think of related to the new _CRAMPS commit series that directly affects the (other) DE0_Nano_SoC_DB25 quartus project:

https://github.com/machinekit/mksocfpga/commit/fca882b8bc25ff2e2e81b4a0979ee2c6bf7c041e#diff-0500c35c248378bdb66c98fea8742ad8L646

https://github.com/machinekit/mksocfpga/commit/fca882b8bc25ff2e2e81b4a0979ee2c6bf7c041e

ArcEye, 2017-09-01:
@luminize 
Try running `nm -C hm2_soc_ol.so`
if that does not produce anything the regular symbols have probably been stripped so try
`nm -D hm2_soc_ol.c`

That should produce this
```
root@armv7-bb:/downloads# nm -D hm2_soc_ol.so
         U __aeabi_unwind_cpp_pr0
         .......
         U hm2_register
         U hm2_unregister
         .......
         U __xstat
```
Then try from the same build
```
root@armv7-bb:/downloads# nm -D hostmot2.so
         U __aeabi_unwind_cpp_pr0
         ......
00015958 T hm2_allocate_bspi_tram
00015ea0 T hm2_bspi_set_read_function
00015a8c T hm2_bspi_setup_chan
00015f54 T hm2_bspi_set_write_function
000159c4 T hm2_bspi_write_chan
00002264 T hm2_get_bspi
0000238c T hm2_get_pktuart
00002420 T hm2_get_sserial
000022f8 T hm2_get_uart
00017598 T hm2_pktuart_read
00017200 T hm2_pktuart_send
00017050 T hm2_pktuart_setup
00003e40 T hm2_register
0001580a T hm2_tram_add_bspi_frame
00016974 T hm2_uart_read
000166b0 T hm2_uart_send
000164f8 T hm2_uart_setup
00004c84 T hm2_unregister
        .......
         U __xstat
```
These prints are from a natively built package prior to any commits from this work

The hm2_unregister is specifically exported at https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.c#L1752

Seems peculiar that that symbol would be missing but not `hm2_register`, which must occur first ☹️

luminize, 2017-09-01:
Here are 3 outputs, from hm2_soc_ol, hostmot2 and hm2_pci
https://gist.github.com/luminize/1c0cad043854a53451e714ff5ed3412a
https://gist.github.com/luminize/e632ff0c23a5ce3ac5ddf9c7317e6824
https://gist.github.com/luminize/688eb6ad21ee5ab3bec484de72ae1c98

[hm2_soc_ol, line 60](https://gist.github.com/luminize/1c0cad043854a53451e714ff5ed3412a#file-output-txt-L60)
```
00000f18 t hm2_soc_write
         U hm2_unregister
00000b8c t _init
```

[hostmot2, line 450](https://gist.github.com/luminize/e632ff0c23a5ce3ac5ddf9c7317e6824#file-output-txt-L450)
```
00017ed0 t hm2_uart_write
00004e94 T hm2_unregister
0001bb2c t hm2_waitirq
```

[hm2_pci, line 55](https://gist.github.com/luminize/688eb6ad21ee5ab3bec484de72ae1c98#file-output-txt-L55)
```
         U hm2_register
         U hm2_unregister
0000081c t _init
```

ArcEye, 2017-09-01:
Looks to be pretty much as it should be.
Can't do a direct comparison because you used the -C switch and the binary package I opened had all the symbols stripped.
Might mean something to the others, or it might be nothing to do with it.

machinekoder, 2017-11-16:
@luminize
In your output above the line `<stdin>:1: Realtime module 'hostmot2' loaded` is missing. I had got exactly the same error message when I forgot to load the `hostmot2` module.