machinekit/machinekit #1269: possible aarch64 support

bjharper, 2017-09-03:
I have had some success compiling and running LinuxCNC on aarch64 (Ubuntu 16.04) using Realtime preempt 4.11 kernel but I am struggling a little with machine kit.  I seems to always end up with an error looking for sys/io.h
```
Compiling machinetalk/support/sizes.c
In file included from rtapi/rtapi.h:1016:0,
                 from hal/lib/hal.h:126,
                 from ./emc/motion/motion.h:80,
                 from machinetalk/support/sizes.c:2:
rtapi/rtapi_io.h:39:43: fatal error: sys/io.h: No such file or directory
compilation terminated.
Makefile:420: recipe for target 'objects/machinetalk/support/sizes.o' failed
```

Basic steps is to install all dependencies then
$ debian/configure -pr
$ sudo mk-build-deps -ir
$ cd src
$ ./autogen.sh
$ ./configure --with-rt-preempt  --with-posix
$ make -j2

Kernel im working with at the moment is development Realtime PREEEMT branch "'Linux odroid64 4.11.12-rt10 #6 SMP PREEMPT RT Sat Aug 26 17:27:03 AEST 2017 aarch64 aarch64 aarch64 GNU/Linux"

Im working on a HAL driver for the Odroid C2 and its GPIO pins so I dont know if I really need the io.h support in rtapi (no PCI interface ill be hitting GPIO registers directly hopefully) and if I need outb / inb functions then it looks like I could probably emulate them myself.

Looks like I will need to setup an aarch64 specific configuration in the machinekit source or maybe install for further dependencies for aarch64?  Im compiling machinekit on an aarch64 platform also, no cross-compiling.

sirop, 2017-09-03:
@bjharper 
Can't you use either Debian Jessie or Stretch instead?
See https://forum.odroid.com/viewtopic.php?f=138&t=27449 .

bjharper, 2017-09-03:
Thanks I will try Debian and see how it goes.

sirop, 2017-09-03:
Could you please publish your instructions for building Linux odroid64 4.11.12-rt10,
as then I'd try it on my site too.

ArcEye, 2017-09-03:
> rtapi/rtapi_io.h:39:43: fatal error: sys/io.h: No such file or directory

You will probably find it is in a subdirectory within /usr/include that relates to the arch type.
Ubuntu especially kept moving things around, I just used to symlink the contents to the main dir and all worked fine.

bjharper, 2017-09-03:
Linux odroid64 4.11.12-rt10 has some limitations for the moment (the main one being no hdmi output),  I was testing LinuxCNC over remote X11 sessions but want to use MachineKit for the extra features such as web UI (particularly as I have no HDMI) and more customised devices such as what has been done with Beaglebone and Pi.  HDMI support for C2 should be soon enough in mainline Linux but 4.11 is enough to get me started for now.

The kernel build is straightforward on an aarch64 platform but the installation is a bit more complex.
- clone the git from https://git.kernel.org/pub/scm/linux/kernel/git/rt/linux-rt-devel.git/log/?h=linux-4.11.y-rt
- run a defconfig
- modify the kernel config for Realtime
CONFIG_PREEMPT_RT_FULL
CONFIG_HZ_1000
CONFIG_EMBEDDED
CONFIG_SCHED_SMT (not sure if this is the smartest choice but seems to work well in testing)

The kernel itself can be a little troublesome to install as it must be made into a compatible uImage for uboot and booted with a different configuration from uboot but I will provide some instruction on the build as soon as I can document it a little clearer.

I did a search through /usr/include but no io.h on Ubuntu.  I notice in my kernel source there is an io.h but most likely not the user library im after for aarch64?  I will try again on Debian to see if I can work it out or it has different libraries.

Thanks!

sirop, 2017-09-03:
@bjharper 
Thanks so far. 

It would be interesting to compare the latency of Odroid C2 with that of Pi3 https://github.com/koppi/mk/blob/master/Machinekit-RT-Preempt-RPI.md#latency-evaluation .

bjharper, 2017-09-03:
`odroid@odroid64:\~$ uname -a
Linux odroid64 4.11.12-rt10 #6 SMP PREEMPT RT Sat Aug 26 17:27:03 AEST 2017 aarch64 aarch64 aarch64 GNU/Linux
odroid@odroid64:~$ sudo cyclictest -t1 -p 80 -n -i 10000 -l 10000
\# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.93 0.42 0.16 2/218 27058          

T: 0 (27052) P:80 I:10000 C:  10000 Min:      7 Act:   10 Avg:   10 Max:      23`

that was done with a `odroid@odroid64:~$ cat /dev/zero > /dev/null` on the odroid and a `$ sudo ping -i 0.01 10.1.1.12` from my host to the odroid.   Impressive I guess but I can bust the max out if I really try.  (ie 100 x glxgears running over ssh & X11 or heavy disk IO got it up to 200 ms)

sirop, 2017-09-03:
Yes, very  impressive latency results.

Some time ago with an older kernel 3.XX there was a problem with the network driver on C2.
Have you noticed anything like that with the newer kernel?

Anyway: waiting for  kernel installation instructions.

bjharper, 2017-09-03:
> Some time ago with an older kernel 3.XX there was a problem with the network driver on C2.
I actually had issues with the USB drivers in 3.10.y kernel, disabling usb got the numbers down nicely but I was not pushing the network to tell if it could be a problem.  In 4.11 the latency is a worse with a 100Mbit connection, 1GBit gives better results.  I did some ftrace latency tests and could see ethernet getting in the way if I maxed it out and the same with emmc card for disk.

I will do install instructions for stretch as soon as I can.

ArcEye, 2017-09-03:
>rtapi/rtapi_io.h:39:43: fatal error: sys/io.h: No such file or directory

>You will probably find it is in a subdirectory within /usr/include that relates to the arch type.
Ubuntu especially kept moving things around, I just used to symlink the contents to the main dir and all worked fine.

On an armhf system, the path is `/usr/include/arm-linux-gnueabihf/sys/io.h`
Yours will be the same or very similar

bjharper, 2017-09-04:
I've wrote a quick README on how I installed on meveric's Debian Stretch and then the 4.11-y-rt kernel.  Please let me know if there are any questions, errors or something does not work.

http://www.builtonline.net/c2rt-v4.11.12-rt11/README

I havent attempted compiling machinekit as yet on Stretch, but I have checked Debian for io.h.  No sign of it (does not exist in /usr/include/aarch64-linux-gnu/sys/io.h or /usr/include/aarch64-linux-gnu/asm/io.h or anywhere).  I will check the kernel config to see if it should be installed as part of the headers as maybe this is where I am going wrong.

ArcEye, 2017-09-04:
> I havent attempted compiling machinekit as yet on Stretch, but I have checked Debian for io.h. No sign of it (does not exist in /usr/include/aarch64-linux-gnu/sys/io.h or /usr/include/aarch64-linux-gnu/asm/io.h or anywhere). 

In that case it is probably the same situation I had when I compiled machinekit on the Pine64 some time back.

There was no io.h for 64 bit ARM and I had to create my own.
The ARM architecture does not support some of the calls (ioperm), the asm code is different and assumptions about ports and base addresses don't work.

This is the file I wrote which works with Pine64
What may well not be right for your board is the base address PCI_IOBASE.
I had to compute this after determining the start memory address that insmod uses to load modules, as explained in the comments.

```
* 
A largely dummy io.h to statisfy the requirements of Machinekit  
on aarch64
The functions still require a valid base address, which appears to be
#define PCI_IOBASE              ((void __iomem *)(MODULES_VADDR - SZ_2M))
MODULES_VADDR being the address at which insmod begins loading kernel modules
*/

#ifndef	_SYS_IO_H
#define	_SYS_IO_H

#include <features.h>

// as per arch/.../memory.txt
#define PCI_IOBASE   0xfee00000UL
// ((void __iomem *)(MODULES_VADDR - SZ_2M))
//((void __iomem *)PCI_IO_START)

__BEGIN_DECLS

/* If TURN_ON is TRUE, request for permission to do direct i/o on the
   port numbers in the range [FROM,FROM+NUM-1].  Otherwise, turn I/O
   permission off for that range.  This call requires root privileges.

   Portability note: not all Linux platforms support this call.  Most
   platforms based on the PC I/O architecture probably will, however.
   E.g., Linux/Alpha for Alpha PCs supports this.  */
//extern int ioperm (unsigned long int __from, unsigned long int __num,
//                   int __turn_on) __THROW;

/* Set the I/O privilege level to LEVEL.  If LEVEL>3, permission to
   access any I/O port is granted.  This call requires root
   privileges. */

// Does not appear this call is supported
// dummy function to return 0 (success and keep pci_write happy

//int iopl(int __level) { return 0; }

#if defined __GNUC__ && __GNUC__ >= 2

// Does not appear this call is supported
// dummy function to return 0 (success and keep pci_write happy

static __inline int iopl(int __level) { return 0; }

static __inline unsigned char 
inb (unsigned short int __port)
{
    unsigned char val;
    unsigned long addr = ((unsigned long) __port + PCI_IOBASE);
    
    asm volatile("ldrb %w0, [%1]" : "=r" (val) : "r" (addr));
    return val;
}

static __inline unsigned short int
inw (unsigned short int __port)
{
    unsigned short val;
    unsigned long addr = ((unsigned long) __port + PCI_IOBASE);
  
    asm volatile("ldrh %w0, [%1]" : "=r" (val) : "r" (addr));
    return val;
}

static __inline unsigned int
inl (unsigned short int __port)
{
    unsigned int val;
    unsigned long addr = ((unsigned long) __port + PCI_IOBASE);
  
    asm volatile("ldrh %w0, [%1]" : "=r" (val) : "r" (addr));
    
    return val;
}

///////////////////////////////////////////////////////////////////

static __inline void
outb (unsigned char __value, unsigned short int __port)
{
    unsigned long addr = ((unsigned long) __port + PCI_IOBASE);
    asm volatile("strb %w0, [%1]" : : "r" (__value), "r" (addr));
}

static __inline void
outw (unsigned short int __value, unsigned short int __port)
{
    unsigned long addr = ((unsigned long) __port + PCI_IOBASE);
    asm volatile("strh %w0, [%1]" : : "r" (__value), "r" (addr));
}

static __inline void
outl (unsigned int __value, unsigned short int __port)
{
    unsigned long addr = ((unsigned long) __port + PCI_IOBASE);
    asm volatile("strh %w0, [%1]" : : "r" (__value), "r" (addr));
}

#endif	/* GNU C */

__END_DECLS
#endif /* _SYS_IO_H */
```

sirop, 2017-09-04:
What about https://kernel.googlesource.com/pub/scm/linux/kernel/git/rt/linux-rt-devel/+/v4.11.12-rt11/include/asm-generic/io.h ?

bjharper, 2017-09-06:
Thanks @ArcEye im looking at the code and looking into a dummy address to use for the Odroid C2 and PCI_IOBASE

@sirop I think https://kernel.googlesource.com/pub/scm/linux/kernel/git/rt/linux-rt-devel/+/v4.11.12-rt11/include/asm-generic/io.h is kernel libraries but we need userspace which is not being exported to /usr/include for aarch64 on Odroid C2

bjharper, 2017-09-07:
Its now compiling doing what you have suggested @ArcEye For PCI_IOBASE I used what was reported at startup for the PCI virtual address allocation  PCI I/O : 0xffff7dfffee00000 - 0xffff7dffffe00000   (    16 MB)

Im not sure this is right, need to look at it further (insmod was using a different address)

I had some troubles building various versions of zeromq and czmq but got a working config on stretch.

The GUI is working over remote X and emcweb is working when enabled in sim mode so I can now starting playing with some config and looking at the HAL driver.

Thanks!

ArcEye, 2017-09-07:
Glad you got there.

Regards czmq and zeromq, I made some packages, see https://github.com/machinekit/machinekit/issues/1228#issuecomment-316364736

They work fine, I am building on Stretch amd64, i386 and armhf using them.

bjharper, 2017-09-07:
I did actually see them but im using aarch64 and didnt think you would have a port there?

ArcEye, 2017-09-07:
Of course you are, my brain is trying to do too many things at once!

I may have later, but it is someway down the list.  I just built from sources as you did and then
forgot all about it.

If you want to produce a package, because you are likely to do this repeatedly or want to circulate it,
the easiest way is to use `checkinstall`.

Build the library then from its root run `checkinstall install=no make install`
Answer the q's and change whatever and it will output a usable .deb

That is what I did initially with the other Stretch packages until I got a build system sorted.

bjharper, 2017-09-11:
I uploaded the packages I created here http://www.builtonline.net/c2rt-v4.11.12-rt11/  Sorry im not the best with debian packaging so they are /usr/local but should work ok.  Ill try to provide some detail on the machinekit build process as there were some dependencies required for stretch minimal image.  The 2 packages zeromq and czmq where the only ones I needed to compile from source and then the sys/io.h

I have been working on a basic gpio hal driver for the C2 starting as a base with the Beaglebone hal_bb_gpio.c to control GPIO and also incorporating an option to build machinekit for the C2.  So far so good everything is compiling ok (I think) optimised for A-53 cores with neon,  floating point and compiler specific options enabled.

I now need to create a hal config which raises a question, am I better to have pin config in a devicetree for the board or in a .hal config file in userspace or both?  I need to either make the hal driver search the devicetree or check a hal config file for pin configuration.  Im using hal_gpio_demo.hal file (Raspberry) as a test hal config and starting point for debug on the Odroid C2.

machinekoder, 2017-10-01:
@bjharper I described how to create a GPIO HAL driver [here](https://machinekoder.com/machinekit-on-the-c-h-i-p-9-computer/). For the pin config, I suggest using the device tree overlay to stay consistent with other GPIO drivers. The solution used for the BBB and CHIP passing the GPIO pins via an HAL argument works very well and does not need any checking of the GPIO configuration from the HAL driver.

machinekoder, 2017-12-05:
@bjharper What is the status of your project? I have a spare ODroid C2 I', thinking about putting to use with Machinekit.