machinekit/machinekit #1294: Problem with RT_PREEMPT and newinst in Python API

machinekoder, 2017-10-20:
With recent changes, it seems that the `newinst` function of the Python HAL API is broken on RT_PREEMPT. In particular, already instantiated components are not detected as RT anymore.

How to reproduce:
`test.py`
```test.py
from machinekit import rtapi as rt

rt.init_RTAPI()

signame = 'test'
safetyLatch = rt.newinst('safety_latch', 'safety-latch.%s-error' % signame)
signame = 'test2'
safetyLatch = rt.newinst('safety_latch', 'safety-latch.%s-error' % signame)
```

Run:
```bash
realtime start
python test.py
realtime stop
```

Error:
```
Traceback (most recent call last):
  File "test.py", line 6, in <module>
    safetyLatch = rt.newinst('safety_latch', 'safety-latch.%s-error' % signame)
  File "hal/cython/machinekit/rtapi.pyx", line 251, in machinekit.rtapi.RTAPIcommand.newinst (hal/cython/machinekit/rtapi.c:6023)
RuntimeError: 'safety_latch' not an RT component
```

Possible source of the error:
https://github.com/machinekit/machinekit/blob/master/src/hal/cython/machinekit/rtapi.pyx#L159

machinekoder, 2017-10-20:
@ArcEye have there been any recent changes that change the representation of the HAL type?

machinekoder, 2017-10-20:
I have no clue why this stabs us in the back right now, but I think I found the problem:

`test.py`
```python
print(rt.classify_comp('safety_latch'))
c = hal.components['safety_latch']
print(c)
print(c.type == hal.TYPE_RT)
print(c.type is hal.TYPE_RT)
```

Result
```
1
<machinekit.hal.Component object at 0xb62cecf0>
True
False
```

The problem is the identity comparison in the check. I have no idea why this fails now but not before.

ArcEye, 2017-10-20:
> @ArcEye have there been any recent changes that change the representation of the HAL type?

None that I can think of.
I take it the czmq API changes could not be connected?
I don't tend to touch anything pythonic 😉

ArcEye, 2017-10-21:
From a brief piece research, the reason for the difference appears to be
`==` checks that the operands are equivalent / same value as each other
`is` checks that they refer to the same object

Therefore `1 == True` returns True, but `1 is True` will return False
which seems quite close to your example above.

closed 2017-11-16 by machinekoder