machinekit/machinekit #1306: Broken SoC FPGA on Zynq after Multicore Merge (No IRQ based thread control)

dkhughes, 2017-10-31:
Hi everyone,

It looks like after the multicore merge some of the work @mhaberler did for external interrupt sync is missing, and it's causing the FPGA configs that depend on PREEMPT RT and nowait to fail. I'm not as good with the internal rtapi details as he was, but I would like to begin working back in the missing pieces. Were the following removed for any specific reason or were they just missed (most of these come from #941)? 

Allow nowait on PREEMPT-RT:
[https://github.com/machinekit/machinekit/pull/941/commits/9e1d0c88286166f438a4ddcac8b826e1ff70db04](https://github.com/machinekit/machinekit/pull/941/commits/9e1d0c88286166f438a4ddcac8b826e1ff70db04)

Remove the FP thread tag which is unused now:
[https://github.com/machinekit/machinekit/pull/941/commits/e7f5d5f02bee10c46b44d58869d50987279e1a37](https://github.com/machinekit/machinekit/pull/941/commits/e7f5d5f02bee10c46b44d58869d50987279e1a37)

Actually perform the nowait condition:
[https://github.com/mhaberler/machinekit/blob/cf7c8e3f30c45d1c6bcc20b02d1e178c3db802cc/src/hal/lib/hal_thread.c#L84](https://github.com/mhaberler/machinekit/blob/cf7c8e3f30c45d1c6bcc20b02d1e178c3db802cc/src/hal/lib/hal_thread.c#L84)

Minor changes really, I just want to run it by someone with a little more experience before I hack them back in.

dkhughes, 2017-10-31:
The nowait condition has been rolled into the wait hook inside rt-preempt implementation. I think that means the only issue is nowait not allowed when preempt-rt is the threading provider:

[https://github.com/machinekit/machinekit/commit/9e1d0c88286166f438a4ddcac8b826e1ff70db04](https://github.com/machinekit/machinekit/commit/9e1d0c88286166f438a4ddcac8b826e1ff70db04)

I will test configs as soon as it's done compiling locally.

ArcEye, 2017-10-31:
I think the problem is just chronology.
@mhaberler did those changes after we finished the multicore work.
They obviously did not get backported into it, so were lost.
_mea culpa_ I expect 😦 

One thing I would ask is that you do not use the
#if 0 / #endif contruct to comment out and just use proper annotated comment blocks.

The significance of the former can easily be missed when skimming code.

dkhughes, 2017-10-31:
"just use proper annotated comment blocks."
Yep, no problem.  I agree it's much clearer to read.

Allowing preempt-rt threads to use nowait did get me passed that issue, but my configs have some dependencies that I need to port forward to be able to test fully and make sure the functionality is all there. I'll put together a pull request as soon as I get something running consistently.

dkhughes, 2017-11-02:
Fixed with #1308.

closed 2017-11-02 by dkhughes