machinekit/machinekit #1310: Convert setup.sh to .bbio files

machinekoder, 2017-11-03:
## Description
Early configurations used custom `setup.sh` bash scripts to configure pin muxing on the BBB. In the meantime, the BB universal IO has been introduced.

In most cases this is pretty much a copy and paste job and therefore great for beginners.

Example BBIO file:
<https://github.com/machinekit/machinekit/blob/master/configs/ARM/BeagleBone/Fabrikator-Mini-CRAMPS/CRAMPS.bbio>

Example `setup.sh`: <https://github.com/machinekit/machinekit/blob/master/configs/ARM/BeagleBone/Probotix/setup.sh>

There is also a [graphical BBIOConfig editor](https://github.com/machinekoder/BBIOConfig)

## Task
Find and convert setup.sh files in <https://github.com/machinekit/machinekit/tree/master/configs/ARM/BeagleBone>

JTrantow, 2017-11-08:
setup.sh also checks for PRU files. Is this redundant with run.py calling launcher.check_installation() in which case setup.sh could be removed?

JTrantow, 2017-11-09:
I looked at configs/ARM/BeagleBone/Fabrikator-Mini-CRAMPS/run.py which uses launcher.load_bbio_file('cramps2_cape.bbio'). This works fine if you are launching the GUI and using run.py but what about if someone wants to run the config from the command line or desktop?  setup.sh gets called from .hal no matter how the config is started.

Can launcher.load_bbio_file() get moved inside the .hal? Seems like this is where it belongs.

machinekoder, 2017-12-08:
@JTrantow I think that's a good idea. 

The right place would be hardware.py (cramps.py) for the Python-based configs. 

For HAL language files you can use `loadusr config-pin -f cramp.bbio` at the beginning of the file.

Let me know if it works.