machinekit/machinekit #1316: Unable to apt get-install  dirmngr

mac44mag, 2017-11-15:
Is anyone else having this problem?

Problem:  Unable to set repository and install Machinekit package(s) due to an apparent failure to install dirmngr.

Machinekit link: http://www.machinekit.io/about/ 

System: Linux YETI-Linux 4.9.0-4-amd64 #1 SMP Debian 4.9.51-1 (2017-09-28) x86_64 GNU/Linux
Computer: AMD 2435 dual processor w/24 GB RAM.

------------------------------
Instructions from Machinekit site:

sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv 43DDF224
sudo sh -c \
  "echo 'deb http://deb.machinekit.io/debian stretch main' > \
  /etc/apt/sources.list.d/machinekit.list"
sudo apt-get update
---------------------------------
Running the first statement generates:

root@YETI-Linux:/home/leonard# sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv 43DDF224
Executing: /tmp/apt-key-gpghome.7Z9yU5iGNi/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv 43DDF224
gpg: failed to start the dirmngr '/usr/bin/dirmngr': No such file or directory
gpg: connecting dirmngr at '/tmp/apt-key-gpghome.7Z9yU5iGNi/S.dirmngr' failed: No such file or directory
gpg: keyserver receive failed: No dirmngr

------------------------------ 
Attempting to get-install dirmngr returns:

root@YETI-Linux:/etc/apt# sudo apt-get install dirmngr
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package dirmngr is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'dirmngr' has no installation candidate
-------------------------------

Where do I go from here?  My apologies if this is misplaced or previously answered.

Thank  you in advance!

Leonard

ArcEye, 2017-11-16:
You haven't said what distro you are using, yeti seems to be some sort 
of audio program?

If you are using Stretch, which the kernel suggests, there is a package

https://packages.debian.org/stretch/dirmngr

If you are using something else, we don't support it.

If the distro is properly Debian based, the original package may work 
installed manually.

Your kernel does not appear to be rt, so not sure what you are hoping to 
do with machinekit.

mac44mag, 2017-11-16:
THANK YOU!  That appears to be what I need...he said, hopefully.
I'm using the Debian-9.2.1-amd64-DVD-1.iso downloaded from: https://cdimage.debian.org/debian-cd/current/amd64/iso-dvd/ .   Yeti is my network name.

I am new to Linux, but I have a friend who has used it in one form or another for 10+ years.  I will get his assistance to install the package manually.

Thank you for your time and assistance!!  I will be in touch.

Leonard (mac44mag)

mac44mag, 2017-11-16:
If there's any other info that you need, I'd be more than happy to attempt to supply it.

Leonard

luminize, 2017-11-18:
I encountered the same on a vanilla Debian stretch installation, but `sudo apt-get install dirmngr` solved it. So this is not Machinekit related imo.

closed 2017-11-18 by luminize