machinekit/machinekit #1318: Debian Stretch RT-Preempt Latency

rafterrc, 2017-11-17:
Before beginning the install I ran a latency test on the computer (Intel Core i5 3400 series). with the Debian Wheezy LinuxCNC latency test via Live CD. It came up with ~9,000ns. I thought good to go. After getting Stretch and Machinekit up on the computer, I ran the latency test with the version in Machinekit. The result is 131,000ns!? (BTW, latency test does not give results for base period. I have no clue how to configure.) The Machinekit latency test result is based on the servo number.

When I check scheduling, I get that things run under root or me are Schedule_Other and priority listed as 0. It seems latency test is not running RT. I checked to make sure that RT-Preempt kernel is installed, and it is. Pstree runs through halcmd with no rtapi in view, but rtapi is running by itself according to pstree.

Is there anything wrong with this? If not, then it seems a Mesa board is in order. If so, then what do I need to do?

rafterrc, 2017-11-17:
I found the fix to get 25us thread and 1ms thread. I am now confused as servo thread is now coming out as 71,000ns and base at ~73Kns. But now I get a message that msgd and rtapi are stopped.

rafterrc, 2017-11-17:
From the log:

rtapi_app exit detected - Scheduled shutdown.

This report occurs right after:  zeroconf: 'Registered Log service on Debian.local ..."

cdsteinkuehler, 2017-11-18:
Requests for help should be sent to the Machinekit Google group, this
issue tracker is for specific bugs, ideally with details to reproduce.

The LinuxCNC LiveCD likely uses an RTAI kernel (unless they have
recently changed) which will provide lower latencies than a PREEMPT_RT
or Xenomai kernel typically used with Machinekit.  You need to make
sure you are running a real-time kernel and have the appropriate
Machinekit packages installed for the kernel you are using.

On 11/17/2017 11:53 AM, rafterrc wrote:
> Before beginning the install I ran a latency test on the computer (Intel Core i5 3400 series). with the Debian Wheezy LinuxCNC latency test via Live CD. It came up with ~9,000ns. I thought good to [...]

-- 
Charles Steinkuehler
charles@steinkuehler.net

darenschwenke, 2017-11-18:
The google group: https://groups.google.com/forum/#!forum/machinekit

rafterrc, 2017-11-18:
As I have gotten into the diagnosis, it looks like a bug to me. It looks like rtapi_app is not running as root, as when I run latency-test as root, then I get the extra message that rtapi_app is not running with root or enough privilege to cordon off memory. That is either a bug or operator error. If operator error, then there is something missing in the documentation (your programming of the operator). 

There is an rtapi.ini file. Is that edited when either configuring for step or pnc, and when rtapi_app is given root privelege? Then running latency test first (without root privilege) gives a bogus answer, and forcing a rerun of latency test after configuring.

From the outsiders perspective; that is, someone with little knowledge of the software and it's configuration process, it is difficult to tell whether it's a bug or operator ignorance. Perhaps this Github page should be restricted to developers, and the developers are responsible for determining bugs from the Google page and porting them over here.

closed 2017-11-25 by ArcEye