machinekit/machinekit #1323: Building machinekit on Debian Buster

ArcEye, 2017-11-22:
Have done a quick preview of Buster, to see what might be in store.

Buster is only Alpha 1 status, so essentially Debian 9, with new kernel, gcc etc but not diverged too far yet

- Machinekit builds and runs 'as is' from the same development package list as I use for Stretch 👍  

- python-gtksourceview2 has been deprecated 👎 
This means that there is another nail in the coffin of gladevcp, because it is used in editor / viewer
widgets within it (eg.. hal_sourceview)

- gcc-7.2.0-16 does not throw up any further esoteric errors, from those ironed out with gcc-6 👍 

- python-2.7 still ships, there are still too many libs not ported to 3.0 to ignore it 👍 

Essentially looks like the packaging for Stretch can be used, minus the gladevcp dependency.

The bigger question, which Linuxcnc have started looking at seriously, is the completely untenable reliance upon gtk2, glade2 and eventually python 2.7, if trying to distribute something which will run unsupported on existing distros.

Soon they won't just look tired, they will cease to work.

ArcEye, 2017-11-25:
There is currently no rt-preempt kernel for Buster in the archive.
The kernel for Posix is 4.13, which has not had a rt patchset released yet, but doubtless this will be resolved nearer release date.

The latest patchset is 4.9.61, which works fine.

ArcEye, 2017-11-30:
PR #1324 readies the build system for Buster packaging with neutral effect on existing distros.

"I have not yet changed the machinekit repo or build system to incorporate a Buster build.
Anticipate Buster being a bit of a 'moving target' until at Beta stage.
This just assures readiness as I have built specific docker images for Buster builds."

I will incorporate into my own repo builder  before long.

"The thing it should throw into sharp relief, is the beginning of the end for gladevcp.
With the deprecation of `python-gtksourceview2`, gladevcp will not build its viewer widgets and the adoption of gtk3, glade3 and python3(.6) is likely to kill it off completely.

Buster should probably be the distro where we give serious thought to adopting Qt, replacing python-gtk, tcl etc. with sustainable widgets and code.
Work along these lines is being promoted in the linuxcnc project too.

Axis still works and pyvcp (which relies upon bwidget) still works, so there is a bit of leeway yet."

ArcEye, 2017-12-02:
See also https://github.com/LinuxCNC/linuxcnc/issues/341 re halgui

ArcEye, 2018-07-31:
I have been using buster / sid for a further 8 months without problems.

The much vaunted deprecation's have not yet occurred, python-gnome2 for instance is still available in sid.

Closing

the-snowwhite, 2019-09-27:
python-gnome2 has now been removed from buster and sid

So to install freshly made arm64 buster packages (in buster), I used following procedure: 

Downloaded the package from stretch and tested if it would install

    machinekit@mksocfpga-ultra96v1-soc:~$ sudo apt install ./python-gnome2_2.28.1+dfsg-1.2_arm64.deb
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    Note, selecting 'python-gnome2' instead of './python-gnome2_2.28.1+dfsg-1.2_arm64.deb'
    Some packages could not be installed. This may mean that you have
    requested an impossible situation or if you are using the unstable
    distribution that some required packages have not yet been created
    or been moved out of Incoming.
    The following information may help to resolve the situation:

    The following packages have unmet dependencies:
    python-gnome2 : Depends: libbonobo2-0 (>= 2.32.1-3~) but it is not installable
                    Depends: libbonoboui2-0 (>= 2.15.1) but it is not installable
                    Depends: libgnome-2-0 but it is not installable
                    Depends: libgnomeui-0 (>= 2.22.0) but it is not installable
                    Depends: libgnomevfs2-0 (>= 1:2.17.90) but it is not installable
                    Depends: liborbit-2-0 (>= 1:2.14.10) but it is not installable
                    Depends: python-gconf (= 2.28.1+dfsg-1.2) but it is not installable
                    Depends: python-pyorbit (>= 2.0.1-4) but it is not installable
    E: Unable to correct problems, you have held broken packages.

libbonobo2-0    ->  in sid
libbonoboui2-0  ->  in sid
libgnome-2-0    ->  in sid
libgnomeui-0    ->  in sid
libgnomevfs2-0  ->  in sid
liborbit-2-0    ->  in sid
python-gconf    ->  in stretch
python-pyorbit  ->  in sid
all sid vesrions are same as stretch versions

added 
    
    deb http://ftp.dk.debian.org/debian/ stretch main

to /etc/apt/sources.list

    sudo apt update

    machinekit@mksocfpga-ultra96v1-soc:~$ sudo apt install python-gnome2
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    The following additional packages will be installed:
    gconf-service gconf2 gconf2-common gnome-mime-data libart-2.0-2 libbonobo2-0 libbonobo2-common libbonoboui2-0 libbonoboui2-common libgconf-2-4 libgnome-2-0 libgnome-keyring-common libgnome-keyring0
    libgnome2-common libgnomecanvas2-0 libgnomecanvas2-common libgnomeui-0 libgnomeui-common libgnomevfs2-0 libgnomevfs2-common libgnomevfs2-extra libidl-2-0 liborbit-2-0 liborbit2 python-gconf
    python-pyorbit
    Suggested packages:
    gconf-defaults-service libbonobo2-bin libgnomevfs2-bin python-gnome2-doc
    The following NEW packages will be installed:
    gconf-service gconf2 gconf2-common gnome-mime-data libart-2.0-2 libbonobo2-0 libbonobo2-common libbonoboui2-0 libbonoboui2-common libgconf-2-4 libgnome-2-0 libgnome-keyring-common libgnome-keyring0
    libgnome2-common libgnomecanvas2-0 libgnomecanvas2-common libgnomeui-0 libgnomeui-common libgnomevfs2-0 libgnomevfs2-common libgnomevfs2-extra libidl-2-0 liborbit-2-0 liborbit2 python-gconf python-gnome2
    python-pyorbit
    0 upgraded, 27 newly installed, 0 to remove and 0 not upgraded.
    Need to get 8,280 kB of archives.
    After this operation, 37.8 MB of additional disk space will be used.
    Do you want to continue? [Y/n] y
    Get:1 http://ftp.dk.debian.org/debian buster/main arm64 gconf2-common all 3.2.6-5 [1,039 kB]
    Get:2 http://ftp.dk.debian.org/debian buster/main arm64 libgconf-2-4 arm64 3.2.6-5 [421 kB]
    Get:3 http://ftp.dk.debian.org/debian buster/main arm64 gconf-service arm64 3.2.6-5 [410 kB]
    Get:4 http://ftp.dk.debian.org/debian buster/main arm64 gconf2 arm64 3.2.6-5 [423 kB]
    Get:5 http://ftp.dk.debian.org/debian buster/main arm64 gnome-mime-data all 2.18.0-2 [363 kB]
    Get:6 http://ftp.dk.debian.org/debian buster/main arm64 libart-2.0-2 arm64 2.3.21-4 [59.5 kB]
[... 123 lines cut ...]
    Processing triggers for libc-bin (2.28-10) ...
    Processing triggers for man-db (2.8.5-2) ...
    Setting up gconf-service (3.2.6-5) ...
    Setting up python-gconf (2.28.1+dfsg-1.2) ...
    Setting up gconf2 (3.2.6-5) ...
    X11 connection rejected because of wrong authentication.
    Setting up libgnomevfs2-common (1:2.24.4-6.1) ...
    Setting up libgnomevfs2-0:arm64 (1:2.24.4-6.1+b2) ...
    Setting up libgnome2-common (2.32.1-5) ...
    Setting up libgnomevfs2-extra:arm64 (1:2.24.4-6.1+b2) ...
    Setting up libgnome-2-0:arm64 (2.32.1-5+b1) ...
    Setting up libbonoboui2-0:arm64 (2.24.5-4) ...
    Setting up libgnomeui-0:arm64 (2.24.5-3.1) ...
    Setting up python-gnome2 (2.28.1+dfsg-1.2) ...
    Processing triggers for libc-bin (2.28-10) ...

removed 
    
    deb http://ftp.dk.debian.org/debian/ stretch main

from /etc/apt/sources.list

    sudo apt update

installed machinekit arm64 buster debs and tested axis gui fine

catcute3521, 2019-10-21:
I install the runtime packages and get the error with Debian Buster
root@chingluh35:/home/h# apt-get install machinekit-rt-preempt
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 machinekit-rt-preempt : Depends: machinekit (= 0.1.1570528621.git9239acb-1~buster) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

the-snowwhite, 2019-10-21:
Are you trying to install for i386, amd64, armhf or arm64 ?
Did you remember to add the machinekit repo and do a

    apt-get update

?

catcute3521, 2019-10-22:
yes, i install for amd64, and add the machinkit to repo as instruction:
Add Machinekit repository

sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv 43DDF224
sudo sh -c \
  "echo 'deb http://deb.machinekit.io/debian buster main' > \
  /etc/apt/sources.list.d/machinekit.list"
sudo apt-get update
------------------
it is look fine and no error.

the-snowwhite, 2019-10-22:
It looks like [that package is there](http://deb.machinekit.io/debian/pool/main/m/machinekit/machinekit_0.1.1570528621.git9239acb-1~buster_amd64.deb) what if you try:

    apt install machinekit machinekit-rt-preempt

?

cerna, 2019-10-23:
Yeah, it looks like there is an error in version mismatch.

@the-snowwhite 
That does not work.

I will investigate if the error is affecting only the latest packages or if it goes deeper.

the-snowwhite, 2019-10-25:
@cerna  OK Great