machinekit/machinekit #1332: halcmd - Xenomai: halcmd[1311] cannot map MAYDAY page

jallwine, 2017-12-14:
I'm getting errors when starting machinekit as a systemd service. I don't see them when I startup manually. I'm not sure where the problem is, or even where to start looking. 

I'm encountering these errors when using the latest machinekit image built for the Beaglebone Black (https://rcn-ee.com/rootfs/bb.org/testing/2017-02-12/machinekit/bone-debian-8.7-machinekit-armhf-2017-02-12-4gb.img.xz)

I ran the steps in this [document](https://github.com/PocketNC/pocketnc/blob/master/README.md) to set up the machinekit configuration for the Pocket NC, which enables systemd services that start up machinekit. When the services aren't enabled, machinekit seems to work, but when turned on the following errors happen at boot up and the system is very unstable. I can't ssh in, and it locks up after a short period of time when I'm interacting directly with the Beaglebone with a monitor and keyboard (I basically have enough time to disable the services with sudo ~/pocketnc/disableServices.sh && sudo reboot).

These are the first few lines of the error from dmesg (full stack trace [here](https://github.com/machinekit/machinekit/files/1559856/error.txt)):
[   19.501859] Xenomai: halcmd[1311] cannot map MAYDAY page
[   19.956681] Unable to handle kernel NULL pointer dereference at virtual address 00000004
[   19.965136] pgd = de414000, hw pgd = de414000
[   19.969675] [00000004] *pgd=9e7c4831, *pte=00000000, *ppte=00000000
[   19.976229] Internal error: Oops: 817 [#1] THUMB2
[   19.981128] Modules linked in: g_ether libcomposite cpufreq_userspace

ArcEye, 2017-12-14:
This appears to be a Xenomai kernel error, rather than anything to do with machinekit per se.

Google `Xenomai:  cannot map MAYDAY page` 
or 
`Xenomai: Unable to handle kernel NULL pointer dereference at virtual address 00000004`
or
`Xenomai: Internal error: Oops: 817 [#1] THUMB2`
and you get quite a lot of similar hits over different systems.  They are all Xenomai / ARM related.

Have you tried a later image with a rt-preempt kernel?
You will probably have to go onto the forum with this and hopefully someone will give you a link to a much newer image which is not xenomai based.
https://groups.google.com/forum/#!forum/machinekit

jallwine, 2017-12-14:
I was having problems with the newer rt-preempt kernel as well, but I may try again and dig into that rather than dig too deep with this.

closed 2018-01-05 by ArcEye