machinekit/machinekit #1334: sudo make setuid failed

SteveBetter, 2017-12-19:
Hi, everyone!
  I installed xenomai 3.0.6 previously and build machinekit from source later with confifuration
"./configure --with-xenomai --with-xeno-config=/usr/xenomai/bin/xeno-config".

  When I run "make" successfully, and "sudo make setuid", I get the following error, and can't run machinekit normally. Can anyone help?

robot@ubuntu:~/Downloads/machinekit-master/src$ sudo make setuid
../scripts/check-build-vs-configure-sha: line 11: not: command not found

***********************************************************
WARNING: configure was last run at Tue Dec 19 19:11:59 CST 2017 at git version 
WARNING: now the git version is: not in a git repo
WARNING: consider executing 'make clean; sh autogen.sh; ./configure <args>' before running make
***********************************************************

make: Entering directory '/home/robot/Downloads/machinekit-master/src'
test -f ../libexec/pci_read && chown root ../libexec/pci_read && chmod 4750 ../libexec/pci_read || true
test -f ../libexec/pci_write && chown root ../libexec/pci_write && chmod 4750 ../libexec/pci_write || true
test -f ../libexec/rtapi_app_xenomai && chown root ../libexec/rtapi_app_xenomai && chmod 4750 ../libexec/rtapi_app_xenomai || true;
Warning:  Logging appears to be not working

          Found rsyslogd configuration and /var/log/linuxcnc.log
          exists, but a test did not appear in the log.
          Please investigate.

Makefile:842: recipe for target 'setuid' failed
make: [setuid] Error 1 (ignored)
make: Leaving directory '/home/robot/Downloads/machinekit-master/src'

ArcEye, 2017-12-19:
You need to take this to the forum, it is an error in your setup, not an issue with machinekit

The first non fatal warning is because the SHA at configure does not match the repo
Most commonly occurs when you run configure then add something or change something and don't run configure again

The setuid process will not occur if the tests fail
Looks like your rsyslog daemon is not running, your /var/log/linuxcnc.log does not have correct permissions or similar

ArcEye, 2017-12-19:
PS.
To see what the test was that failed, see https://github.com/machinekit/machinekit/blob/master/scripts/check-system-configuration.sh#L73

The two causes I mentioned above are the most usual culprits.
linuxcnc.log should be have permissions of 644 set and the below will check that rsyslog is running.
```
root@INTEL-i7:/# service rsyslog status
● rsyslog.service - System Logging Service
   Loaded: loaded (/lib/systemd/system/rsyslog.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2017-12-19 13:56:54 GMT; 4h 4min ago
     Docs: man:rsyslogd(8)
           http://www.rsyslog.com/doc/
 Main PID: 573 (rsyslogd)
    Tasks: 4 (limit: 4915)
   CGroup: /system.slice/rsyslog.service
           └─573 /usr/sbin/rsyslogd -n

Dec 19 13:56:54 INTEL-i7 systemd[1]: Starting System Logging Service...
Dec 19 13:56:54 INTEL-i7 rsyslogd[573]: command 'SystemLogRateLimitInterval' is currently not permitted - did you already set it via a RainerScript command (v6+ config)? [v8.31.0 try http://www.rsyslog.com/e/22
Dec 19 13:56:54 INTEL-i7 rsyslogd[573]: command 'SystemLogRateLimitBurst' is currently not permitted - did you already set it via a RainerScript command (v6+ config)? [v8.31.0 try http://www.rsyslog.com/e/2222 
Dec 19 13:56:54 INTEL-i7 rsyslogd[573]:  [origin software="rsyslogd" swVersion="8.31.0" x-pid="573" x-info="http://www.rsyslog.com"] start
Dec 19 13:56:54 INTEL-i7 systemd[1]: Started System Logging Service.
```

SteveBetter, 2017-12-20:
@ArcEye Thank you for you reply. I will go to the forum. Let me finish this problem first.
I change mode of /var/log/linuxcnc.log. sudo make setuid runs with no error, but machinekit can't run successfully neither.
I must say that I did not change or add anything.
../scripts/check-build-vs-configure-sha: line 11 is below:
11 GIT_CONFIG_SHA=not in a git repo

robot@ubuntu:~/Downloads/machinekit-master/src$ sudo chmod 777 /var/log/linuxcnc.log 
robot@ubuntu:~/Downloads/machinekit-master/src$ sudo make setuid
../scripts/check-build-vs-configure-sha: line 11: in: command not found

***********************************************************
WARNING: configure was last run at Wed Dec 20 09:01:15 CST 2017 at git version 
WARNING: now the git version is: not in a git repo
WARNING: consider executing 'make clean; sh autogen.sh; ./configure <args>' before running make
***********************************************************

make: Entering directory '/home/robot/Downloads/machinekit-master/src'
test -f ../libexec/pci_read && chown root ../libexec/pci_read && chmod 4750 ../libexec/pci_read || true
test -f ../libexec/pci_write && chown root ../libexec/pci_write && chmod 4750 ../libexec/pci_write || true
test -f ../libexec/rtapi_app_xenomai && chown root ../libexec/rtapi_app_xenomai && chmod 4750 ../libexec/rtapi_app_xenomai || true;
make: Leaving directory '/home/robot/Downloads/machinekit-master/src'

robot@ubuntu:~/Downloads/machinekit-master$ . ./scripts/rip-environment
robot@ubuntu:~/Downloads/machinekit-master$ machinekit
MACHINEKIT - 0.1
Machine configuration directory is '/home/robot/Downloads/machinekit-master/configs/sim'
Machine configuration file is 'emcweb.ini'
Starting Machinekit...
Warning - /home/robot/Downloads/machinekit-master/libexec/rtapi_app_posix not setuid
'sudo make setuid' missing?
/home/robot/Downloads/machinekit-master/scripts/realtime: line 192: /home/robot/Downloads/machinekit-master/libexec/rtapi_app_posix: No such file or directory
rtapi_app startup failed - aborting
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
io started
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
halcmd loadusr io started
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
Shutting down and cleaning up Machinekit...
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
Warning - /home/robot/Downloads/machinekit-master/libexec/rtapi_app_posix not setuid
'sudo make setuid' missing?
Cleanup done
Machinekit terminated with an error.  You can find more information in the log:
    /home/robot/linuxcnc_debug.txt
and
    /home/robot/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal

I

ArcEye, 2017-12-20:
> I installed xenomai 3.0.6 previously and build machinekit from source later with confifuration
"./configure --with-xenomai --with-xeno-config=/usr/xenomai/bin/xeno-config".
```
/home/robot/Downloads/machinekit-master/libexec/rtapi_app_posix: No such file or directory
rtapi_app startup failed - aborting
```
Don't know what you are actually doing, looks like trying to run an old version of xenomai on Ubuntu.

Note you are not starting a rtai_app_xenomai but the posix version which does not exist, so it fails.

You need to take this to the forum and explain exactly what you are trying to do and on what platform etc.

closed 2018-01-05 by ArcEye