machinekit/machinekit #1340: Website: add google groups to search

machinekoder, 2018-01-02:
I think it would be nice if the search functionality on the Machinekit websites also includes the Machinekit Google Group because it contains lot's of valuable information not in the docs.

ArcEye, 2018-01-02:
I can add it to the search params in the next couple of days.
May take a while to propagate it, that is down to Googles netbots.

ArcEye, 2018-01-03:
The problem with this is that we can never own the site we want indexed, since Google does.

If I can get access to the header here https://groups.google.com/forum/#!forum/machinekit
I may be able to insert a meta-tag into it which will satisfy the verification process.
Have a feeling only @mhaberler has that, he probably set it up

[image]

ArcEye, 2018-01-03:
I have sent him and email, so see how we go

machinekoder, 2018-01-03:
Okay good. Right now the search searches all google groups it seems.

mhaberler, 2018-01-03:
@ArcEye @machinekoder  you are now owners of the google group

ArcEye, 2018-01-04:
That does not work unfortunately.
The welcome message appears to be pretty much plain text and does not come within the index.html.
Even entering the meta-tag into the text does not result in verification.

All the other methods require ownership or access to the domain which we do not have.

As Google provides a search facility with the forum, I don't suppose they are much concerned about others specifically searching it from another engine.

If we can find a search syntax that only yields results from that forum, we could set up a link.
However `https://groups.google.com/forum/#!forum/machinekit: HAL` and similar do not bring up results exclusively from our forum.

Open to ideas.

ArcEye, 2018-01-18:
Finally fixed it via a separate route.

Adding the specific forum to the google search proved impossible due to the ownership issues as above.
Adding the forum url into the search specs resulted in just the first bit of the url being parsed by google and results from all forums being displayed.

Hit upon a search url which resulted in specific results from our forum only.
`https://www.google.co.uk/search?&q=site%3Agroups.google.com%2Fd%2Fmsg%2Fmachinekit%2F%20`

The final char is a space and anything appended to it is searched

Got rid of the clunky branded search input box for the google cse and used a plain size limited input for google site search and another for Forum search.

After much buggering about to try to get alignment between disparate elements in the page, the result works 😄 

[image]

https://github.com/machinekit/machinekit-docs/pull/278 refers

sirop, 2018-01-18:
Overlaps  in Opera 50.0 on Windows 8.1

[image]

ArcEye, 2018-01-18:
That is not the page I created, it has not been rendered yet.

Look at http://preview.machinekit.io

sirop, 2018-01-18:
The it looks like 

[image]

in Opera 50.0 .

ArcEye, 2018-01-18:
Opera seems to have decided there is room for the first box and Firefox thinks it only fits on the next line
I'll experiment with pushing it down

ArcEye, 2018-01-18:
OK, adding a link to the Forum in the top line, pushes the search boxes down in Opera

[image]

and also displays OK still in Firefox

[image]

I'll push another PR and test

ArcEye, 2018-01-18:
https://github.com/machinekit/machinekit-docs/pull/279

ArcEye, 2018-01-18:
Now built and tested.
I reverted to the overlay results method, because it worked best in both browser types.

If checking, make sure you clear your cache and force refresh the browser (Ctrl F5 in Firefox), 
or you may get peculiar results since not only the webpage code but the google cse settings have changed.

Interestingly there is a distinct difference to the results pages from Firefox with adblocker enabled

[image]

and Opera without.

[image]

If you entered `comp` instead of `instcomp`, you would get half a page of 'ambulance chasing' law firms adverts prior to the actual results in Opera.

Even disabling adblocker, does not produce adverts in Firefox

Pretty much says it all about google's sole motivation and why I don't touch Chrome or anything derived from it.

ArcEye, 2018-08-01:
Resolved as far as can be

closed 2018-08-01 by ArcEye