machinekit/machinekit #1341: libegl1 package problem when installing MK on the beagleboard X15

luminize, 2018-01-03:
When I follow the step `sudo mk-build-deps -ir` I get an error from which I don't know how to recover.
This step is trying to install libegl but there seems to be a problem replacing the file of the one already in libegl1-mesa:armhf 17.1.3-2rcnee0~stretch+20170627

My initial setup has been from a "stock" debian stretch beagleboard X15 image from https://beagleboard.org/latest-images
```
debian@BeagleBoard-X15:~/machinekit$ uname -a
Linux BeagleBoard-X15 4.9.35-ti-r44 #1 SMP PREEMPT Sat Jul 1 00:54:33 UTC 2017 armv7l GNU/Linux
```
the specific error:
```
Get:1 http://repos.rcn-ee.com/debian stretch/main armhf libegl1 armhf 0.2.999+git20170802-2 [27.2 kB]
Fetched 27.2 kB in 0s (53.7 kB/s)  
(Reading database ... 97424 files and directories currently installed.)
Preparing to unpack .../libegl1_0.2.999+git20170802-2_armhf.deb ...
Unpacking libegl1:armhf (0.2.999+git20170802-2) ...
dpkg: error processing archive /var/cache/apt/archives/libegl1_0.2.999+git20170802-2_armhf.deb (--unpack):
 trying to overwrite '/usr/lib/arm-linux-gnueabihf/libEGL.so.1.0.0', which is also in package libegl1-mesa:armhf 17.1.3-2rcnee0~stretch+20170627
```
`sudo apt --fix-broken install` gets me in an eternal broken packages loop :)
full output of the installation step below:
```
debian@BeagleBoard-X15:~/machinekit$ sudo mk-build-deps -ir
dh_testdir
dh_testroot
dh_prep
dh_testdir
dh_testroot
dh_install
dh_install: Compatibility levels before 9 are deprecated (level 7 in use)
dh_installdocs
dh_installdocs: Compatibility levels before 9 are deprecated (level 7 in use)
dh_installchangelogs
dh_compress
dh_fixperms
dh_installdeb
dh_installdeb: Compatibility levels before 9 are deprecated (level 7 in use)
dh_gencontrol
dh_md5sums
dh_builddeb
dpkg-deb: building package 'machinekit-build-deps' in '../machinekit-build-deps_1.0_all.deb'.

The package has been created.
Attention, the package has been created in the current directory,
not in ".." as indicated by the message above!
Selecting previously unselected package machinekit-build-deps.
(Reading database ... 97420 files and directories currently installed.)
Preparing to unpack machinekit-build-deps_1.0_all.deb ...
Unpacking machinekit-build-deps (1.0) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies...Starting pkgProblemResolver with broken count: 0
Starting 2 pkgProblemResolver with broken count: 0
Done
 Done
Starting pkgProblemResolver with broken count: 0
Starting 2 pkgProblemResolver with broken count: 0
Done
The following additional packages will be installed:
  libegl1
The following NEW packages will be installed:
  libegl1
0 upgraded, 1 newly installed, 0 to remove and 171 not upgraded.
147 not fully installed or removed.
Need to get 27.2 kB of archives.
After this operation, 71.7 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://repos.rcn-ee.com/debian stretch/main armhf libegl1 armhf 0.2.999+git20170802-2 [27.2 kB]
Fetched 27.2 kB in 0s (73.8 kB/s)
(Reading database ... 97424 files and directories currently installed.)
Preparing to unpack .../libegl1_0.2.999+git20170802-2_armhf.deb ...
Unpacking libegl1:armhf (0.2.999+git20170802-2) ...
dpkg: error processing archive /var/cache/apt/archives/libegl1_0.2.999+git20170802-2_armhf.deb (--unpack):
 trying to overwrite '/usr/lib/arm-linux-gnueabihf/libEGL.so.1.0.0', which is also in package libegl1-mesa:armhf 17.1.3-2rcnee0~stretch+20170627
Errors were encountered while processing:
 /var/cache/apt/archives/libegl1_0.2.999+git20170802-2_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
(Reading database ... 97424 files and directories currently installed.)
Removing machinekit-build-deps (1.0) ...
mk-build-deps: Unable to install all build-dep packages
```

cdsteinkuehler, 2018-01-03:
On 1/3/2018 5:56 AM, Bas de Bruijn wrote:
> When I follow the step `sudo mk-build-deps -ir` I get an error from which I don't know how to recover.
> This step is trying to install libegl but there seems to be a problem replacing the file of the one already in libegl1-mesa:armhf 17.1.3-2rcnee0~stretch+20170627

That's a good question for the BeagleBoard Google Group...it's not
just for the BeagleBone, it covers other boards as well, including the
X15.

It looks like you've got a custom package from Robert installed (based
on rcnee0 in the version string).  I suspect there needs to be some
apt pinning setup or something.

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2018-01-03:
The quick-fix solution where dpkg refuses to overwrite an existing file is usually just to move it or rename it and run install again.  
Then replace the new one with the original, which presumably has some required change or RCN would use the stock package.

Doesn't fix any long term incompatibility through.

luminize, 2018-01-03:
@cdsteinkuehler I'll ask on the beagleboard group. @ArcEye I'll try your fix.
Thanks both!

luminize, 2018-01-03:
solution here:
https://groups.google.com/forum/#!category-topic/beagleboard/beagleboard-x15/8J9Clq3NI0w

> This happened when i merged mesa: 17.3.x over 17.1.3 
>
> It should resolve with: 
>
> sudo apt update 
> sudo apt dist-upgrade 
> sudo apt --fix-broken install 
>
> For "i.mx6" hardware 17.3.x brought us "Desktop" OpenGL 2.0... sadly 
> only software bug fixes for the X15...

closed 2018-01-03 by luminize