machinekit/machinekit #1343: keystick cannot handle more than three axis

l29ah, 2018-01-04:
It is impossible to control those, so it's pretty much useless on such a machine as one needs to home the unavailable axis as well somehow to be able to run gcode.

ArcEye, 2018-01-04:
You have given us no information, so how can we have any idea what you are trying to do and on what machine?
Keystick is a legacy program from linuxcnc, we don't claim it is useful for anything.

ArcEye, 2018-01-04:
This might give some more info than the very minimal docs we inherited from linuxcnc.
http://www.cnc-club.ru/forum/viewtopic.php?f=15&t=4346&hilit=keystick&p=104568
However it seems to be mostly concerned with the latency benefit of using it from what I can tell.

Keystick was written for a 3 axis mill.  If you want to run some unspecified machine with more than 3 axes, you will need Axis, Cetus, Machineface, gmoccapy or one of the GUIs which does cater for that.

The forum is really the place for any queries on those.
https://groups.google.com/forum/#!forum/machinekit

closed 2018-01-04 by ArcEye