machinekit/machinekit #1347: A machinekit configuration cannot be launched via machinekit-client if there is an underscore in the path to the ini-file.

everclear72216, 2018-01-06:
Reproduction:
- copy a known working configuration to a directory that contains an underscore somewhere in its path
- create a mklauncher configuration (launcher.ini and launcher.py) that launches the newly created configuration
- start mklauncher with the new configuration
- start machinekit-client on the remote host and select to launch the advertised configuration

Observation:
- the client will show that the sever is launching the configuration an after a few seconds it will show "returned with exit code 1"

Expectation:
- It just works

machinekoder, 2018-01-16:
Can you please give an example. I was not able verify this with "foo_bar" as directory name.

everclear72216, 2018-01-17:
Thank you for looking into it. I will do so next weekend.