machinekit/machinekit #1348: GSoC project ideas

machinekoder, 2018-01-18:
We will apply for taking part in Google Summer of Code this year.

Please add any project ideas you have to this issue and rate them by complexity. We should have easy, low hanging fruits and also intermediate and complex task to make this interesting for students.

Alessandro-Barbieri, 2018-01-22:
Package as an official Debian package

Alessandro-Barbieri, 2018-01-22:
Add machinekit as a package to rosdistro

Alessandro-Barbieri, 2018-01-22:
Merge bugfixes from linuxcnc

machinekoder, 2018-01-23:
Package or otherwise make Machinekit-HAL easier to use beyond just
Debian (ie: make a Yocto meta-layer, gentoo portage configuration, etc)

machinekoder, 2018-01-23:
Build a ROS-Controller node on top of Machinekit-HAL

machinekoder, 2018-01-23:
Port (smoothie|grbl|sailfish|etc) to run on top of Linux +
Machinekit-HAL

machinekoder, 2018-01-23:
Create example projects for Machinekit HAL.

machinekoder, 2018-01-23:
having PRU on a beaglebone interrupt firing the servo thread (like the fpga De0-nano and mesa can do)

machinekoder, 2018-01-23:
a (Qt/C++) library/framework for making applications as easy as reconfiguring HAL, but at the userland level. Could be the basis for more sophisticated projects like CNC stack :) Drag and drop application decision logic. Interfacing with realtime thru hal/machinetalk? Some sturdy
 design for others to build on.

machinekoder, 2018-01-23:
use Google blockly to write an user application, where blockly configures and starts HAL? Inputs from remote components and physical inputs.

machinekoder, 2018-01-23:
PLC open editor to use Machinekit? http://www.plcopen.org http://www.openplcproject.com/plcopen-editor https://beremiz.org/apps

machinekoder, 2018-01-23:
logging functionality, write data periodically to a database

machinekoder, 2018-01-23:
more language bindings? Golang? Erlang?

machinekoder, 2018-01-23:
LinuxCNC on top of Machinekit HAL

machinekoder, 2018-01-23:
A better "out of the box" experience.

machinekoder, 2018-01-23:
Move the project build system from Automake to CMake.

machinekoder, 2018-01-23:
Replace NML with Machinetalk.

ArcEye, 2018-01-23:
>@Alessandro-Barbieri 
Merge bugfixes from linuxcnc

What bugs, what fixes?

machinekoder, 2018-01-23:
I created a project page here: https://github.com/machinekit/machinekit/projects/1
@machinekit/maintainers please add your ideas there.

ArcEye, 2018-01-23:
> Move the project build system from Automake to CMake.

The current system works perfectly well.  
This would be time consuming, with no benefit to the actual code.

ArcEye, 2018-01-23:
> LinuxCNC on top of Machinekit HAL

Discussed off-list this morning.
If you could add Linuxcnc on top of Machinekit-HAL, you would just get Machinekit again.

Machinekit is Linuxcnc, but so far diverged HAL wise, as to be completely incompatible.

I think the goal would have to be to build a machinekit-cnc cnc stack which can interface with machinekit-hal.
If you can manage that, then porting from linuxcnc might be achievable, if it proves desirable.

However, if the cnc stack stays the same, there is no point building it, because it already exists combined with HAL in machinekit
The only way which makes sense to me is the replacement of the NML messaging system with machinetalk or similar, so that you can start and run both stacks, with communication between them.
There is also scope for more modular interpreters etc.

Consideration then:
If HAL is providing the hardware interface to a 5i25 board, but the interpreter / planner / motion are generating the move instructions, there has to be a communication layer between them and HAL runs them
or
does the CNC stack ask HAL for a hm2 driver and specific board driver and get passed a pointer to their interfaces and use them directly or some other way to share the resource.

I don't hold out any hope of Linuxcnc adopting our changes to HAL, or a wholesale replacement of NML and that is where the major divergence would come about.

The best we can probably hope for is that we end up with a machinekit-cnc stack basics that is similar enough to linuxcnc, that porting across any improvements is not major work, leaving us to concentrate on the machinekit-hal and messaging interface.

My thoughts anyway.

machinekoder, 2018-01-23:
@ArcEye I agree with you. This task is probably too big for GSoC anyway.

Alessandro-Barbieri, 2018-01-23:
@ArcEye https://github.com/LinuxCNC/linuxcnc/issues/379

ArcEye, 2018-01-24:
> @ArcEye LinuxCNC#379

For some reason my reply has disappeared

This is not suitable for a GSOC application, it is a bread and butter Issue subject to be raised by that medium.  See #1350

machinekoder, 2018-01-24:
I'll close this issue since the application is in progress and we don't need this preliminary ideas page anymore.

closed 2018-01-24 by machinekoder