machinekit/machinekit #1351: Latest hm2 hostmot2 firmware for mesanet cards

machinekoder, 2018-01-25:
What is the latest firmware (source) for the mesanet cards possibly with hardware threads synchronization (https://github.com/machinekit/machinekit/issues/687)?

I'm using the latest source tree from the mesanet website and it works, but I was not sure if there is anything newer as I will start to work on new features shortly.

ArcEye, 2018-01-25:
Linuxcnc have done a bit on smart serial recently plus some other fixes
https://github.com/LinuxCNC/linuxcnc/commits/master/src/hal/drivers/mesa-hostmot2

Will need to check that all changes are relevant to us.
You particularly need to be aware of the thread period stuff changing with multicore.

If you just copy the hostmot2.c file across, it will not work with Machinekit.
https://github.com/machinekit/machinekit/commit/bd6c9a55ce610a5c73c97a49b7ba3eb3e1d106ad#diff-7d9da9f3383bdf41ad3e4e4163cd607f

ArcEye, 2018-01-25:
Just spotted `firmware` so ignore the above.

I am still using the original firmware on my cards because it works. :)