machinekit/machinekit #1355: referencing LCNC issues in commits from a machinekit fork will add wrong relations

luminize, 2018-02-11:
referencing linuxcnc issues in commits (hashtag + number) on a machinekit fork links back to machinekit organisation repos issues while these numbers definitely don't match.
for example like here:
https://github.com/machinekit/machinekit/issues/134#ref-commit-c00bd8b

ArcEye, 2018-02-11:
I think this is a Github thing.
Any reference is assumed to be one to THAT repo.
You cannot get a proper reference to a commit or issue in another repo in the same organisation, unless you include the full URL and that sometimes does not work.

luminize, 2018-02-11:
Ok, well it's really annoying when you read thru an issue and see all nonsense references. I'll look if we can turn the referencing from forked repo's off. Then we can decide if we want that.

cdsteinkuehler, 2018-02-11:
On 2/11/2018 11:25 AM, Bas de Bruijn wrote:
> Ok, well it's really annoying when you read thru an issue and see
> all nonsense references. I'll look if we can turn the referencing
> from forked repo's off. Then we can decide if we want that.

I think it works properly if you use the full user/repo#issue format
(or user/repo@commit) ie:

linuxcnc/linxucnc#134

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2018-08-03:
@luminize is there anything further on this or can it be closed?

closed 2018-08-03 by luminize