machinekit/machinekit #1359: Q about possible missing symbols in liblinuxcnchal.so

luminize, 2018-03-16:
I've succesfully created a C++ component (shared lib, can be loaded in hal, funct added to a thread etc) and I want to load this not via halcmd but thru a C++ executable, similar like rtapi_app_main() does.
```C++
# include <control_component2.h>

ControlComponent2 *cr;

int main()
{
    cr = new ControlComponent2();
}
```
Linking the executable fails:
```
[ 75%] Linking CXX executable create_comp
libcontrol_component2.so: undefined reference to `hal_export_functf'
```

looking into the symbols of liblinuxcnchal.so I can find no `hal_export_functf` although it should be there according [these lines](https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal_lib.c#L420-L426) in hal_lib.c
```
EXPORT_SYMBOL(hal_export_funct);
EXPORT_SYMBOL(hal_export_functf);
EXPORT_SYMBOL(hal_export_xfunctf);
EXPORT_SYMBOL(halg_export_xfunctf);
EXPORT_SYMBOL(hal_add_funct_to_thread);
EXPORT_SYMBOL(hal_del_funct_from_thread);
EXPORT_SYMBOL(hal_call_usrfunct);
```

for reference, these symbols are available:
```
bas@xw6600:~/projects/cmake-component$ objdump -TC ../machinekit/lib/liblinuxcnchal.so | grep halg_xinit
000000000001a2fb g    DF .text	0000000000000a7b  Base        halg_xinitfv
000000000001a1eb g    DF .text	0000000000000110  Base        halg_xinitf
```

I'd like to fix that, any ideas where?

ArcEye, 2018-03-16:
hal.h :L960
Have a look at the `#ifdef RTAPI` a few lines above.

Having `-DRTAPI` in your CFLAGS building components, does not change how the RIP of Machinekit you are trying to link against was built.
If it was not built with `RTAPI` defined, a lot will be missing.

Run `nm -C liblinuxcnchal | grep hal_export_functf`

You may well find it is missing.

luminize, 2018-03-16:
indeed it is. So how do I set the flags to do so?

ArcEye, 2018-03-16:
I just hacked last time I needed them exported and commented out the `#ifdef RTAPI` around the bits I wanted.

The better way would be to add `-DRTAPI` to the CFLAGS in configure or similar so long as that does not have a knock on elsewhere.

luminize, 2018-03-16:
Found the CFLAGS in the Makefile, changing DULAPI to DRTAPI gave me all complaints (defining thread_flavor_id and such, and also gave errors). Couldn't find another place where of how to configure this. So following your solution for now, commenting out the offending `#ifdef RTAPI` sections

ArcEye, 2018-03-16:
For C code, if you used instcomp, which uses Makefile.modinc etc., it will compile and link the `hal_export_(x)functf` calls, which it does in fact for all instantiated components.

I haven't bothered drilling down to see exactly where it differs, I just took the quick fix when I was messing about with C++ components and hit the same problem.

luminize, 2018-03-16:
Success! (sort of, I could not find to compile for RTAPI, so I took the shortcut).

I commented out these lines:
https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal.h#L960
https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal.h#L1045
and then it took me some time to realize to take out these too (after some swearing and "why o why did the symbol not get exported" yelling) 
https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal_funct.c#L11
https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal_funct.c#L146

running the program (with realtime already started) now loads the comp and exposes the function, which can be added to a thread.

Thanks @ArcEye

closed 2018-03-16 by luminize