machinekit/machinekit #1364: Get rid of multiple rtapi_app and comp binaries

zultron, 2018-03-30:
I want to reopen the argument for folding together per-flavor binaries (comps and `rtapi_app`) into one that will run in all (userland) thread flavors.  This issue is related to #874, but doesn't address the v1 vs. v2 question, and adds the issue of combining per-flavor comps.  Some of the discussion there is directly related to this issue.

The primary motivation is to reduce unnecessary complication in both build and run-time systems.  @ArcEye could surely rant at length about how the current scheme has complicated build-related things like packaging and out-of-tree `icomp` builds.  In a recent whirl with manually building a C++ comp out of tree, @luminize and I found we had to define `THREAD_FLAVOR_ID`, which is ultimately meaningless to the build.  Dealing with these problems has already cost several people many hours of time dealing with these problems, and I agree these problems are *not* solved and will continue to plague us.

Folding together the `rtapi_app_*` binaries also addresses issue (1) in #874.  It doesn't solve issue (2) there by itself, but it may or may not help by reducing the problem down to only the v1/v2 comp separation.

The main points for how to do this are as follows:
- Modify the build system to build just one copy of comps.  A minimal flavor-specific module, e.g. `flavor-rt-preempt.so`, containing essentially the code in e.g. `src/rtapi/rt-preempt.[ch]`, will be built and stuffed into the same directory with other comps.
- Modify the flavor detection system to make data about the flavor available at run-time to rtapi.  That data would come from the above-mentioned `flavor-*.so` module, and be shared through an interface similar to the current `rtapi_switch`.
- In RTAPI, esp. `rtapi_app`, eliminate any per-flavor `#ifdef` statements, replacing them with run-time conditionals based on the above.  Preferably, isolate anything like this into the `flavor-*.so` module so that flavor-specific code isn't spread out all over rtapi and hal sources.  The `rtapi_app_*` binaries may then be merged.
- At run-time, detect thread flavor as we do now, and load the appropriate `flavor-*.so` module.
- Fix run-time init scripts (`realtime`) as needed.

I don't see any real risks here.  In-tree, per-flavor comp builds are in fact binary identical, and the same binary will run with any thread flavor.  The most concrete concern presented in #874, that `#ifdef flavor == foo` will break something, actually wouldn't be possible:  any such build-time, flavor-specific macros wouldn't exist, since they would be converted to run-time `if` conditionals using run-time flavor data.

This will also address legitimate concerns about reducing build time or disk space usage, including @ArcEye's distaste for the default building of flavors irrelevant to the running kernel.  Even if the current default stays the same, and autoconf detects and builds all flavors whose required build dependencies are met, the extra time and disk space will be only that needed to build at most two extra `flavor-*.so` modules.  In terms of source size, `xenomai.[ch]` and `rt-preempt.[ch]` are small at about 500 LOC each.

ArcEye, 2018-04-01:
> @ArcEye could surely rant at length................

Moi? Surely not!

Long time since I looked at #874, we are certainly no further down the road towards most of it.

I too had fun attempting to build C++ components, when something as basic as `hal_export_xfunctf()`,
without which no instantiated component can run, is wrappered in `#ifdef RTAPI`.
Then if you define `RTAPI`, you also need to define `THREAD_FLAVOR_ID` or that fails too.

A brief look at `Makefile`, `Makefile.modinc.in` and `Makefile.inc` shows quite a lot of redundent checks for kernel threads which are no longer used.  
This holds for `configure.ac` and `debian/configure` too.

Perhaps as a first step, I will remove all the hard kernel thread artifacts and in doing so remove at least 2 conditional tests ( _#ifeq( $(BUILD_SYS), kbuild)_ and _#ifeq( $(BUILD_SYS), user_dso)_ ).

zultron, 2018-08-03:
IMO this is still worth doing, and a combination of @ArcEye's work on a stand-alone HAL build and @bobvanderlinden's [cmake build system][1] would be lean, clean and beautiful.

It would be even more worthwhile if LinuxCNC could be taught to be built upon Machinekit HAL, so we could enjoy the latest LCNC bugfixes and updates, but keep the powerful Machinekit HAL underneath.

At one point, I recall @machinekoder asking the LCNC community if there's interest in a QtQuickVCP port, and the highly-conditional display of interest was about as positive as it gets.  If that could happen, QtQuickVCP would enjoy brand-new popularity with a much larger community than it is currently exposed to.

And with all these dreams come true at once, the entire CNC application could be dropped, relieving the project of a major burden.

[1]: https://github.com/bobvanderlinden/machinekit/commits/cmake