machinekit/machinekit #1374: rtapi_print_msg() not working from within comps

zultron, 2018-05-11:
`rtapi_print_msg()` doesn't seem to work from within a .comp.  The following test demonstrates the problem in Machinekit, and works properly in LCNC.

`test.comp`:
``` C
component test;
pin out bit token_pin;

function _;
license "GPL";
;;

FUNCTION(_) {
  rtapi_print_msg(RTAPI_MSG_ERR, "Testing rtapi_print_msg()\n");
}
```

`test.hal`:
```
loadrt threads name1=t period1=10000000
loadrt test names=test
addf test t
start
loadusr -w sleep 0.5
```

`test.sh`:
``` shell
#!/bin/bash -e

. scripts/rip-environment
COMP="$(which halcompile || which comp)"
$COMP --install test.comp
halrun -f test.hal
```

sirop, 2018-05-11:
Does it depend on HAL debug level?
 I.e., would
`rtapi_print_msg(RTAPI_MSG_INFO, "Testing rtapi_print_msg()\n");`
work?

zultron, 2018-05-12:
This is a "duh" moment for me.  I originally had `export SYSLOG_TO_STDERR=1 DEBUG=5` early in `test.sh`, then needlessly took it out while testing in LCNC (which likes `halrun -V` instead), and then forgot to put it back when running against Machinekit again.  Sorry for the noise.

closed 2018-05-12 by zultron