machinekit/machinekit #1393: Please do NOT open any new issues in machinekit - open issue in either HAL or CNC stack

ArcEye, 2018-08-04:
All issues are being moved to either machinekit-hal or machinekit-cnc repos

Some issues which span those stacks are held in machinekit-cnc but referenced also in machinekit-hal

4 issues have been left in machinekit

These have so many comments that moving them proved difficult without loss, they will remain here

jallwine, 2019-10-02:
Is this still the recommended way to build from source? http://www.machinekit.io/docs/developing/machinekit-developing/

It references cloning this repository. If it's been split up into machinekit-hal and machinekit-cnc, is that still correct?

luminize, 2019-10-02:
> On 2 Oct 2019, at 19:46, John Allwine <notifications@github.com> wrote:
> 
> Is this still the recommended way to build from source? http://www.machinekit.io/docs/developing/machinekit-developing/
> 
> It references cloning this repository. If it's been split up into machinekit-hal and machinekit-cnc, is that still correct?
> 

Hi John,

most changes have gone into mk-hal. If you don’t care about the newest, then i’d recommend building from the Machinekit repo.
The CNC stack has not changed a lot iirc.

We haven’t really advertised this.

If you want to build the machinekit-cnc stack, you can do that by cloning Machinekit-hal and running `scripts/build_with_cnc` script. This pulls in the Machinekit CNC stack.

I’m not sure what the best way is for committing and keeping both stacks separate. Maybe @ArcEye can chime in?

Anything specific you want to develop?

Cheers,
Bas

jallwine, 2019-10-02:
I'd like to see the changes that were implemented in LinuxCNC to address this issue integrated into MachineKit: https://github.com/LinuxCNC/linuxcnc/issues/550

Another thing I'm looking into is increasing the max compensation table size for an axis. We have a continuous rotation B axis on the Pocket NC that has soft limits of -9999 to 9999. In order to compensate for that full range with the current max compensation table size of 256, we can only use 4 points per revolution. A larger compensation table would be the easiest way to address this, but the ideal solution would be to be able to mark an axis as continuous some how and be able to apply the full compensation table out to any absolute location.

Anyway, just want to make sure I'm following the recommended practices. Thanks for the info!